zwave-js/zwave-js #1369: [bug] Aeotec Smart Switch 6 - Fails to complete Interview

tommyjlong, 2021-01-13:
**Version**
1.0.0-alpha.2.51f715c

Build/Run method
- [ ] Docker
- [ ] PKG
- [x] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: 1.0.0-alpha.2.51f715c

**Describe the bug**
After a few hours, all my devices have reached "Complete" but
 the Aeotec Smart Switch 6 still never reaches "Complete".
Note: This is a case of transitioning from OZW/zwave2mqtt to zwavejs2mqtt.
Note: The Smart Switch 6 has been operational with OZW/zwave2mqtt.
          The Smart Switch 6 reports  power/energy meter updates about every 60 sec.

**To Reproduce**
Steps to reproduce the behavior:
(With "Log to file" set to on)
1. GUI Refresh Info (in my case the Smart Switch 6 is node 6).
The following zwave-js2mqtt logs:

```
14:22:01.691 CNTRLR   [Node 006] Beginning interview - last completed stage: None
14:22:01.692 CNTRLR   [Node 006] new node, doing a full interview...
14:22:01.693 CNTRLR » [Node 006] querying protocol info...
14:22:01.719 CNTRLR « [Node 006] received response for protocol info:
                      basic device class:    Static Controller
                      generic device class:  Binary Switch
                      specific device class: Binary Power Switch
                      is a listening device: true
                      is frequent listening: false
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
                      version:               4
14:22:01.719 CNTRLR   [Node 006] Interview stage completed: ProtocolInfo
14:22:01.720 CNTRLR » [Node 006] pinging the node...
14:22:01.859 CNTRLR « [Node 006] ping successful
14:22:01.860 CNTRLR » [Node 006] querying node info...
14:22:02.165 CNTRLR « [Node 006] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Binary Switch
                      · Multilevel Switch
                      · Color Switch
                      · Configuration
                      · All Switch
                      · Meter
                      · Clock
                      · Association
                      · Association Group Information
                      · Manufacturer Specific
                      · Version
                      · Firmware Update Meta Data
                      · Powerlevel
                      controlled CCs:
                      · Device Reset Locally
                      · Hail
14:22:02.199 CNTRLR   [Node 006] Interview stage completed: NodeInfo
14:22:02.204 CNTRLR   [Node 006] ManufacturerSpecificCC: doing a complete interview...
14:22:02.204 CNTRLR » [Node 006] querying manufacturer information...
14:22:02.509 CNTRLR « [Node 006] received response for manufacturer information:
                        manufacturer: AEON Labs (0x86)
                        product type: 0x0103
                        product id:   0x60
14:22:02.510 CNTRLR   [Node 006] trying to load device config
14:22:02.511 CNTRLR   [Node 006] no device config loaded
[... 61 lines cut ...]
14:22:42.204 CNTRLR   [Node 006] ZWavePlusCC: doing a complete interview...
14:22:42.204 CNTRLR » [Node 006] querying Z-Wave+ information...
14:22:52.353 CNTRLR   [Node 006] Interview attempt 3/5 failed, retrying in 15000 ms...
14:23:07.354 CNTRLR   [Node 006] Beginning interview - last completed stage: NodeInfo
14:23:07.360 CNTRLR   [Node 006] ManufacturerSpecificCC: doing a partial interview...
14:23:07.361 CNTRLR   [Node 006] trying to load device config
14:23:07.362 CNTRLR   [Node 006] no device config loaded
14:23:07.386 CNTRLR   [Node 006] VersionCC: doing a partial interview...
14:23:07.387 CNTRLR   [Node 006] trying to load device config
14:23:07.403 CNTRLR   [Node 006] device config loaded
14:23:07.404 CNTRLR   [Node 006] ZWavePlusCC: doing a complete interview...
14:23:07.405 CNTRLR » [Node 006] querying Z-Wave+ information...
14:23:17.555 CNTRLR   [Node 006] Interview attempt 4/5 failed, retrying in 20000 ms...
14:23:37.556 CNTRLR   [Node 006] Beginning interview - last completed stage: NodeInfo
14:23:37.562 CNTRLR   [Node 006] ManufacturerSpecificCC: doing a partial interview...
14:23:37.564 CNTRLR   [Node 006] trying to load device config
14:23:37.565 CNTRLR   [Node 006] no device config loaded
14:23:37.587 CNTRLR   [Node 006] VersionCC: doing a partial interview...
14:23:37.588 CNTRLR   [Node 006] trying to load device config
14:23:37.599 CNTRLR   [Node 006] device config loaded
14:23:37.600 CNTRLR   [Node 006] ZWavePlusCC: doing a complete interview...
14:23:37.600 CNTRLR » [Node 006] querying Z-Wave+ information...
14:23:47.745 CNTRLR   [Node 006] Failed all interview attempts, giving up.

```

robertsLando, 2021-01-13:
cc @AlCalzone

AlCalzone, 2021-01-13:
That invalid message looks like a security encapsulated command which fails to deserialize. Could be due to incorrect network key, could be something else.
Can you send me the network key on a secure channel and make sure the log is complete or at least contains the previously sent NonceReport?

robertsLando, 2021-01-13:
@tommyjlong Could it be you added the device with a network key and then you changed it?

tommyjlong, 2021-01-13:
I went back to OZW and captured its logs to confirm that my Smart Switch 6 does not use security keys (nonce is not used).  
The other devices that do use encryption, I have not found any issues with them when using zwavejs2mqtt.
However I do NOT see any NonceReports in any the zwavejs logs for any of these secured devices.  
Is there a log level setting I need to set?

@AlCalzone I will send you separate info to help figure out the deserialization issue

robertsLando, 2021-01-13:
> Is there a log level setting I need to set?

@tommyjlong You should set loglevel to `silly` level in zwave settings. Also enable log to file and then send us that file

tommyjlong, 2021-01-13:
I set it to silly, `"logEnabled":true,"logLevel":"silly","logToFile":true`
and did the GUI->refresh Info and here is what came out of the log:
Note: Node 6 is the Aeotec Smart Switch 6. Node 2 is an Aeotec Multisensor6 (encrypted).  You will also see in the log where Node 6 and Node2 are periodically sending updates autonomously.

```
16:18:30.819 CNTRLR   [Node 006] Interview stage completed: ProtocolInfo
16:18:30.819 CNTRLR » [Node 006] pinging the node...
16:18:30.827 SERIAL » 0x01080013060100258e48                                              (10 bytes)
16:18:30.827 DRIVER » [Node 006] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      142
                      └─[NoOperationCC]
16:18:30.831 SERIAL « [ACK]                                                                   (0x06)
16:18:30.838 SERIAL « 0x0104011301e8                                                       (6 bytes)
16:18:30.839 SERIAL » [ACK]                                                                   (0x06)
16:18:30.844 DRIVER « [RES] [SendData]
                        was sent: true
16:18:30.952 SERIAL « 0x010700138e00000c69                                                 (9 bytes)
16:18:30.952 SERIAL » [ACK]                                                                   (0x06)
16:18:30.954 DRIVER « [REQ] [SendData]
                        callback id:     142
                        transmit status: OK
16:18:30.958 CNTRLR « [Node 006] ping successful
16:18:30.959 CNTRLR » [Node 006] querying node info...
16:18:30.965 SERIAL » 0x01040060069d                                                       (6 bytes)
16:18:30.966 DRIVER » [Node 006] [REQ] [RequestNodeInfo]
                        payload: 0x06
16:18:30.968 SERIAL « [ACK]                                                                   (0x06)
16:18:30.996 SERIAL « 0x01040160019b                                                       (6 bytes)
16:18:30.997 SERIAL » [ACK]                                                                   (0x06)
16:18:30.999 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
16:18:31.251 SERIAL « 0x011a00498406140410015e25263370273281855972867a73ef5a82b3          (28 bytes)
16:18:31.255 SERIAL » [ACK]                                                                   (0x06)
16:18:31.258 DRIVER « [Node 006] [REQ] [ApplicationUpdateRequest]
                        payload: 0x8406140410015e25263370273281855972867a73ef5a82
16:18:31.264 CNTRLR   [Node 006] The node is alive.
16:18:31.265 CNTRLR « [Node 006] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Binary Switch
                      · Multilevel Switch
                      · Color Switch
                      · Configuration
                      · All Switch
                      · Meter
                      · Clock
                      · Association
                      · Association Group Information
                      · Manufacturer Specific
                      · Version
                      · Firmware Update Meta Data
                      · Powerlevel
                      controlled CCs:
                      · Device Reset Locally
                      · Hail
16:18:31.292 CNTRLR   [Node 006] Interview stage completed: NodeInfo
16:18:31.295 CNTRLR   [Node 006] ManufacturerSpecificCC: doing a complete interview...
16:18:31.295 CNTRLR » [Node 006] querying manufacturer information...
16:18:31.315 SERIAL » 0x0109001306027204258f3d                                            (11 bytes)
16:18:31.316 DRIVER » [Node 006] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      143
                      └─[ManufacturerSpecificCCGet]
16:18:31.322 SERIAL « [ACK]                                                                   (0x06)
16:18:31.325 SERIAL « 0x0104011301e8                                                       (6 bytes)
16:18:31.325 SERIAL » [ACK]                                                                   (0x06)
16:18:31.327 DRIVER « [RES] [SendData]
                        was sent: true
16:18:31.482 SERIAL « 0x010700138f00001074                                                 (9 bytes)
[... 906 lines cut ...]
16:20:55.295 CNTRLR   [Node 006] [~] [Meter] value[65537]: 44.745 => 44.745             [Endpoint 0]
16:20:55.297 CNTRLR   [Node 006] [~] [Meter] previousValue[65537]: 44.745 => 44.745     [Endpoint 0]
16:20:55.298 CNTRLR   [Node 006] [~] [Meter] deltaTime[65537]: 64 => 56                 [Endpoint 0]
16:20:55.299 SERIAL » [ACK]                                                                   (0x06)
16:20:55.302 DRIVER « [Node 006] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:        Electric
                          scale:       kWh
                          rate type:   Consumed
                          value:       44.745
                          time delta:  56 seconds
                          prev. value: 44.745
16:20:55.395 SERIAL « 0x0118000400061232022174000000000000000000000001050096              (26 bytes)
16:20:55.397 CNTRLR   [Node 006] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
16:20:55.401 CNTRLR   [Node 006] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
16:20:55.403 SERIAL » [ACK]                                                                   (0x06)
16:20:55.406 DRIVER « [Node 006] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      0
                          time delta: 0 seconds

```

AlCalzone, 2021-01-13:
Node 6 claims it supports Z-Wave+ CC, but does not respond when we query that CC's info.

blhoward2, 2021-01-14:
If these are both smart switch 6s, I’m not sure how that’s true as they’re all zwave plus.

closed 2021-01-27 by AlCalzone

tommyjlong, 2021-01-29:
I would like to try out this fix.  I have a question on installation with zwavejsmqtt using npm (install, run build).  I just installed zwavejs2mqtt tagged version 1.0.1 and it appears that it installs the zwavejs tagged v6.1.0 driver which does not contain this fix.  Is there a way I can use npm to get zwavejs2mqtt to install the zwavejs driver from main (with the fix) instead of the tagged v6.1.0 branch?  Sorry I am not familiar with nodejs very much.

AlCalzone, 2021-01-29:
If you go into the zwavejs2mqtt repo and open the docs there you should find a description.
I'm on mobile right now, can't look it up

tommyjlong, 2021-01-29:
Thanks.  I've looked all through the docs.  Closest I found was docs on using a custom docker build but couldn't find anything on custom npm install. If you come across something and can provide a pointer, it would be much appreciated.

AlCalzone, 2021-01-29:
Oh, npm... https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github

tommyjlong, 2021-01-31:
It took a while to get the install procedure worked out, and I am happy to say, that node-zwavejs v6.1.1 solved this problem and my Smartswitch 6 completed the interview process!! Thanks :)

Madelaide, 2021-03-04:
Is this fix in the latest (zw js to mqtt 0.6.0 addon and ha 21.3.0) release? It does not seem to be working for me.

I have two SS6 which do no interview; the ones that were already on the zstick did interview and work fine.

AlCalzone, 2021-03-04:
Please open a new issue and fill in the requested info. That log snippet is not nearly enough to say anything.

Madelaide, 2021-03-04:
Hi
Yes okay, i thought it was still this ticket as it is the same issue, but that is fine; I won't have access to the system for a few days but will raise it when I return and I'll turn on debug logs vs info (or is TRACE required?)

PS: i finally found the zw version in the integration, if it was fixed in 6.1.1 has it regressed?

Network Connected
ws://a0d7b954-zwavejs2mqtt:3000
Driver Version: 6.6.0
Server Version: 1.1.0

AlCalzone, 2021-03-04:
> has it regressed?

🤷🏻‍♂️ I really need to see the detailed log. `dead` could just mean you have a bad connection.

Madelaide, 2021-03-04:
I understand, I'll try and provide it as soon as I can. The aeotec smart switch 6 are the worst RF devices in my network, see very few neighbours . They see 1-4 where an immediately adjacent fibaro, or even older aeotec smart switch 2 will see 20-30 neighbours.

Madelaide, 2021-03-04:
[zwavejs_357.log](https://github.com/zwave-js/node-zwave-js/files/6087219/zwavejs_357.log)

Okay, I managed to remote in and run an interview - node 135 is the new node, today at 10 am you can see it fail - interestingly if you search for Node 135, yesterday at 150734 node 135 did a partial interview but seemed to fail - nothing in the UI, and also 136.

Node 11 is another switch, but already in the node list, at the risk of losing it I reinterview, and it completed okay. hopefully this log points out the issue(s)

Madelaide, 2021-03-05:
PS: If you need this in a different ticket I am out of time today but can try move/recreate it over the weekend

AlCalzone, 2021-03-05:
As I suspected: The affected nodes do not acknowledge the commands we send them. That usually indicates a connectivity problem. I can see NoAck a few times for nodes 100, 112, 135, 136 in your log.

Madelaide, 2021-03-05:
For reference, node 11 is 15m away through two double-brick walls and a stud wall, node 135/6 are 4m and 3m, respectively, LOS. I can try moving them within 1m next week as I have gone away for the long-weekend local holiday (mini break).

AlCalzone, 2021-03-05:
Try doing a network heal. I'm lately seeing many issues that can be explained by bad routes.

Madelaide, 2021-03-06:
Thankyou. Will try after a stick backup. Last heal I did killed the network and had to revert to a 2 month old archive.

AlCalzone, 2021-03-06:
How's that even possible? You can't change the routes on nodes with a backup.

blhoward2, 2021-03-06:
Maybe poor connectivity and they restored an old cache?

tommyjlong, 2021-03-06:
I can add, that my Smart Switch 6 from a networking/RF point of view was both bad and good.  I initially wanted to use in on the 1st floor of my house, but the range was very bad (less than 7m through 1 wall).  I moved it to my 2nd floor much further away through more walls and it worked very well.  Fortunate for me it was where my washing machine was located, so I use it strictly as an energy meter.  I have not had any issues since this problem was fixed, but I am still on 1.0.1/6.1.1.  I will try and upgrade in the next day or two just to confirm it still works.

Madelaide, 2021-03-07:
> How's that even possible? You can't change the routes on nodes with a backup.

The stick became unresponsive in ha after an attempted heal. It was effectively dead in ha -even after snapshot rollback. 

So I zensys restored the stick. I didn’t care about the routes. It worked and could talk to it with ha again. So I always backup the switch before trying to heal.

Madelaide, 2021-03-07:
> If they mean a backup of their docker or something maybe poor connectivity and they restored an old cache, thereby requiring less interviewing? I agree a backup of the stick would do little.

890 MHz over the house is very good but added two repeaters anyway. 

I did purchases a USB loop back tester made in Australia and it found errors on the USB extension I was using. Annoying as it was expensive. But using a test passing cable now.

Madelaide, 2021-03-11:
PS: I moved the controller within 10 cm of the node in question (135) and it interviewed okay.

AlCalzone, 2021-03-11:
That confirms my initial suspicion that the node does not have correct routes. The only way to get that corrected is to heal it while the controller is at its usual location.

Another workaround (assuming that it can even be included where it is) would be:
* stop zwave-js
* factory reset the node
* start zwave-js
* replace the node with itself (replace failed node)

Madelaide, 2021-03-11:
I would love to heal them, but the command has no effect on the nodes. The neighbour list remains unchanged (and I've tried single healing 4 other different types of devices with neighbours). Maybe I need to raise an issue for that...

AlCalzone, 2021-03-11:
Then maybe the devices really don't see any other neighbors. You (I) should be able to see in a log what happens exactly.

cdemi, 2021-04-22:
I see this issue as closed so I am not sure if I am doing something wrong, but I am experiencing the same issue with Aeotec Smart Switch 6. Exactly after inclusion in the interview stage, the node is stuck in "NodeInfo":

[image]

AlCalzone, 2021-04-22:
Got a log?

cdemi, 2021-04-22:
Sorry I'm pretty new to HomeAssistant/Z-Wave JS since I am moving from SmartThings. I will figure out how to get the log file and attach it soon

AlCalzone, 2021-04-22:
If you open a new issue, you should find some hints in the issue template. It looks like you're using zwavejs2mqtt, so https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs could help you aswell.
Please make sure to post the correct log - I need the `zwave-js` logfile.

PR #1505 (2021-01-27): feat: allow and handle response timeouts for GET-type CC commands
This PR paves the way for a more permissive handling of missing node responses. Based on the discussion in https://github.com/zwave-js/node-zwave-js/discussions/1415, we **have** to expect a timeout for every GET-type CC command and we **have** to handle it in a way that a missing response does not break everything (e.g. abort and restart interviews for uncritical queries).

As a side-effect the code that previously used `ignoreTimeout` is more readable now.

closes: #1305
fixes: #1369