zwave-js/zwave-js #1370: [bug] Building36 T10 (ADC T-2000) thermostat fails to load climate entity for Home Assistant

bachya, 2021-01-13:
**Version**

Build/Run method
- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: 1.0.0-alpha.2

**Describe the bug**
I am attempting to get my [Building 36 T10 thermostat](https://products.z-wavealliance.org/products/1161) working with Home Assistant and am unable to see the `climate` entity as a device within z2m.

First, I gather from the docs that devices in the `climate` Home Assistant domain need to have a configuration provided. I wrote this configuration:

```json
{
 "400-1-1": [
  {
   "type": "climate",
   "object_id": "thermostat",
   "values": [
    "49-1-Air temperature",
    "64-0-mode",
    "66-0-state",
    "67-1-setpoint-1",
    "67-1-setpoint-2",
    "68-0-mode"
   ],
   "mode_map": {
    "off": 0,
    "heat": 1,
    "cool": 2,
    "auto": 3
   },
   "fan_mode_map": {
    "auto_low": 0,
    "on_low": 1,
    "circulate": 6
   },
   "setpoint_topic": {
    "heat": "67-1-setpoint-1",
    "cool": "67-1-setpoint-2"
   },
   "default_setpoint": "67-1-setpoint-1",
   "discovery_payload": {
    "min_temp": 50,
    "max_temp": 85,
    "modes": [
     "off",
     "heat",
     "cool",
     "auto"
    ],
    "fan_modes": [
     "auto_low",
     "on_low",
     "circulate"
    ],
    "action_topic": "66-0-state",
    "action_template": "{{ value_json.value | lower }}",
    "current_temperature_topic": "49-1-Air temperature",
    "current_temperature_template": "{{ value_json.value }}",
    "fan_mode_state_topic": "68-0-mode",
    "fan_mode_command_topic": true,
    "mode_state_topic": "64-0-mode",
    "mode_command_topic": true,
    "temperature_state_template": "{{ value_json.value }}",
    "temperature_command_topic": true
   }
  }
 ]
}
```

...and stored it in `store/customDevices.json`. I didn't find those docs especially clear, but I tried to follow the format and used the data from [the OZW database](https://github.com/OpenZWave/open-zwave/blob/master/config/building36/b36-t10.xml) (which I also wrote).

I am able to see the node when `zwavejs2mqtt` loads:

[image]

I am able to see four `binary_sensor`/`sensor` devices:

[image]

...but the `climate` device is missing. No amount of refreshing the node, rediscovering the devices, etc. changes that.

**Additional context**
I do see this error in the Docker console logs (which are the gateway logs, I assume):

```
2021-01-07 23:55:10.417 INFO ZWAVE: Node 2: value updated: 128-0-level 0 => 0
2021-01-07 23:55:10.418 INFO ZWAVE: Node 2: value updated: 128-0-isLow false => false
2021-01-07 23:55:10.426 INFO ZWAVE: Node 2 is alive
2021-01-07 23:55:10.492 INFO ZWAVE: Node 2: value added 2-49-0-Air temperature => 20.62
2021-01-07 23:55:10.493 INFO ZWAVE: Node 2: value added 2-49-0-Humidity => 29
2021-01-07 23:55:10.494 INFO ZWAVE: Node 2: value added 2-64-0-mode => 1
2021-01-07 23:55:10.494 INFO MQTT: Subscribing to zwavejs2mqtt/+/+/+/+/set
2021-01-07 23:55:10.495 INFO ZWAVE: Node 2: value added 2-64-0-manufacturerData => undefined
2021-01-07 23:55:10.495 INFO ZWAVE: Node 2: value added 2-66-0-state => 8
2021-01-07 23:55:10.496 INFO ZWAVE: Node 2: value added 2-112-0-1 => 0
2021-01-07 23:55:10.496 INFO ZWAVE: Node 2: value added 2-112-0-2 => 2
2021-01-07 23:55:10.496 INFO ZWAVE: Node 2: value added 2-112-0-3 => 2
2021-01-07 23:55:10.497 INFO ZWAVE: Node 2: value added 2-112-0-4 => 1
2021-01-07 23:55:10.497 INFO ZWAVE: Node 2: value added 2-112-0-5 => 704643072
2021-01-07 23:55:10.497 INFO ZWAVE: Node 2: value added 2-112-0-6 => 704644352
2021-01-07 23:55:10.498 INFO ZWAVE: Node 2: value added 2-112-0-7 => 704643072
2021-01-07 23:55:10.498 INFO ZWAVE: Node 2: value added 2-112-0-8 => 10
2021-01-07 23:55:10.498 INFO ZWAVE: Node 2: value added 2-112-0-9 => 10
2021-01-07 23:55:10.499 INFO ZWAVE: Node 2: value added 2-112-0-10 => 704732672
2021-01-07 23:55:10.499 INFO ZWAVE: Node 2: value added 2-112-0-11 => 0
2021-01-07 23:55:10.499 INFO ZWAVE: Node 2: value added 2-112-0-12 => 20
2021-01-07 23:55:10.500 INFO ZWAVE: Node 2: value added 2-112-0-13 => 25
2021-01-07 23:55:10.500 INFO ZWAVE: Node 2: value added 2-112-0-14 => 60
2021-01-07 23:55:10.500 INFO ZWAVE: Node 2: value added 2-112-0-15 => 704886272
2021-01-07 23:55:10.500 INFO ZWAVE: Node 2: value added 2-112-0-16 => 704732672
2021-01-07 23:55:10.501 INFO ZWAVE: Node 2: value added 2-112-0-17 => 704886272
2021-01-07 23:55:10.501 INFO ZWAVE: Node 2: value added 2-112-0-18 => 704771072
2021-01-07 23:55:10.501 INFO ZWAVE: Node 2: value added 2-112-0-19 => 0
2021-01-07 23:55:10.502 INFO ZWAVE: Node 2: value added 2-112-0-20 => 5
2021-01-07 23:55:10.502 INFO ZWAVE: Node 2: value added 2-112-0-21 => 10
2021-01-07 23:55:10.502 INFO ZWAVE: Node 2: value added 2-112-0-22 => 25
2021-01-07 23:55:10.502 INFO ZWAVE: Node 2: value added 2-112-0-23 => 1
2021-01-07 23:55:10.503 INFO ZWAVE: Node 2: value added 2-112-0-24 => 15
2021-01-07 23:55:10.503 INFO ZWAVE: Node 2: value added 2-112-0-25 => 0
2021-01-07 23:55:10.503 INFO ZWAVE: Node 2: value added 2-112-0-26 => 1
2021-01-07 23:55:10.504 INFO ZWAVE: Node 2: value added 2-112-0-27 => 30
2021-01-07 23:55:10.504 INFO ZWAVE: Node 2: value added 2-112-0-28 => 15
2021-01-07 23:55:10.504 INFO ZWAVE: Node 2: value added 2-112-0-30 => 0
2021-01-07 23:55:10.505 INFO ZWAVE: Node 2: value added 2-112-0-31 => 704650752
2021-01-07 23:55:10.505 INFO ZWAVE: Node 2: value added 2-112-0-32 => 704650752
2021-01-07 23:55:10.505 INFO ZWAVE: Node 2: value added 2-112-0-33 => 704645632
2021-01-07 23:55:10.505 INFO ZWAVE: Node 2: value added 2-112-0-34 => 1
2021-01-07 23:55:10.506 INFO ZWAVE: Node 2: value added 2-112-0-35 => 0
2021-01-07 23:55:10.506 INFO ZWAVE: Node 2: value added 2-114-0-manufacturerId => 400
2021-01-07 23:55:10.506 INFO ZWAVE: Node 2: value added 2-114-0-productType => 1
2021-01-07 23:55:10.506 INFO ZWAVE: Node 2: value added 2-114-0-productId => 1
2021-01-07 23:55:10.507 INFO ZWAVE: Node 2: value added 2-128-0-level => 0
2021-01-07 23:55:10.507 INFO ZWAVE: Node 2: value added 2-128-0-isLow => false
2021-01-07 23:55:10.507 INFO ZWAVE: Node 2: value added 2-134-0-libraryType => 3
2021-01-07 23:55:10.508 INFO ZWAVE: Node 2: value added 2-134-0-protocolVersion => 4.54
2021-01-07 23:55:10.508 INFO ZWAVE: Node 2: value added 2-134-0-firmwareVersions => 1.20
2021-01-07 23:55:10.508 INFO ZWAVE: Node 2: value added 2-134-0-hardwareVersion => undefined
2021-01-07 23:55:10.511 ERROR GATEWAY: Error while discovering device climate_thermostat of node 2: Cannot read property 'id' of undefined Cannot read property 'id' of undefined
TypeError: Cannot read property 'id' of undefined
    at Gateway.valueTopic (/usr/src/app/lib/Gateway.js:868:23)
    at Gateway.discoverDevice (/usr/src/app/lib/Gateway.js:1126:16)
    at /usr/src/app/lib/Gateway.js:360:40
    at Array.forEach (<anonymous>)
    at Gateway.onNodeStatus (/usr/src/app/lib/Gateway.js:360:17)
    at ZwaveClient.emit (events.js:314:20)
    at ZwaveClient.EventEmitter.emit (domain.js:483:12)
    at ZwaveClient.onNodeStatus (/usr/src/app/lib/ZwaveClient.js:255:10)
    at ZwaveClient.onNodeReady (/usr/src/app/lib/ZwaveClient.js:283:16)
    at ZWaveNode.emit (events.js:326:22)
    at ZWaveNode.EventEmitter.emit (domain.js:483:12)
    at ZWaveNode.onReadyChange (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:441:19)
    at /usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:230:10
    at Interpreter.update (/usr/src/app/node_modules/xstate/lib/interpreter.js:253:17)
    at /usr/src/app/node_modules/xstate/lib/interpreter.js:125:23
    at Scheduler.process (/usr/src/app/node_modules/xstate/lib/scheduler.js:60:13)
2021-01-07 23:55:10.514 INFO ZWAVE: Node 2 ready: Building 36 Technologies - ADC-T 2000 (Alarm.com Smart Thermostat)
```

Here are (a) the exported Node 2 configuration and (b) the `zwave-js` logs: https://gist.github.com/bachya/b2eb36f3be24878c0c78e5494f34977c

From the `zwave-js` logs, it appears that the driver receives climate mode data from the node:

```
23:55:24.843 CNTRLR   [Node 002] [~] [Thermostat Mode] mode: 1 => 1                     [Endpoint 0]
23:55:24.846 SERIAL » [ACK]                                                                   (0x06)
23:55:24.849 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[ThermostatModeCCReport]
                          mode: Heat
23:55:24.855 SERIAL » 0x0109001305029840253e21                                            (11 bytes)
23:55:24.856 DRIVER » [Node 005] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      62
                      └─[SecurityCCNonceGet]
23:55:24.856 CNTRLR « [Node 002] received current thermostat mode: Heat
23:55:24.856 CNTRLR   [Node 002] [~] [Thermostat Mode] interviewComplete: tr [Endpoint 0] [internal]
```

AlCalzone, 2021-01-13:
I'll need to see a log from zwave-js with a complete interview (after refreshNode)

bachya, 2021-01-13:
@AlCalzone Here is a complete, "silly"-level zwave-js log (complete startup and interview). Reminder that Node 2 is the one we're interested.

https://gist.github.com/b74b19be49582b83dbec4332fb4af173

bachya, 2021-01-13:
Thanks for this info, @AlCalzone. Probably too early to tell, but does this indicate a permanent incompatibility that will prevent zwave-js from supporting this device? Or more debugging to do?

As always, I'm at your disposal – let me know how I can help.

AlCalzone, 2021-01-13:
I'll need to figure out what to do with this. Probably just skip the part with the bugged responses.

robertsLando, 2021-01-13:
@AlCalzone So does an error in the interview stop all the interview?

AlCalzone, 2021-01-13:
Not if I ignore the timeout. That seems to happen in this case but I am not sure why that causes the problems you're seeing

AlCalzone, 2021-01-27:
@bachya https://github.com/zwave-js/node-zwave-js/pull/1505 should stop the interview from aborting.
The invalid command still won't be parsed, but that isn't critical at least.

bachya, 2021-01-27:
Thanks for the update, @AlCalzone. I've installed that PR into the `dev` z2m container and am starting fresh; will report back shortly.

bachya, 2021-01-27:
Alright, after a complete startup and interview, z2m shows the `climate` Home Assistant device:

[image]

...with this device JSON:

```json
{
  "type": "climate",
  "object_id": "climate",
  "discovery_payload": {
    "min_temp": 5,
    "max_temp": 40,
    "temp_step": 0.5,
    "modes": [
      "off",
      "heat",
      "cool",
      "auto"
    ],
    "mode_state_topic": "zwavejs2mqtt/nodeID_2/64/0/mode",
    "mode_state_template": "{{ {0: \"off\",1: \"heat\",2: \"cool\",3: \"auto\"}[value_json.value] | default('off') }}",
    "mode_command_topic": "zwavejs2mqtt/nodeID_2/64/0/mode/set",
    "current_temperature_topic": "zwavejs2mqtt/nodeID_2/49/0/Air_temperature",
    "current_temperature_template": "{{ value_json.value }}",
    "temperature_state_topic": "zwavejs2mqtt/nodeID_2/67/0/setpoint/1",
    "temperature_state_template": "{{ value_json.value }}",
    "temperature_command_topic": "zwavejs2mqtt/nodeID_2/67/0/setpoint/1/set",
    "temperature_unit": "C",
    "precision": 0.1,
    "device": {
      "identifiers": [
        "zwavejs2mqtt_0xc8341d86_node2"
      ],
      "manufacturer": "Building 36 Technologies",
      "model": "Alarm.com Smart Thermostat (ADC-T 2000)",
      "name": "nodeID_2",
      "sw_version": "1.20"
    },
    "name": "nodeID_2_climate",
    "unique_id": "zwavejs2mqtt_0xc8341d86_Node2_climate"
  },
  "values": [
    "49-0-Air temperature",
    "64-0-mode",
    "67-0-setpoint-1",
    "67-0-setpoint-2"
  ],
  "mode_map": {
    "off": 0,
    "heat": 1,
    "cool": 2,
    "auto": 3
  },
  "setpoint_topic": {
    "1": "67-0-setpoint-1",
    "2": "67-0-setpoint-2"
  },
  "default_setpoint": "67-0-setpoint-1",
  "discoveryTopic": "climate/nodeID_2/climate/config",
  "persistent": false,
  "ignoreDiscovery": false,
  "id": "climate_climate"
}
```

When I preview the entity in Home Assistant, I notice a couple of other oddities – not sure if they are related to zwave-js or z2m, so feel free to punt me elsewhere if needed.

1. My Home Assistant climate entity doesn't give me the option to set my fan mode, which OZW did previously.
2. The current temperature of the entity is 158°F, which obviously isn't correct. I have the thermostat set to 70°F, but from the device JSON above, it appears that the default unit is C? 158°F does, indeed, equal 70°C, so something is backward there.

Latest logs for both zwave-js and z2m: https://gist.github.com/bachya/88576de4eaf119232f447c69de956d49

robertsLando, 2021-01-27:
> My Home Assistant climate entity doesn't give me the option to set my fan mode

Because it's not supported yet in discovery

AlCalzone, 2021-01-27:
Okay so, everything works on the zwave-js side?

bachya, 2021-01-27:
@AlCalzone Here's the zwave-js log: https://gist.githubusercontent.com/bachya/88576de4eaf119232f447c69de956d49/raw/e6ebf60fa00134eaacf856a2e1a8d4da419c934b/zwavejs_1.log

AlCalzone, 2021-01-27:
```
15:29:09.273 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         70 °F
...
15:29:27.450 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Cooling
                          value:         77 °F
```
Looks like there's some conversion happening outside zwave-js.

bachya, 2021-01-27:
That first entry is correct, in that at the time, my thermostat was set to 70°F. @robertsLando Is it possible that the strange representation of 158°F in Home Assistant likes with z2m, as well?

AlCalzone, 2021-01-27:
BTW: That's two differeent setpoints, the second one is for the "Cooling" capability. (a bit weird though that it is higher than "Heating").