zwave-js/zwave-js #1371: Correctly parse `Compatibility` flags when importing OZW config file

robertsLando, 2021-01-13:
Check for example https://github.com/OpenZWave/open-zwave/blob/master/config/building36/b36-t10.xml

Ref #1370

AlCalzone, 2021-01-13:
We need to figure out what they mean and how they translate to our format.

robertsLando, 2021-01-13:
@AlCalzone IMO it mean you have to override/add those CC using that props instead of the one reported from the device

AlCalzone, 2021-01-13:
so roughly it translates to `compat.commandClasses.add`. Does this only affect the root endpoint or individual endpoints too?

robertsLando, 2021-01-13:
It depends if there is the `<Instance>` specified (instance is the endpoint).

```xml
<Instance index="1" />
```

AlCalzone, 2021-01-13:
Guess we could try converting that to out compat options.

Accepting PR 😀

robertsLando, 2021-01-13:
Will look at that this afternoon if I find some time :)

blhoward2, 2021-01-14:
https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags

AlCalzone, 2021-01-14:
Oh nice. That also means that the linked file above doesn't actually use any compat flags.

robertsLando, 2021-01-14:
> Oh nice. That also means that the linked file above doesn't actually use any compat flags.

Yep

blhoward2, 2021-01-14:
Is that an ozw 1.4 or 1.6 config file? I don’t recall seeing explicit mappings like that in other 1.6 files, though maybe I’m wrong.

closed 2021-02-23 by AlCalzone