zwave-js/zwave-js #1376: Add property to Driver to return driver state

cgarwood, 2021-01-13:
**Is your feature request related to a problem? Please describe.**
If you fire up zwave-js with zwave-js-server but have an application connect to the zwave-js-server after the "driver ready" and/or "all nodes ready" events have fired, there's not a good way for the application to know the state of the network since it missed those events.

**Describe the solution you'd like**
My recommendation is for an additional property on the Driver class that returns the state of the network. ZWave-JS core could update that property when it emits the event. That way if an application fires up and misses one of the events it can still easily know what state the network is in.

AlCalzone, 2021-01-13:
Off the top of my head:
- driver ready?
- which nodes ready?
- all nodes ready?
- interview for which nodes completed?

cgarwood, 2021-01-13:
Important states I can think of from an application standpoint:
- is the driver ready
- are all nodes queried or are we still querying

AlCalzone, 2021-01-14:
@cgarwood I've opened PR #1396 to add two properties (driver ready, all nodes ready). I think this should be enough for your use case.
You should be able to check whether all queries are done using
```ts
const allInterviewsCompleted = [...driver.controller.nodes.values()].every(n => n.interviewStage === InterviewStage.Complete);
```

PR #1396 (2021-01-16): feat(driver): add ready and allNodesReady indicators
This PR adds two properties to the driver class which indicate whether
a) the driver "ready" event has been emitted (`driver.ready`)
b) the "all nodes ready" event has been emitted (`driver.allNodesReady`)

fixes: #1376
closed 2021-01-16 by AlCalzone