zwave-js/zwave-js #1389: propertyKeyName seems not to be exposed for Meter CC

AlCalzone, 2021-01-14:
... and maybe others.

AlCalzone, 2021-01-14:
@robertsLando FYI, people seem to be getting topics like `50/1/value/65537` instead of `50/1/value/electric_kWh_consumed` because of this

bdbogjoe, 2021-01-14:
let me know if you need any testing

robertsLando, 2021-01-14:
> @robertsLando FYI, people seem to be getting topics like 50/1/value/65537 instead of 50/1/value/electric_kWh_consumed because of this

Oh that's bad :cry:

AlCalzone, 2021-01-14:
[image]

I can't reproduce. Two different tests, both claim that the propertyKeyName is there. Any chance that property gets lost somewhere in zwavejs2mqtt?

AlCalzone, 2021-01-14:
Might be because of https://github.com/zwave-js/zwavejs2mqtt/issues/239

closed 2021-01-14 by AlCalzone