zwave-js/zwave-js #1400: MCOHome bad config.json integration for generation 5 devices

S1M8N, 2021-01-14:
** Is your feature request related to a problem?**
"Yes"
A clear and concise description of the nature of the problem :

To start, sorry for my English...

I'm starting with the zwavejs2mqtt and I noticed some problem with the configuration files. I will explain to you in as much detail as possible what I would not qualify as a "bug" but more as a code error.

**Introduction:** 
I have around 40 z-wave modules, 35 of which are MCOHome brand devices. The problem is found in the configuration files of the MCOhome devices. I have 4 types of switches (I quote you the label of these, here: MH-s411 (1 button light switch), MH-s412 (2 button light switch), MH-s314 (4 light switch buttons), MH-c421 (shutter) and MH-dt411 (dimmer)). All the devices mentioned are generation 5 (productID = 0x5102) and I provide a complete configuration file in .xml format for all the devices.

**Details about light switches:**
For the Light Switches, there is one thing that (I guess) you didn't pay attention to. This is the generation. There have been several generations of these modules. In your configuration files, you do not differentiate the two ... This is a shame because generation 5 (0x5102) does not have the same "information parameters" as version 4 (0x4102). Indeed, version 4 only had as parameter "Saving state before power failure". However, the new generations have many more parameters (like changing the intensity of the led). I think it would be nice to make these changes in your configuration files for me and for the next people who will be using the new generation 5 home MCO models.

**Details on the shutter panel:**
My MH-c421 device is first not added in the configuration files but also has problems when adding it, it is not recognized as a shutter but as an ON / OFF light switch. It is impossible for me to stop my awning or to raise / lower it as I wish ... I have as a simple action "binary switch"
Link img for help you: https://ibb.co/7tbmb1R, https://ibb.co/G7VZybs

**Dimmer Details:**
For the dimmer switch, I did not do any test with the zwavejs2mqtt. I only provide the configuration file in .xml format.

I will attach my .xml configuration documents for all the modules mentioned here and you can see the differences with the information entered in your configuration files. In addition I have the manufacturer_specific.xml file with the productType and the productId of all the devices mentioned above.

You can without further look on the mcohome.com site to download the PDF configuration documents for the devices mentioned above because mine may not be completely complete.

I sincerely hope that you can do something to improve the various points mentioned above with my help.

PS: I think I noticed a problem with the mh-s521.json configuration file. It does not exist on the mcohome site .... Moreover this reference is not that of a shutter and in the file we find light configurations with a label for shutter panels. This is very special ...

Thank you for taking the time for this request, I would be very grateful. If you need help, don't hesitate. If you are missing any files, let me know.

Simon

[Devices gen 5.zip](https://github.com/zwave-js/node-zwave-js/files/5817239/Devices.gen.5.zip)

closed 2021-01-14 by S1M8N

AlCalzone, 2021-01-14:
Why close?

S1M8N, 2021-01-14:
I have generated two Issues… De : AlCalzoneEnvoyé le :jeudi, 14 janvier 2021 23:28À : zwave-js/node-zwave-jsCc : S1M8N; State changeObjet :Re: [zwave-js/node-zwave-js] MCOHome bad config.json integration for generation 5 devices (#1400) Why close?