zwave-js/zwave-js #1401: MCOHome bad config.json integration for generation 5 devices

S1M8N, 2021-01-14:
** Is your feature request related to a problem?**
"Yes"
A clear and concise description of the nature of the problem :

To start, sorry for my English...

I'm starting with the zwavejs2mqtt and I noticed some problem with the configuration files. I will explain to you in as much detail as possible what I would not qualify as a "bug" but more as a code error.

**Introduction:** 
I have around 40 z-wave modules, 35 of which are MCOHome brand devices. The problem is found in the configuration files of the MCOhome devices. I have 4 types of switches (I quote you the label of these, here: MH-s411 (1 button light switch), MH-s412 (2 button light switch), MH-s314 (4 light switch buttons), MH-c421 (shutter) and MH-dt411 (dimmer)). All the devices mentioned are generation 5 (productID = 0x5102) and I provide a complete configuration file in .xml format for all the devices.

**Details about light switches:**
For the Light Switches, there is one thing that (I guess) you didn't pay attention to. This is the generation. There have been several generations of these modules. In your configuration files, you do not differentiate the two ... This is a shame because generation 5 (0x5102) does not have the same "information parameters" as version 4 (0x4102). Indeed, version 4 only had as parameter "Saving state before power failure". However, the new generations have many more parameters (like changing the intensity of the led). I think it would be nice to make these changes in your configuration files for me and for the next people who will be using the new generation 5 home MCO models.

**Details on the shutter panel:**
My MH-c421 device is first not added in the configuration files but also has problems when adding it, it is not recognized as a shutter but as an ON / OFF light switch. It is impossible for me to stop my awning or to raise / lower it as I wish ... I have as a simple action "binary switch"
Link img for help you: https://ibb.co/7tbmb1R, https://ibb.co/G7VZybs

**Dimmer Details:**
For the dimmer switch, I did not do any test with the zwavejs2mqtt. I only provide the configuration file in .xml format.

I will attach my .xml configuration documents for all the modules mentioned here and you can see the differences with the information entered in your configuration files. In addition I have the manufacturer_specific.xml file with the productType and the productId of all the devices mentioned above.

You can without further look on the mcohome.com site to download the PDF configuration documents for the devices mentioned above because mine may not be completely complete.

I sincerely hope that you can do something to improve the various points mentioned above with my help.

PS: I think I noticed a problem with the mh-s521.json configuration file. It does not exist on the mcohome site .... Moreover this reference is not that of a shutter and in the file we find light configurations with a label for shutter panels. This is very special ...

Thank you for taking the time for this request, I would be very grateful. If you need help, don't hesitate. If you are missing any files, let me know.

Simon

[Devices gen 5.zip](https://github.com/zwave-js/node-zwave-js/files/5817286/Devices.gen.5.zip)

AlCalzone, 2021-01-14:
Thanks for the detailed report. Many of our config files were imported from various sources and we can't test them ourselves. So reports like yours help a lot.

To distinguish between generation 5 and older devices, we should split the configuration files. Would you be willing to help with that? You can find a detailed description of our config format [here](https://zwave-js.github.io/node-zwave-js/#/development/config-files).
We also have a converter script, but I don't think it can work with local xml files yet. @robertsLando, might be worth an addition?

For the `MH-c421` I would need to see a zwave-js logfile. You should be able to enable that somewhere in the zwavejs2mqtt options, but I'm not sure where. When you create that log, please do a fresh interview ("refresh info").

robertsLando, 2021-01-15:
> We also have a converter script, but I don't think it can work with local xml files yet. @robertsLando, might be worth an addition?

We could add this support too, yep :) or, for now, it could manually place it in tmp dir (this also require to update manufacturers.xml file).

To enable logs just open the ui, go to settings, there are 2 logs settings one for the application and it's on general section and one for zwavejs that is on zwave section.

S1M8N, 2021-01-15:
> Thanks for the detailed report. Many of our config files were imported from various sources and we can't test them ourselves. So reports like yours help a lot.
>
> To distinguish between generation 5 [...]

--------------------------------------------------------------------------------------------------------------------------------------------

Hello @AlCalzone,

Thank you for your quick reply. I'll start by talking about configuration files and come back to the topic of mh-c421.

**Configuration.json file:**
I was able to make progress today on the configuration of my 3 light switches and my shutter switch. Taking the example of the files already implemented, I could see that there was a lot of imperfection in the consistency of the files. I still tried to do my best. it would still be good to check my configuration files to be sure the data is correct. They are attached to this message.

I have not yet made the configuration file for the dimmer switch but once that will be done I will send you a message in this post.

**MH-c421:**
For this one, I was able to activate the log options you told me and I had access to the following information: Image link : https://ibb.co/f9FbX1g 
If that's not enough, don't hesitate to point it out to me.

If you need anything else I am available. Hope my files will do the job;).

Thank you again for your kindness and responses, hope to find a solution as soon as possible.

Simon

AlCalzone, 2021-01-15:
I'm not sure how proficient you are with GitHub. The easiest way for us to include the updated config files would be a pull request. We could also discuss possible changes there.

If not, let me know and I'll look at the files offline.

AlCalzone, 2021-01-15:
The log is not what I had in mind, sry. @robertsLando can you help out again please?

S1M8N, 2021-01-15:
> I'm not sure how proficient you are with GitHub. The easiest way for us to include the updated config files would be a pull request. We could also discuss possible changes there.
> 
> If not, let me know and I'll look at the files offline.

I don't know github at all in the art. If there is anything I need to do, let me know.

> The log is not what I had in mind, sry. @robertsLando can you help out again please?

Ok, I do not see either ... If you have additional information...

robertsLando, 2021-01-15:
I think there is no easy way to export logs apart from making them exportable from the UI. 

If he is using docker he could try to use the `docker logs <containerId>` command

S1M8N, 2021-01-15:
That it's ok ? :
[Log zwavejs2mqtt file.zip](https://github.com/zwave-js/node-zwave-js/files/5823061/Log.zwavejs2mqtt.file.zip)

AlCalzone, 2021-01-15:
Sorry, no. Your log contains this:
```
Logging to file:
/usr/src/app/bin/zwave-1.log
```
I need that file

S1M8N, 2021-01-15:
Ok thank normaly ist that : 
[zwave-1 log.zip](https://github.com/zwave-js/node-zwave-js/files/5823225/zwave-1.log.zip)

AlCalzone, 2021-01-15:
So, I took a look at your c421 log. First the device tells us that it supports these CCs:
```
07:57:34.026 CNTRLR « [Node 023] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Association
                      · Association Group Information
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Powerlevel
                      · Binary Switch
                      · Multilevel Switch
                      · Configuration
                      · Scene Actuator Configuration
                      · Scene Activation
                      · Notification
                      · Meter
                      · Basic
                      · Firmware Update Meta Data
                      · Multilevel Switch
```
but then when we request the supported versions, it tells us this:
```
07:57:34.148 CNTRLR » [Node 023] querying CC versions...
07:57:34.148 CNTRLR » [Node 023]   querying the CC version for Multilevel Switch...
07:57:34.197 CNTRLR   [Node 023]   does NOT support CC Multilevel Switch (0x26)
07:57:34.197 CNTRLR » [Node 023]   querying the CC version for Binary Switch...
07:57:34.252 CNTRLR   [Node 023]   supports CC Binary Switch (0x25) in version 1
```

As a result, the driver removes Multilevel Switch from the support list and you're only seeing a binary switch in the UI.

S1M8N, 2021-01-16:
> So, I took a look at your c421 log. First the device tells us that it supports these CCs:
>
> ```
> 07:57:34.026 CNTRLR « [Node 023] node info received
>                       supported [...]

Ok, this is annoying ... There is no solution? For my part I don't know how to do ... I hope we can sort this out ...

AlCalzone, 2021-01-16:
We have a way to override the reports using compatibility options: https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=commandclassesadd
So we should be able to say "this device does support Multilevel Switch V1"

S1M8N, 2021-01-16:
> So, I took a look at your c421 log. First the device tells us that it supports these CCs:
>
> ```
> 07:57:34.026 CNTRLR « [Node 023] node info received
>                       supported [...]

Ok

S1M8N, 2021-01-16:
> We have a way to override the reports using compatibility options: https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=commandclassesadd
> So we should be able to say "this device does support Multilevel Switch V1"

ok, are you going to work on it any time soon? Personally I do not know enough ... Make me a proposal ... (I did not purposely close the subject ...)

AlCalzone, 2021-01-16:
I can do it, just need to find the time.

S1M8N, 2021-01-16:
> I can do it, just need to find the time.

ok thank you. For the rest of the config files to add, how will you proceed?

AlCalzone, 2021-01-16:
I'll let you know when they are added so you can test.

AlCalzone, 2021-01-28:
@blhoward2 do you know if these devices are on your import list?

S1M8N, 2021-01-30:
Hello AlCalzone,

Following our discussion, I wanted to know when it will be possible to put the configuration files online (in the github) because as long as this is not done I cannot add my switches on my z-wave key and therefore to a lot of thing I'm stuck.

I also have my shutter panel which is considered to be a switch. This is very problematic for me and I hope it will be sorted out soon enough ...

Also if updates are available to solve the problem of my shutter panel how can I do them?

Thank you for your help and for the time you spent helping me.

Have a good day

Simon

blhoward2, 2021-01-30:
I'll check my list today and see if they're in the pile. @S1M8N thank you for your patience. We're in the middle of importing around 1,500 new devices so making other changes creates conflicts as we do so.

jcam, 2021-02-10:
Taking a look at this one.
For c421, is the issue that it is reporting as a binary switch vs a multilevel switch, or is there something else missing about it being a cover?

For the S314, I looked at the manual here: http://www.mcohome.com/ProductDetail/3894254.html
The parameters don't seem to match your file. Are the parameters definitely correct for the 1502/version 5 switch?
Thank you,
Jesse

S1M8N, 2021-02-10:
Hello @jcam

For part 1, the problem is that the c421 report as a binary switch but this is a multi-level switch (it's a shutter module).

For the last part, the configuration files are not completely accurate or finished because the example file I took had already not been done correctly. So I did not see that in the pdf of the MCO home there was even more configuration ... I am very sincerely sorry ... Do not hesitate to inform me of how you would like to proceed for the future 

Thank you very much

Simon

jcam, 2021-02-10:
No problems at all! I just wanted to make sure I should follow the PDF. I am close to having something ready for initial testing, hopefully this evening after work i can finish

S1M8N, 2021-02-10:
> No problems at all! I just wanted to make sure I should follow the PDF. I am close to having something ready for initial testing, hopefully this evening after work i can finish

Ok I remain available.

Thank you very much

jcam, 2021-02-10:
here's my WIP:
https://github.com/zwave-js/node-zwave-js/pull/1696/files

still need to do the dimmer and do another pass for cleanup

AlCalzone, 2021-02-12:
fixed in #1696

closed 2021-02-12 by AlCalzone