zwave-js/zwave-js #1402: [bug] Invalid data with Danalock V3-BTZE

cinadr, 2021-01-14:
**Version**

Build/Run method
- [ ] Docker
- [ ] PKG
- [x] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: 1.0.0-beta.0.8c826f1 (Windows Server 2019 Datacenter)

**Describe the bug**
Danalock V3-BTZE not updating status. 
ZWave-JS logs showing _Message with invalid data received. Dropping it_

**To Reproduce**
Manually turn Danalock or use proprietary app to change lock status.

**Expected behavior**
It should update with every status change of the phisical lock status.

**Additional context**
This might be  bug in node-zwave-js, I cannot decide.
Node info: [node_2.zip](https://github.com/zwave-js/zwavejs2mqtt/files/5816380/node_2.zip)
ZWave.log: [zwavejs_108.zip](https://github.com/zwave-js/zwavejs2mqtt/files/5816383/zwavejs_108.zip)
Relevant zwave.log excerpt :
From 19:36:24.884 to 19:36:26.529: Normal operation (open attempt and required update went well)
From 19:36:43.504 to 19:36:44.992: Erroneus operation (close attempt returns invalid data message)
```
19:36:24.884 SERIAL « 0x0108000400020298402b                                              (10 bytes)
19:36:24.884 SERIAL » [ACK]                                                                   (0x06)
19:36:24.885 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
19:36:24.893 SERIAL » 0x01110013020a9880a0c05375e6895f9c056765                            (19 bytes)
19:36:24.893 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      103
                      └─[SecurityCCNonceReport]
                          nonce: 0xa0c05375e6895f9c
19:36:24.895 SERIAL « [ACK]                                                                   (0x06)
19:36:24.901 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:36:24.901 SERIAL » [ACK]                                                                   (0x06)
19:36:24.902 DRIVER « [RES] [SendData]
                        was sent: true
19:36:24.932 SERIAL « 0x0108000400020298402b                                              (10 bytes)
19:36:24.933 SERIAL » [ACK]                                                                   (0x06)
19:36:24.933 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
19:36:24.937 SERIAL » 0x01030016ea                                                         (5 bytes)
19:36:24.937 DRIVER » [REQ] [SendDataAbort]
19:36:24.943 SERIAL « [ACK]                                                                   (0x06)
19:36:24.945 SERIAL » 0x01110013020a9880b4902ce551b8e3d60568b1                            (19 bytes)
19:36:24.945 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      104
                      └─[SecurityCCNonceReport]
                          nonce: 0xb4902ce551b8e3d6
19:36:24.947 SERIAL « [ACK]                                                                   (0x06)
19:36:24.954 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:36:24.955 SERIAL » [ACK]                                                                   (0x06)
19:36:24.956 DRIVER « [RES] [SendData]
                        was sent: true
19:36:24.976 SERIAL « 0x0108000400020298402b                                              (10 bytes)
19:36:24.976 SERIAL » [ACK]                                                                   (0x06)
19:36:24.977 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
19:36:24.984 SERIAL » 0x01030016ea                                                         (5 bytes)
19:36:24.984 DRIVER » [REQ] [SendDataAbort]
19:36:24.988 SERIAL « [ACK]                                                                   (0x06)
19:36:24.989 SERIAL » 0x01110013020a9880598ec55ea71e8ee405691e                            (19 bytes)
19:36:24.989 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
[... 215 lines cut ...]
19:36:44.938 SERIAL » [ACK]                                                                   (0x06)
19:36:44.939 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
19:36:44.945 SERIAL » 0x01110013020a98807e63b0f192e356e0057003                            (19 bytes)
19:36:44.945 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      112
                      └─[SecurityCCNonceReport]
                          nonce: 0x7e63b0f192e356e0
19:36:44.947 SERIAL « [ACK]                                                                   (0x06)
19:36:44.953 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:36:44.953 SERIAL » [ACK]                                                                   (0x06)
19:36:44.954 DRIVER « [RES] [SendData]
                        was sent: true
19:36:44.967 SERIAL « 0x0121000400021b9881e89460da2ebfa1e70ebd901ec46b4a1c583b1eea74d665e (35 bytes)
                      508b2
19:36:44.968 DRIVER   Message with invalid data received. Dropping it:
                      0x0121000400021b9881e89460da2ebfa1e70ebd901ec46b4a1c583b1eea74d665e508b2
19:36:44.968 SERIAL » [ACK]                                                                   (0x06)
19:36:44.991 SERIAL « 0x01070013700000049f                                                 (9 bytes)
19:36:44.992 SERIAL » [ACK]                                                                   (0x06)
19:36:44.992 DRIVER « [REQ] [SendData]
                        callback id:     112
                        transmit status: OK
```

AlCalzone, 2021-01-14:
Since the messages are encrypted, I can't tell anything without the network key.
You can send it to me via e-mail.

cinadr, 2021-01-15:
My network key is

D4C57BDD4CC466D30E2EC4C56A89FD11

Thanks

cinadr, 2021-01-15:
Replied and deleted. You should get it via email. Can't find your address though. Thank you.

AlCalzone, 2021-01-15:
I thought it was displayed somewhere but it seems not. I've added an address to my profile.

cinadr, 2021-01-15:
I've sent you an e-mail.

AlCalzone, 2021-01-15:
Thanks, I think I know what's going on.

AlCalzone, 2021-01-15:
Or maybe not... I thought we were expiring the nonces too early, but it seems like this device is trying to double-communicate with us.
Starting at `19:36:43.505` you see that the device requests a new nonce for encryption and then another one while we are still in the process of sending that back. That causes the first send process to be aborted mid-flight and the nonce invalidated. 
The device however received the first (now invalid) nonce and continues to communicate with it.

cinadr, 2021-01-15:
That was I think the same issue within OpenHAB before. The link is [here](https://community.openhab.org/t/danalock-not-reporting-status/91860). I don't konw if that helps or not. The solution was to ignore these duplicate entries...

AlCalzone, 2021-01-15:
I already had a "fix" for this, but it aborts the ongoing transmission instead of just not queuing the duplicate response. I'll post a PR shortly for you to test.

AlCalzone, 2021-01-15:
Try https://github.com/zwave-js/node-zwave-js/pull/1409 please.
In case you are not using docker, the bot will post an install link for a test build shortly.

cinadr, 2021-01-15:
Got this (no UI update, shows invalid dropped data:
```
20:33:59.490 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.491 SERIAL » [ACK]                                                                   (0x06)
20:33:59.493 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:33:59.495 SERIAL » 0x01110013020a988076b02078a83f2139059e67                            (19 bytes)
20:33:59.496 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      158
                      └─[SecurityCCNonceReport]
                          nonce: 0x76b02078a83f2139
20:33:59.498 SERIAL « [ACK]                                                                   (0x06)
20:33:59.504 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:33:59.504 SERIAL » [ACK]                                                                   (0x06)
20:33:59.504 DRIVER « [RES] [SendData]
                        was sent: true
20:33:59.528 SERIAL « 0x010700139e00000376                                                 (9 bytes)
20:33:59.528 SERIAL » [ACK]                                                                   (0x06)
20:33:59.528 DRIVER « [REQ] [SendData]
                        callback id:     158
                        transmit status: OK
20:33:59.550 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.551 SERIAL » [ACK]                                                                   (0x06)
20:33:59.551 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:33:59.563 SERIAL » 0x01110013020a9880b645471dcf9d9a00059f16                            (19 bytes)
20:33:59.563 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      159
                      └─[SecurityCCNonceReport]
                          nonce: 0xb645471dcf9d9a00
20:33:59.565 SERIAL « [ACK]                                                                   (0x06)
20:33:59.573 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:33:59.573 SERIAL » [ACK]                                                                   (0x06)
20:33:59.575 DRIVER « [RES] [SendData]
                        was sent: true
20:33:59.597 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.597 SERIAL » [ACK]                                                                   (0x06)
20:33:59.598 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:33:59.628 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.628 SERIAL » [ACK]                                                                   (0x06)
20:33:59.629 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:33:59.656 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.656 SERIAL » [ACK]                                                                   (0x06)
20:33:59.657 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:33:59.683 SERIAL « 0x0108000400020298402b                                              (10 bytes)
20:33:59.683 SERIAL » [ACK]                                                                   (0x06)
20:33:59.687 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:34:00.871 SERIAL « 0x010700139f000082f6                                                 (9 bytes)
20:34:00.871 SERIAL » [ACK]                                                                   (0x06)
20:34:00.871 DRIVER « [REQ] [SendData]
                        callback id:     159
                        transmit status: OK
20:34:00.873 SERIAL » 0x01110013020a98809ffac375307f0e1d05a0c7                            (19 bytes)
20:34:00.873 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      160
                      └─[SecurityCCNonceReport]
                          nonce: 0x9ffac375307f0e1d
20:34:00.875 SERIAL « [ACK]                                                                   (0x06)
20:34:00.884 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:34:00.885 SERIAL » [ACK]                                                                   (0x06)
20:34:00.886 DRIVER « [RES] [SendData]
                        was sent: true
20:34:00.906 SERIAL « 0x01070013a000000348                                                 (9 bytes)
[... 168 lines cut ...]
20:34:19.090 SERIAL » [ACK]                                                                   (0x06)
20:34:19.091 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
20:34:19.094 SERIAL » 0x01110013020a988052010e2d8eceeae805a973                            (19 bytes)
20:34:19.095 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      169
                      └─[SecurityCCNonceReport]
                          nonce: 0x52010e2d8eceeae8
[... cut ...]
20:34:19.126 DRIVER « [REQ] [SendData]
                        callback id:     169
                        transmit status: OK
20:34:19.155 SERIAL « 0x0121000400021b988157ae4990f80e7076503a7d87a66c87088677e1513155ce6 (35 bytes)
                      c3068
20:34:19.155 DRIVER   Message with invalid data received. Dropping it:
                      0x0121000400021b988157ae4990f80e7076503a7d87a66c87088677e1513155ce6c3068
20:34:19.155 SERIAL » [ACK]                                                                   (0x06)
```

AlCalzone, 2021-01-15:
Okay, next try. I've updated the branch and the bot is building a new release (do you even need that?)

LordMike, 2021-01-17:
I'm seeing these `Message with invalid data received. Dropping it` messages too, it seems a lot of messages are going out (ConfigurationCC interview), and then it pauses with this message every second or so..

Viewing a larger piece of logs, I noticed they came in pairs of 6. The below snippet contains 6 of these messages.

Nodes involved:

* 26 - LogicSoft ZDB5100
* 27 - LogicSoft ZDB5100
* 28 - LogicSoft ZDB5100
* 30 - LogicSoft ZDB5100
* 31 - LogicSoft ZDB5100
* 32 - LogicSoft ZDB5100

```
01:51:07.430 DRIVER « [RES] [SendData]
                        was sent: true
01:51:07.532 SERIAL « 0x010700138c00000b6c                                                 (9 bytes)
01:51:07.532 SERIAL » [ACK]                                                                   (0x06)
01:51:07.533 DRIVER « [REQ] [SendData]
                        callback id:     140
                        transmit status: OK
01:51:07.604 SERIAL « 0x01150004001c0f700b0020004b65792033206d6f6465c1                    (23 bytes)
01:51:07.605 SERIAL » [ACK]                                                                   (0x06)
01:51:07.606 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[ConfigurationCCNameReport]
                          parameter #:       32
                          name:              Key 3 mode
                          reports to follow: 0
01:51:07.610 SERIAL » 0x010b00131e04700a0020258d0f                                        (13 bytes)
01:51:07.610 DRIVER » [Node 030] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      141
                      └─[ConfigurationCCNameGet]
                          parameter #: 32
01:51:07.613 SERIAL « [ACK]                                                                   (0x06)
01:51:07.616 SERIAL « 0x0104011301e8                                                       (6 bytes)
01:51:07.616 SERIAL » [ACK]                                                                   (0x06)
01:51:07.616 DRIVER « [RES] [SendData]
                        was sent: true
01:51:07.712 SERIAL « 0x010700138d00000a6c                                                 (9 bytes)
01:51:07.712 SERIAL » [ACK]                                                                   (0x06)
01:51:07.713 DRIVER « [REQ] [SendData]
                        callback id:     141
                        transmit status: OK
01:51:07.789 SERIAL « 0x01150004001e0f700b0020004b65792033206d6f6465c3                    (23 bytes)
01:51:07.789 SERIAL » [ACK]                                                                   (0x06)
01:51:07.790 DRIVER « [Node 030] [REQ] [ApplicationCommand]
                      └─[ConfigurationCCNameReport]
                          parameter #:       32
                          name:              Key 3 mode
                          reports to follow: 0
01:51:07.794 SERIAL » 0x010b00131f04700a0020258e0d                                        (13 bytes)
01:51:07.795 DRIVER » [Node 031] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      142
                      └─[ConfigurationCCNameGet]
                          parameter #: 32
01:51:07.796 SERIAL « [ACK]                                                                   (0x06)
01:51:07.800 SERIAL « 0x0104011301e8                                                       (6 bytes)
01:51:07.800 SERIAL » [ACK]                                                                   (0x06)
01:51:07.800 DRIVER « [RES] [SendData]
                        was sent: true
01:51:07.922 SERIAL « 0x010700138e00000d68                                                 (9 bytes)
01:51:07.923 SERIAL » [ACK]                                                                   (0x06)
01:51:07.923 DRIVER « [REQ] [SendData]
                        callback id:     142
                        transmit status: OK
01:51:07.997 SERIAL « 0x01150004001f0f700b0020004b65792033206d6f6465c2                    (23 bytes)
01:51:07.998 SERIAL » [ACK]                                                                   (0x06)
01:51:07.998 DRIVER « [Node 031] [REQ] [ApplicationCommand]
[... 183 lines cut ...]
                      value format:        Enumerated
                      value size:          1 bytes
                      min value:           0
                      max value:           4
                      default value:       0
                      is read-only:        false
                      is advanced (UI):    false
                      has bulk support:    true
                      alters capabilities: false
01:51:58.564 CNTRLR » [Node 031] querying parameter #32 value...
01:51:58.567 SERIAL « [ACK]                                                                   (0x06)
01:51:58.568 SERIAL « 0x0104011301e8                                                       (6 bytes)
01:51:58.568 SERIAL » [ACK]                                                                   (0x06)
01:51:58.568 DRIVER « [RES] [SendData]
                        was sent: true
01:51:58.631 SERIAL « 0x010700139500000779                                                 (9 bytes)
01:51:58.631 SERIAL » [ACK]                                                                   (0x06)
01:51:58.632 DRIVER « [REQ] [SendData]
                        callback id:     149
                        transmit status: OK
01:51:58.702 SERIAL « 0x010b0004002005700d00200088                                        (13 bytes)
01:51:58.703 DRIVER   Message with invalid data received. Dropping it:
                      0x010b0004002005700d00200088
01:51:58.703 SERIAL » [ACK]                                                                   (0x06)
```

AlCalzone, 2021-01-17:
@LordMike can you put that into a separate issue please? This is unrelated aside from the fact that it displays the same error.

PR #1409 (2021-01-18): fix: ignore duplicate NonceGet instead of aborting previous transmission
In order to fix #1059 we aborted the ongoing NonceReport transmissions if a device requested a new one. This causes #1402 however. The more logical fix seems to be **not** to reply to duplicate requests while the transmission of a nonce is ongoing.

fixes: #1402
closed 2021-01-18 by AlCalzone