zwave-js/zwave-js #1419: Zooz ZST10 700 "Unable to write true on 37-1-targetValue" for Fibaro FGWPB-121 device.

fxl90, 2021-01-16:
**Describe the bug**

When using a Zooz ZST10 700 series stick, I am not able to turn on/off a Fibaro FGWPB-121 device. When I click "on" through the zwave2mqtt UI, I hear an audible click from the switch but it does not change current state.

**Device information**

Which device(s) is/are affected (make/model)?  Zooz ZST10 700 and Fibaro FGWPB-121
What are the node IDs? Node 1 and 2

**Last Known Working Configuration**
- [x]New Device:  **Zooz ZST10 700**
- [x ] Previously working device (node-zwave-js): **Aeotec Gen5 Stick**
    - Which library version/docker image/adapter version? **zwavejs2mqtt:dev and zwavejs2mqtt:latest**
    - Have you made any recent configuration changes to the device? Describe. None, only change was switching to Zoos ZST10 700 series
    
**Installation information**
How did you install `node-zwave-js`?
- [x ] Zwavejs2mqtt (latest) docker image
- [x ] Zwavejs2mqtt (dev) docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Control Panel
2. Click on Node 2 
3. Scroll down to Binary Switch and click
4. Click On or Off
5. No response
6. See error in log

**Additional context**
Add any other context about the problem here.

**Logfile:** 
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5824903/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5824904/zwavejs_1.log)

AlCalzone, 2021-01-16:
The get request after the set results in us receiving a `Application Status CC`, which we are not equipped to handle yet. So you don't see the updated value.

AlCalzone, 2021-01-19:
Please test https://github.com/zwave-js/node-zwave-js/pull/1453

AlCalzone, 2021-01-27:
Closing for now - feel free to reopen if the issue persists.

closed 2021-01-27 by AlCalzone