zwave-js/zwave-js #1421: Unable to set setpoint temperature without decimal for Everspring AC301

andresn81, 2021-01-16:
**Describe the bug**

I'm trying to set the temperature for my Everspring AC301 but it only works when doing numbers with a decimal, for example 22.5 works but not 22, 22.1 works, etc. 

**Device information**

Which device(s) is/are affected (make/model)?  Everspring AC301
What are the node IDs? 8

**Last Known Working Configuration**
- [x] New device

**Installation information**
How did you install `node-zwave-js`?
- [x] Pkg

**Additional context**
I remembered that I saw something about Heatit TRM3 a while ago ([link here](https://github.com/OpenZWave/open-zwave/files/5483761/Test.of.thermostat.Setpoint.Command.Class.pdf)) where they said that if the precision is wrong, the temperature will not be set. 

So... I went into core/values/primitive.js and had a look. Found encodeFloatWithScale() and I tried to see what the precision returned for 21 and 21.5. For 21, it was 0 (which converted the value to "21") and for 21.5 it was 1 (which converted the value to "215"). Then I hard-coded the precision to 1 in encodeFloatWithScale(), restarted zwave-js and BANG. It worked like a charm.

I don't know where the real issue is though, if there is different interpretations of the zwave standard or a bug in zwave-js. 

**Logfile:**

Included a few different logfiles:

[set_21_celsius.txt](https://github.com/zwave-js/node-zwave-js/files/5825062/set_21_celsius.txt)
[set_22_celsius.txt](https://github.com/zwave-js/node-zwave-js/files/5825064/set_22_celsius.txt)
[set_22.1_celsius.txt](https://github.com/zwave-js/node-zwave-js/files/5825065/set_22.1_celsius.txt)
[set_23.5_celsius.txt](https://github.com/zwave-js/node-zwave-js/files/5825066/set_23.5_celsius.txt)

This is the logfile for when I changed the precision to 1:
[set_22_celsius_precision_1.txt](https://github.com/zwave-js/node-zwave-js/files/5825063/set_22_celsius_precision_1.txt)

AlCalzone, 2021-01-16:
Can you show me your cache files too please?

andresn81, 2021-01-16:
Sure, here they are.

[f688c43c.txt](https://github.com/zwave-js/node-zwave-js/files/5825135/f688c43c.txt)
[f688c43c.metadata.txt](https://github.com/zwave-js/node-zwave-js/files/5825134/f688c43c.metadata.txt)
[f688c43c.values.txt](https://github.com/zwave-js/node-zwave-js/files/5825136/f688c43c.values.txt)

AlCalzone, 2021-01-16:
Ok that does look like a misinterpretation of the Z-Wave specs by the device. The device manual mentions that the setpoint precision is 1:
[image]

The specs say that the precision is used to indicate how many decimal places are included in the Value field - so that's related to how the current command is meant to be interpreted.

andresn81, 2021-01-16:
Alright, I guess Heatit does the same with TRM3. Maybe there needs to be some kind of flag in the config for these devices then. Or if the precision could be set manually when sending the command.

AlCalzone, 2021-01-16:
I'll add a flag.

kpine, 2021-01-17:
This is the OZW issue for the TRM3: https://github.com/OpenZWave/open-zwave/issues/2435. Not sure if it's the same as this, but that problem was due to the size value. The device requires the size to always be 2 bytes even if the encoded value could fit in one byte.

AlCalzone, 2021-01-17:
Seems to be very similar. My plan is to add a compat flag to fix the setpoint encoding parameters to specific values.

FYI: The recommended fix for TRM3 is to upgrade the firmware ;)

kpine, 2021-01-17:
> FYI: The recommended fix for TRM3 is to upgrade the firmware ;)

From my amateur viewpoint, I would agree, but Heat-it is adamant that they have done it to spec and the firmware is a "workaround" for OZW. 🤷‍♂️

AlCalzone, 2021-01-17:
> From my amateur viewpoint, I would agree, but Heat-it is adamant that they have done it to spec and the firmware is a "workaround" for OZW.

Yeah I can't get it into my head how anyone would interpret the specs that way. But I've also read that they only released an inofficial firmware because they can't change it without losing certification. 🤷🏻‍♂️

jcam, 2021-01-18:
Would other thermostats fail if there was always a decimal? i.e. generalize by always sending 20.0?

AlCalzone, 2021-01-18:
I don't like to do workarounds by default.

AlCalzone, 2021-01-18:
@andresn81 try https://github.com/zwave-js/node-zwave-js/pull/1443 please

andresn81, 2021-01-18:
Sure, I'll try tonight. Also bought a Z-TRM3, will install it and try there too.

AlCalzone, 2021-01-18:
Oh that's perfect!
@larstobi can you check too please? I can't find the unofficial firmware anymore, so this might interest more Z-TRM3 users.

larstobi, 2021-01-18:
I don't have the official firmware either... Maybe @roflmao has it?

roflmao, 2021-01-18:
My three Z-TRM3's have the modified firmware :\

AlCalzone, 2021-01-18:
I meant checking with the unofficial firmware if they still work with this change :)

roflmao, 2021-01-18:
Just set the setpoint to a value and it worked, any special temperatures you want me to test?

AlCalzone, 2021-01-18:
maybe a whole number once and then one with a decimal point.

roflmao, 2021-01-18:
@AlCalzone both 22 and 23,5 works ;)

AlCalzone, 2021-01-18:
So now I just need verification that @andresn81's device works too

andresn81, 2021-01-18:
I now have tried both devices and they are working properly, it is possible to change the setpoint on them.

PR #1443 (2021-01-18): feat(config): add compat option overrideFloatEncoding and use it for Z-TRM3 and AC301
fixes: #1421
and allows to operate Z-TRM3 without updating to the unofficial firmware.
closed 2021-01-18 by AlCalzone