zwave-js/zwave-js #1422: Aeotec Recessed Door/Window 7 Sensor adding as unknown

crzykidd, 2021-01-16:
**Describe the bug**
When adding the Aeotec Recess Door/Window 7 Sensor it is added as unknown.  
Device: ZW187-A
Firmware 1.4

Looking at the logs the product id shows up as 0xbb, while in the zw187.json it shows it as "0x00bb"

Slack thread on troubleshooting:
https://zwave2mqtt.slack.com/archives/C01DQ24SDKL/p1610822732275500

**Device information**

Which device(s) is/are affected (make/model)?  Aeotec ZW187-A 
What are the node IDs? 20

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X ] Previously working device (other platform)
    - Which platform? zwave2mqtt
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [X ] Zwavejs2mqtt (latest) docker image
- [ X] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**Logfile:** 
[zwavejs_1-aeotec_recessed_door_7.log](https://github.com/zwave-js/node-zwave-js/files/5825106/zwavejs_1-aeotec_recessed_door_7.log)

AlCalzone, 2021-01-16:
Thats simple. We don't have a matching config file for `0102:00bb`. The only one for ZW187 has `0002:00bb`.
Do you know if there's a difference between ZW187-A and the other versions?

@blhoward2 got some insight maybe?

blhoward2, 2021-01-16:
Let me check. I have both

blhoward2, 2021-01-16:
@AlCalzone The parameters are all the same and ozw uses one file for both. The 0x0002 model is for Europe and 0x103 for the US/Canada/Mexico.

AlCalzone, 2021-01-16:
Okay then we can add the ids to the existing file

PR #1425 (2021-01-16): feat(config): add additional aeotec door/window model
Closes #1422
closed 2021-01-16 by AlCalzone