zwave-js/zwave-js #1426: Homeseer WD200+ dimmer not providing updates to MQTT topics on change

crzykidd, 2021-01-16:
**Describe the bug**
When setting the changeValue topic with MQTT to turn on/off or set dim level the light changes to the correct setting but the MQTT topics currentValue and targetValue never get updated.
Discussion of issue: https://zwave2mqtt.slack.com/archives/C01DQ24SDKL/p1610818512273700

It looks like SupervisionCCReport command is failing in the zwave logs.

**Device information**
Which device(s) is/are affected (make/model)?  Homeseer WD200+
What are the node IDs? 22

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X] Previously working device (other platform)
    - Which platform? zwave2mqtt/HA Native Zwave
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [X] Zwavejs2mqtt (latest) docker image
- [X] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
Steps to reproduce the behavior:
1. issue:  z2m/nodeID_22/38/0/targetValue/set to 0
2. light turns off, but MQTT targetValue and CurrentValue do not get updated with a new timestamp or value.
3. issue:  z2m/nodeID_22/38/0/targetValue/set to 50 
4. light is set to dim level of 50, but targetValue and currentValue do not get updated with a new timestamp of value.

**Logfile:** 
[zwavejs_1_WD200_LightOn.log](https://github.com/zwave-js/node-zwave-js/files/5825210/zwavejs_1_WD200_LightOn.log)

AlCalzone, 2021-01-19:
Please test https://github.com/zwave-js/node-zwave-js/pull/1453

crzykidd, 2021-01-20:
@AlCalzone #1453 did not solve this issue.   Looking at new zwavejs logs it is the same message at the end after an update:
`19:25:51.271 SERIAL « 0x010d00040016056c02070000d50059                                    (15 bytes)
19:25:51.271 SERIAL » [ACK]                                                                   (0x06)
19:25:51.272 DRIVER « [Node 022] [REQ] [ApplicationCommand]
                      └─[SupervisionCCReport]
                          session id:          7
                          more updates follow: false
                          status:              NoSupport
                          duration:            [Duration: 0seconds]`

AlCalzone, 2021-01-20:
Oh, right - I just skimmed through the issues with the polling support label.

AlCalzone, 2021-01-23:
@crzykidd It might be that the device in question has a firmware issue. I would like to ask Homeseer about this, but I'll need another log that includes the entire interview process as well as a switching attempt.
You can do that by clicking `refresh info` in the dashboard while the log is active.

crzykidd, 2021-01-24:
@AlCalzone I am attaching new logs.   zwavejs_1.log has node 22 which is the Homeseer WD200+ from zwavejs2mqtt.    For comparison since the switches seem to be working fine in OZW with zwave2mqtt, I am attaching a log for that as well in OZW_log.txt   The switch in question in this log is node003.
[OZW_Log.txt](https://github.com/zwave-js/node-zwave-js/files/5863075/OZW_Log.txt)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5863076/zwavejs_1.log)

PR #1531 (2021-01-30): feat: add compat flag to remove support of CCs
and apply it to HomeSeer WD200+ to remove Supervision support

fixes: #1426
closed 2021-01-30 by AlCalzone