zwave-js/zwave-js #1431: "Fan Switch" specific device type label is missing (Z-Wave Plus Device Type)

kpine, 2021-01-17:
**Is your feature request related to a problem? Please describe.**
The Fan Switch specific device type has no label, it shows as UNKNOWN.

**Describe the solution you'd like**
The specific device type should say `Fan Switch`.

**Describe alternatives you've considered**
None.

**Additional context**
See example dump [here](https://pastebin.com/bKj6jcr9), node 6 for example.
```json
"deviceClass": {
					"basic": "Static Controller",
					"generic": "Multilevel Switch",
					"specific": "UNKNOWN (0x08)",
					"mandatorySupportedCCs": ["Basic", "Multilevel Switch"],
					"mandatoryControlCCs": []
				},
```
Should be
```json
"deviceClass": {
					"basic": "Static Controller",
					"generic": "Multilevel Switch",
					"specific": "Fan Switch",
					"mandatorySupportedCCs": ["Basic", "Multilevel Switch"],
					"mandatoryControlCCs": []
				},
```

AlCalzone, 2021-01-17:
You don't have the updated specs by any chance?

kpine, 2021-01-17:
The [device type spec](https://github.com/zwave-js/node-zwave-js/blob/master/specs/SDS11847%20Z-Wave%20Plus%20Device%20Type%20Specification.pdf
) is in this repo, section 4.8 for fan. Are you asking which document the actual value of SPECIFIC_TYPE_FAN_SWITCH is specified?

AlCalzone, 2021-01-17:
Uhh I think I don't have that part translated to JSON yet, only SDS10242. Or it might be an oversight.

If someone wants to help out - SDS13740 contains the values in a practical C-Code as PDF form 😖

kpine, 2021-01-17:
Huh, I'm missing SDS13740, good to know.

Is there more to it than just adding a new entry here?

https://github.com/zwave-js/node-zwave-js/blob/68c25e2b2372ef8592f8f1b0ca8aef1734c47085/packages/config/config/deviceClasses.json#L237

You have most of others, just not Fan and Color Tunable types.

kpine, 2021-01-17:
The C macros are also defined in [this](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/specs/API_includes/ZW_classcmd.h) header file. Like in SDS13740 they seem to have given up on assigning descriptive names (e.g. Fan Switch).

There are human readable strings in this public [XML](https://github.com/Z-WavePublic/libzwaveip/blob/ade735d6b8f364716334c89f4593b22cc5220d92/config/ZWave_custom_cmd_classes.xml#L116) file.

commit 82fabba7 (2021-01-17): fix(config): add missing specific device types

fixes: #1431
closed 2021-01-17 by AlCalzone