zwave-js/zwave-js #1433: [bug] Invalid data with ZDB5100

LordMike, 2021-01-17:
**Describe the bug**
I was watching logs and noticed that messages were being sent back and forth, until it hit this error message. Then a pause occurs, until messages begin being sent again ... until another one of these.

**Device information**
* Node 26 - LogicSoft ZDB5100
* Node 27 - LogicSoft ZDB5100
* Node 28 - LogicSoft ZDB5100
* Node 30 - LogicSoft ZDB5100
* Node 31 - LogicSoft ZDB5100
* Node 32 - LogicSoft ZDB5100

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [x] Previously working device (other platform)
    - Which platform? open-zwave
    - Have you made any recent configuration changes to the device? Describe.
        None. I'm testing zwave-js before migrating.
    
**Installation information**
How did you install `node-zwave-js`?
- [x] Zwavejs2mqtt (latest) docker image
- [ ] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
Steps to reproduce the behavior:
Unknown.

**Additional context**

**Logfile:** 
Viewing a larger piece of logs, I noticed they came in pairs of 6. The below snippet contains 6 of these messages.

```
01:51:07.430 DRIVER « [RES] [SendData]
                        was sent: true
01:51:07.532 SERIAL « 0x010700138c00000b6c                                                 (9 bytes)
01:51:07.532 SERIAL » [ACK]                                                                   (0x06)
01:51:07.533 DRIVER « [REQ] [SendData]
                        callback id:     140
                        transmit status: OK
01:51:07.604 SERIAL « 0x01150004001c0f700b0020004b65792033206d6f6465c1                    (23 bytes)
01:51:07.605 SERIAL » [ACK]                                                                   (0x06)
01:51:07.606 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[ConfigurationCCNameReport]
                          parameter #:       32
                          name:              Key 3 mode
                          reports to follow: 0
01:51:07.610 SERIAL » 0x010b00131e04700a0020258d0f                                        (13 bytes)
01:51:07.610 DRIVER » [Node 030] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      141
                      └─[ConfigurationCCNameGet]
                          parameter #: 32
01:51:07.613 SERIAL « [ACK]                                                                   (0x06)
01:51:07.616 SERIAL « 0x0104011301e8                                                       (6 bytes)
01:51:07.616 SERIAL » [ACK]                                                                   (0x06)
01:51:07.616 DRIVER « [RES] [SendData]
                        was sent: true
01:51:07.712 SERIAL « 0x010700138d00000a6c                                                 (9 bytes)
[... 213 lines cut ...]
                      value format:        Enumerated
                      value size:          1 bytes
                      min value:           0
                      max value:           4
                      default value:       0
                      is read-only:        false
                      is advanced (UI):    false
                      has bulk support:    true
                      alters capabilities: false
01:51:58.564 CNTRLR » [Node 031] querying parameter #32 value...
01:51:58.567 SERIAL « [ACK]                                                                   (0x06)
01:51:58.568 SERIAL « 0x0104011301e8                                                       (6 bytes)
01:51:58.568 SERIAL » [ACK]                                                                   (0x06)
01:51:58.568 DRIVER « [RES] [SendData]
                        was sent: true
01:51:58.631 SERIAL « 0x010700139500000779                                                 (9 bytes)
01:51:58.631 SERIAL » [ACK]                                                                   (0x06)
01:51:58.632 DRIVER « [REQ] [SendData]
                        callback id:     149
                        transmit status: OK
01:51:58.702 SERIAL « 0x010b0004002005700d00200088                                        (13 bytes)
01:51:58.703 DRIVER   Message with invalid data received. Dropping it:
                      0x010b0004002005700d00200088
01:51:58.703 SERIAL » [ACK]                                                                   (0x06)
```

closed 2021-01-17 by AlCalzone

LordMike, 2021-01-17:
Wow.. These devices seem to not play well.. Wonder if they're out of spec :D

It's the same device that had the configuration getProperties(0) thing.. :P

AlCalzone, 2021-01-18:
> It's the same product that had the configuration getProperties(0) thing.. :P

🙈

PR #1435 (2021-01-17): fix(cc): accept empty configuration name and info
fixes: #1433