zwave-js/zwave-js #1436: Feature request: More concise format for device configs reg. bitsets

LordMike, 2021-01-17:
**Is your feature request related to a problem? Please describe.**
I was changing some configs that I'll add a PR for later, and I was introducing a number of bitsets (changing existing numerical parameters to explain each bit). The current format is quite verbose and tough to write a lot of.

**Describe the solution you'd like**
I'll propose this alternate format. The trick here, I think, is to internally (in zwave-js) convert this config-file format, into the current format available today. It's my impression that other libraries read the device configs, via zwave-js, and not directly from the files available. If that's the case, this would be entirely backwards compatible with existing users.

The current format looks like this:

```json
..
"9[0x01]": {
	"label": "Alarm reaction to Access Control trigger state",
	"description": "React to access control triggers from other Z-Wave devices.",
	"valueSize": 2,
	"minValue": 0,
	"maxValue": 1,
	"defaultValue": 0,
	"readOnly": false,
	"writeOnly": false,
	"allowManualEntry": true
},
"9[0x0100]": {
	"label": "Alarm reaction to Smoke Alarms",
	"description": "React to Smoke Alarms from other Z-Wave devices.",
	"valueSize": 2,
	"minValue": 0,
	"maxValue": 1,
	"defaultValue": 0,
	"readOnly": false,
	"writeOnly": false,
	"allowManualEntry": true
},
..
``` 

It's clunky, as we must repeat a number of values between settings - and come up with multiple labels and so on. I propose the following instead:

```json
..
"9": {
	"label": "Alarm reaction config",
	"description": "React to the following alarms from other Z-Wave devices.",
	"valueSize": 2,
	"minValue": 0,
	"maxValue": 16384,
        // Default stays out here, and is now for the entire value. In this product, it really is 0, but it could be any bitmasked value
	"defaultValue": 0,
	"readOnly": false,
	"writeOnly": false,
	"allowManualEntry": true,
	"bitset": {
            // Each bit is explained here. As it is a bitset, derived partial values will all encompass exactly 1 bit
            // When expanded, the key in this map is used where partial values would normally be set up
	    "0x01": {
                  // Having this as an object allows for future expansion with more data
		  "label": "Access Control trigger state"
		},
	    "0x100": {
		  // The texts could be standalone (ie. remove label+description from the main parameter setting)
		  "label": "React to Smoke Alarms",
		  "description": "React to Smoke Alarms from the z-wave network"
		},
	    "0x200": {
		  // .. or, the texts could be suffixed to the parent setting, duplicating the description between all bitset settings
		  // This would be come "Alarm reaction config: CO Alarm" 
		  "label": "CO Alarm"
		}
	}
},
..
```

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

AlCalzone, 2021-01-17:
I've been thinking about something like this, to also support the use case in #1411 better.

I agree that some of the values could be moved out, but your proposal is going a bit too far IMO. Havin min/max/default values outside the partial params means they are decoupled from them again and you have to do mental gymnastics to figure out what the ranges of the partial params are (defeating the purpose). You might be forgetting that partial params are not necessarily single-bit, they can each target multiple bits simultaneously.

Defining valueSize, readOnly, writeOnly only once does make sense. I could also see combining the labels of the compound and partial params (your 2nd proposal).

LordMike, 2021-01-17:
Ah yes, but min/max etc make no sense for bitsets, which are strictly single-bits. Bitsets occur quite often in configs, so they warrant some simpler way of specifying. Likewise the default value (at least in the docs I've seen) is already some number - so you'd conversely have to do "mental gymnastics" to derive individual bits defaults.

LordMike, 2021-01-17:
You've hinted at mental gymnastics a few times now.. If bitsets are that tricky for users, you could also support a binary field, like `"0b00111001"`, which would become 113

LordMike, 2021-01-17:
Kinda fun. I just noticed another issue with the original ZW175 config - the max of `16384` doesn't allow for enabling all options. You can enable the highest value option, or all of the others - but not all options at once.. :D

It's fixed now - but it's a fun anecdote.

AlCalzone, 2021-01-17:
> Ah yes, but min/max etc make no sense for bitsets, which are strictly single-bits

We could make those optional if the bitmask is only a single bit.

> Likewise the default value (at least in the docs I've seen) is already some number

Hmm, I guess we could allow deriving min/max/default for the bit masks from a common value.

> the max of 16384 doesn't allow for enabling all options.

I thought we had a check for that.

LordMike, 2021-01-17:
> I thought we had a check for that.

Irrelevant. Previous config was a list that only allowed for one value. I would have copied it, which would've made it an error.. That was the fun part :) 

But hey - cool check :)

AlCalzone, 2021-02-09:
superseded by #1675

closed 2021-02-09 by AlCalzone