zwave-js/zwave-js #1446: [question]  Support for ZME_054436 scene commands

haavardNO, 2021-01-19:
I have a ZME_054436 wall controller. Not realy sure if it is a Düwi or zwave.me product.
[image]

Previously I have had this up and running when I was using Indigo as zwave controller (also aeotech usb stick)
The trick is to modify parameter 13 and 14.
It will then be possible to trigger scenes:

Scene 1: Single Click on Upper Part of Paddle
 Scene 2: Single Click on Lower Part of Paddle
 Scene 3: Single Click Dim Up
 Scene 4: Single Click Dim Down
 Scene 5: Single Click Dim Up Stop
 Scene 6: Single Click Dim Down Stop
 Scene 11: Double Click on Upper Part of Paddle
 Scene 12: Double Click on Lower Part of Paddle
 Scene 13: Double Click Dim Up
 Scene 14: Double Click Dim Down
 Scene 15: Double Click Dim Up Stop
 Scene 16: Double Click Dim Down Stop

http://bustoautomatika.lt/wp-content/uploads/ZME_054436_Jungiklis-valdiklis-bevielis-Wall-Controller-Set-for-DUWI-Everlux_I_EN.pdf)

Any chance of adding support for this device? 
(Hope so, becouse I have several of then stuck on my walls around the house)

AlCalzone, 2021-01-19:
@robertsLando can you add a way to manually set a configuration value through the ui (number, size and value)?

That would at least be a workaround for devices we don't have a config file for.

haavardNO, 2021-01-19:
Good point @AlCalzone, but in this specific case, I guess we need a way to handle the scene commands as well?

AlCalzone, 2021-01-19:
The scene commands should be understood when they arrive. I'm guessing all you are missing is the device name and the configuration parameters? If not, I'd need to see a log.

haavardNO, 2021-01-19:
Yes. That is correct.

AlCalzone, 2021-01-19:
I can't promise when I'll find the time. If you want to speed up the process, you could try to provide a config file yourself.
https://zwave-js.github.io/node-zwave-js/#/development/config-files describes the process and how you could import it from OZW or OpenHAB if they do have a file for it.

haavardNO, 2021-01-19:
In best case I am more of a "copy paste" programmer, but I will see what I can figure out.

haavardNO, 2021-01-19:
After reading up on the device I learned that it has very short time awake. So the cause for the issue was that device was not properly interviewed. The trick was to constantly wake the device during configuration. Then full config appeared and I am able to trigger on scenes.

Sorry for the confusion.

AlCalzone, 2021-01-19:
Thanks for the update, closing then...

closed 2021-01-19 by AlCalzone

haavardNO, 2021-02-21:
@AlCalzone not sure what might have happened. But I have not been using this device for a while now, but suddenly today when I try to use it it does not work. When I investigate further I see that in the zwavejs2mqtt dashboard, the sensor_scene states is missing. I can see they are activated in the debug. Any idea what could be causing the changed behaviour?

```
2021-02-21 13:21:34.144 INFO ZWAVE: Node 5: value notification: 43-0-sceneId 2
2021-02-21 13:21:34.148 INFO ZWAVE: Node 5: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-21 13:21:36.958 INFO ZWAVE: Node 5: value notification: 43-0-sceneId 1
2021-02-21 13:21:36.964 INFO ZWAVE: Node 5: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-21 13:21:39.460 INFO ZWAVE: Node 5: value notification: 43-0-sceneId 2
2021-02-21 13:21:39.465 INFO ZWAVE: Node 5: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-21 13:21:40.659 INFO ZWAVE: Node 5: value notification: 43-0-sceneId 1
2021-02-21 13:21:40.664 INFO ZWAVE: Node 5: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
```

[image]
[node_5.json.zip](https://github.com/zwave-js/node-zwave-js/files/6017678/node_5.json.zip)

AlCalzone, 2021-02-21:
It looks like zjs2mqtt receives the events so the problem shouldn't be in node-zwave-js.

@robertsLando ?

robertsLando, 2021-02-22:
@haavardNO I think that the problem is mqtt discovery, @billiaz Any clue?

billiaz, 2021-02-22:
reading the Json We miss completely the Scenes class 43 has only dimmingDuration in this device!?

do you see any scenes also on your gui (zj2m)? I suspect not under Node tab

haavardNO, 2021-02-22:
@billiaz I have this in the UI:
[image]

AlCalzone, 2021-02-22:
@robertsLando https://zwave-js.github.io/node-zwave-js/#/api/node?id=getvaluemetadata
Might have to do what I wrote in the blue box

robertsLando, 2021-02-22:
@AlCalzone I manually call the getValue metadata on each value when node is ready and then I update it when I receive the event, so what could I miss here?

AlCalzone, 2021-02-22:
I'd need the cache files including metadata to verify if this is after all a bug in node-zwave-js

haavardNO, 2021-02-22:
If there is files that I should upload please explain a little more in detail how to fetch them...

AlCalzone, 2021-02-22:
I need these files: https://zwave-js.github.io/node-zwave-js/#/development/troubleshooting?id=providing-the-necessary-information-for-an-issue
Here's how to fetch them: https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report

haavardNO, 2021-02-22:
See below. I have attached the file shown in the image. Please let me know if this is not the correct file. I have also added the node file.

[image]

[node_5.json.zip](https://github.com/zwave-js/node-zwave-js/files/6023727/node_5.json.zip)
[zwavejs2mqtt-store-2.zip](https://github.com/zwave-js/node-zwave-js/files/6023722/zwavejs2mqtt-store-2.zip)

haavardNO, 2021-02-22:
@AlCalzone Also, I need to add that there is something strange going on. Seams as when I disable silly logging the node 5 device disappears. Then when I try to add it from the UI and tripple click the paddle it will pop back with the same node and name as before...

AlCalzone, 2021-02-22:
That is not the correct file. I needed the network cache files <hex-number>.json, <hex-number>.values.jsonl and <hex-number>.metadata.jsonl like the documentation states.

I have no idea why a node would vanish if you enable logging. @robertsLando?

Anyways, I think this is a duplicate of https://github.com/zwave-js/zwavejs2mqtt/issues/471 which has a PR to fix it.

haavardNO, 2021-02-23:
See below, hope this is the correct files :)
[zwavejs2mqtt-store 2.zip](https://github.com/zwave-js/node-zwave-js/files/6027298/zwavejs2mqtt-store.2.zip)

robertsLando, 2021-02-23:
> I have no idea why a node would vanish if you enable logging. @robertsLando?

Nope that has no sense

AlCalzone, 2021-02-23:
@haavardNO can you try to build zwavejs2mqtt with this branch:
https://github.com/zwave-js/node-zwave-js/pull/1849

haavardNO, 2021-02-23:
@AlCalzone sure, happy to test.
I am running zwavejs2mqtt on a rpi with docker. Can you please explain how to install a specified branch?

AlCalzone, 2021-02-23:
Normally, I'd link you here: https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

But AFAIK the docker custom builds are broken right now.
@robertsLando is there a way he can test the zwave-js master branch with docker?

robertsLando, 2021-02-24:
That's the only way we have (with docker)

AlCalzone, 2021-02-24:
Yeah, but Dockerfile.contrib is broken.

robertsLando, 2021-02-24:
Will try to see if I find some time today to fix that

robertsLando, 2021-02-24:
Another way could be that you pack that branch and he try to run npm install instead of using the yarn link

robertsLando, 2021-02-25:
If you need to install nodejs: 

```
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.0/install.sh | bash
nvm install --lts
nvm use --lts
```

Install yarn globally: `npm install -g yarn`

Clone repos:
```
mkdir -p testing && cd testing
git clone https://github.com/zwave-js/node-zwave-js
git clone https://github.com/zwave-js/zwavejs2mqtt

cd node-zwave-js
git checkout <branchname>
yarn install
yarn run build:full
yarn link

cd ../zwavejs2mqtt
git checkout <branch>
npm install
npm run build
yarn link zwave-js
npm run start
```

haavardNO, 2021-02-28:
Ok.
So after quite a lot of hassle, I think I have it running. It behaves the same as before, but is there a way to confirm that I was able to get the correct build?

[image]

haavardNO, 2021-02-28:
@AlCalzone seams like #1849 was merged in to the master..?
So when I updated with docker as well I should get the latest versjon. But still strange results on my side.
```
2021-02-28 20:30:25.627 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 1
2021-02-28 20:30:25.634 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-28 20:30:26.879 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 2
2021-02-28 20:30:26.885 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-28 20:30:28.153 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 3
2021-02-28 20:30:28.161 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-28 20:30:29.142 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 5
2021-02-28 20:30:29.146 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-28 20:30:30.534 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 4
2021-02-28 20:30:30.539 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
2021-02-28 20:30:31.384 INFO ZWAVE: Node 10: value notification: 43-0-sceneId 6
2021-02-28 20:30:31.389 INFO ZWAVE: Node 10: value updated: 43-0-dimmingDuration [Duration: default] => [Duration: default]
```

Still get this in the debug log. 
The scene control tab under configuration seams unstable. It pops up the first time I include the device after a reset. But it can disappear if I reinterview the device.
But the log entry is rock solid.
I have several of these devices, and have been using them for a long time, so I know the can work 😊

AlCalzone, 2021-02-28:
> But it can disappear if I reinterview the device.

Can you share a log of such an interview (where it disappears) and the cache files afterwards?

haavardNO, 2021-03-07:
@AlCalzone  I Have not been able to look at this again before now.
But it is a bit strange, because when I enable logging, and click save the scene activation tab disappears... Not sure if the logs is of any value...

But also verry strange becouse the basic log display the sceneId every time...

AlCalzone, 2021-03-07:
If the UI does something strange, that is an issue in zwavejs2mqtt.
To find out if zwave-js does something wrong, I need a zwave-js log (including the re-interview) where this happens.

haavardNO, 2021-03-07:
Thanks for spending the time @AlCalzone .
[zwavejs2mqtt-store-2.zip](https://github.com/zwave-js/node-zwave-js/files/6097735/zwavejs2mqtt-store-2.zip)
I have exported the log files I can access from the UI in zwavejs2mqtt. Since I am running in docker it is a little difficult to access the files. I hope this is sufficient.

I removed the settings file, since I assume you dont need it.

The way this occurs is as follows:
The scene device is removed from controller.
Reset device
Add device to zwave2mqtt
Enable parameter 13&14 --> 3

At this stage the scene config appears in the UI, but only containing settings for dimming duration.
If I now run a re interview the scene tab disappears.

haavardNO, 2021-03-16:
I have moved to the new WS Server option.
So I am no longer using these devices in the same way. Investigating more pressing issues with my z-wave network at the moment, so I think we can push this issue further ahead.

haavardNO, 2021-03-18:
And I am back with this device again @AlCalzone. The problem is that it is such a nice controller when it works. And I have glued it to a dry wall, so not so easy to get rid of it 😊 .
Did something change in the last updates regarding this device?
When I try to change parameter 13 to 3, scenes, it jumps back when I push send. I I was planning to use events to get it to play with Home Assistant.

[image]
[node_91.json.zip](https://github.com/zwave-js/node-zwave-js/files/6166417/node_91.json.zip)

AlCalzone, 2021-03-18:
Can you make a log where you attempt to change that configuration? Just to make sure the correct value gets sent.
I remember there was a bug in a previous zwavejs2mqtt version.

haavardNO, 2021-03-19:
@AlCalzone below is the logs you requested for node 91. The code below gets printet in the debug window when I push send on the parameter, and value jumps back to default it event jumps back before I wake up the device.

I have done some reinterviews and tried to select it in the menu as requested. Please see attached.
And note that this is regarding setting parameter 13 to 3 on node 91.

```
2021-03-19 18:31:51.315 DEBUG SOCKET: Event ZWAVE_API emitted to J8Vmvh8cdJoqnmy5AAmY
2021-03-19 18:31:51.318 INFO ZWAVE: Calling api writeValue with args: [
{ nodeId: 91, commandClass: 112, endpoint: 0, property: 13 },
0,
[length]: 2
]
2021-03-19 18:31:51.319 INFO ZWAVE: Writing 0 to 91-112-0-13
```

[zwavejs2mqtt-store-6.zip](https://github.com/zwave-js/node-zwave-js/files/6173127/zwavejs2mqtt-store-6.zip)

AlCalzone, 2021-03-19:
Hm I don't see any attempt to set value 3 - zwavejs2mqtt must send the wrong value.
Are you on the latest version? I remember there was a fix for something similar (or this?) not long ago.

haavardNO, 2021-03-19:
Not on the exactly latest. One release behind.
Will update and report back!

haavardNO, 2021-03-20:
@AlCalzone,  I updated to latest version. And it behaves the same. I can select the option, but when I press send on the blue button it jumps back to 0. I tried to reset the device and include it agin. But same results. Had this working for some time ago, but cant remember what version unfortunately.

[image]

[image]
[node_94.json.zip](https://github.com/zwave-js/node-zwave-js/files/6175462/node_94.json.zip)

[image]

[image]

AlCalzone, 2021-03-20:
@robertsLando any clue what might be going on?

robertsLando, 2021-03-22:
hummm I didn't change anything that could break that. Let me check. Seems that the problem is that you are not able to send custom values or values that are in list?

AlCalzone, 2021-03-22:
It's always sending 0, no matter what's selected from the list (if I understood correctly)

haavardNO, 2021-03-22:
Yes. That's correct.
It jumps back to 0 after I press send. And stays there when I wake up the device.

robertsLando, 2021-03-22:
I have merged an update that fixed this on my side. Wait until next patch release 2.4.1

haavardNO, 2021-03-22:
Tanks @robertsLando.
When will that release be ready you think?

robertsLando, 2021-03-22:
In 20 min or less

haavardNO, 2021-03-23:
@robertsLando and @AlCalzone  that fixed that issue. Now able to send configuration to the device. However seams there is a new or other issue. Before I would get an event that I could use to trigger scenes on in Home Assistant. Now I get the following in the debug log:

```
07:53:16.394 CNTRLR « [Node 094] TODO: no handler for application command
07:53:17.675 CNTRLR « [Node 094] TODO: no handler for application command
07:53:18.677 CNTRLR « [Node 094] TODO: no handler for application command
07:53:19.448 CNTRLR « [Node 094] TODO: no handler for application command
07:53:20.165 CNTRLR « [Node 094] TODO: no handler for application command
07:53:20.835 CNTRLR « [Node 094] TODO: no handler for application command
07:53:21.602 CNTRLR « [Node 094] TODO: no handler for application command
07:53:25.263 CNTRLR « [Node 094] TODO: no handler for application command
07:53:26.833 CNTRLR « [Node 094] TODO: no handler for application command
```

robertsLando, 2021-03-23:
@haavardNO I think that means there CC is not implemented

AlCalzone, 2021-03-23:
Can you up the log level please so I can see what's actually being sent?

haavardNO, 2021-03-24:
Yes. See below 😊

```
5:47:39.584 DRIVER « [Node 009] [REQ] [ApplicationCommand]
└─[BinarySensorCCReport]
type: Any
value: true
15:47:39.624 SERIAL « 0x010a0004005e042b0101ff7f (12 bytes)
15:47:39.627 CNTRLR [Node 094] [!] [Scene Activation] sceneId: 1 [Endpoint 0]
15:47:39.629 CNTRLR [Node 094] [~] [Scene Activation] dimmingDuration: "default" => " [Endpoint 0]
default"
15:47:39.631 SERIAL » [ACK] (0x06)
15:47:39.634 DRIVER « [Node 094] [REQ] [ApplicationCommand]
└─[SceneActivationCCSet]
scene id: 1
dimming duration: [Duration: default]
15:47:39.638 CNTRLR « [Node 094] TODO: no handler for application command
15:47:39.652 SERIAL « 0x010a0004005e042b0101ff7f (12 bytes)
15:47:39.654 CNTRLR [Node 094] [!] [Scene Activation] sceneId: 1 [Endpoint 0]
15:47:39.657 CNTRLR [Node 094] [~] [Scene Activation] dimmingDuration: "default" => " [Endpoint 0]
default"
15:47:39.659 SERIAL » [ACK] (0x06)
15:47:39.663 DRIVER « [Node 094] [REQ] [ApplicationCommand]
└─[SceneActivationCCSet]
scene id: 1
dimming duration: [Duration: default]
15:47:39.667 CNTRLR « [Node 094] TODO: no handler for application command
15:47:39.794 SERIAL « 0x0110000400090a7105000000ff070800006c (18 bytes)
15:47:39.796 SERIAL » [ACK] (0x06)
15:47:39.799 DRIVER « [Node 009] [REQ] [ApplicationCommand]
└─[NotificationCCReport]
notification type: Home Security
notification status: 255
notification state: Motion detection
15:47:39.803 CNTRLR [Node 009] [~] [Notification] Home Security[Motion sensor status] [Endpoint 0]
: 0 => 8
```

AlCalzone, 2021-03-24:
Ok, that is just a minor inconvenience. The command is actually handled and an event is emitted:
```
15:47:39.654 CNTRLR [Node 094] [!] [Scene Activation] sceneId: 1 [Endpoint 0]
15:47:39.657 CNTRLR [Node 094] [~] [Scene Activation] dimmingDuration: "default" => " [Endpoint 0]
default"
```
but the generic handler still thinks it needs to handle the command (and therefore logs the message).

haavardNO, 2021-03-25:
Thanks @AlCalzone.
I see now that this is a UI issue only, and that I do receive the event i Home Assistant.