zwave-js/zwave-js #1464: aeotec aerQ ZWA009-A not identified

crzykidd, 2021-01-21:
**Describe the bug**
The US version of this device ZWA009-A has a product id of 0x0102.  Currently the device only lists 0x0002 in: 
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0371/zwa009.json

This can be identified here:
https://products.z-wavealliance.org/products/3982?selectedFrequencyId=-1

The docs are the same for both sensors so all the commands should be the same.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] Zwavejs2mqtt (latest) docker image
- [X ] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`)

AlCalzone, 2021-01-21:
Accepting PR. You can just edit the file and add the additional ID.

AlCalzone, 2021-01-21:
fixed in #1465

closed 2021-01-21 by AlCalzone