zwave-js/zwave-js #1466: First Alert ZWave Combo [question]

ragenhe, 2021-01-21:
I've converted from Home Assistant native zwave support to zwavejs2mqtt.

So far so good.  I hope to contribute to this great project.

Is the First Alert ZWave Combo Smoke / CO2 detector supported?

I have excluded it and included it without issue.  I know it can take some time for a battery device to complete the query process, but I'm not sure if it is just taking a while to show it's details or if it is possible not supported.

Thank you!!

blhoward2, 2021-01-24:
Note that this device has always been terrible to interview under any hub. You generally have to remove the battery and close the door with the button down multiple times to complete the interview process.

ragenhe, 2021-01-24:
Yes, it can be a pain, and I've tried from many angles.

So much so the family is over me messing with the thing, LOL.

I'm game to try anything, even a new hub, but this device and hub have been
paired prior to using zwavejs.

I'll be patient and wait.  The project is great and I want to help it
advance.

blhoward2, 2021-01-24:
I didn’t mean that there isn’t a problem her. They was a note for @AlCalzone that it may still be abnormal in the logs/procedure.

ragenhe, 2021-01-24:
@blhoward2 - I didn't take it that way, no worries at all!!

AlCalzone, 2021-01-27:
@ragenhe please try #1505

ragenhe, 2021-01-27:
I updated my dev docker image. I then entered the container and updated via npm install zwave-js@6.1.1-0-pr-1505-0102cdc.

I then manually woke the device up twice.  I'll give it a little while and will continue the process.

Thanks!

AlCalzone, 2021-01-27:
if it doesn't work, log please ;)

ragenhe, 2021-01-27:
I'm going to continue waking the device and then waiting to see if the end result goal is reached.

But here are the logs so you can see if it is reacting as expected at this point in the process.
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5880992/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5880993/zwavejs_1.log)

ragenhe, 2021-01-27:
Because this device can be..... is..... difficult, I want to make sure I give you everything I have.

I again manually woke this device up twice and then restarted the container.

No progress towards in result goal.  Logs are attached.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5881603/zwavejs_1.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5881604/zwavejs2mqtt.log)

ragenhe, 2021-01-27:
Another manual wake up....

[image]

AlCalzone, 2021-01-27:
Okay, I'll take a look at your logs later

ragenhe, 2021-01-27:
The status in the screen shot it the same still.  Here are the most recent logs.

No rush! Thanks so much!!!

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5882387/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5882388/zwavejs_1.log)

AlCalzone, 2021-01-27:
```
13:52:01.268 DRIVER   version 6.0.2
```
@ragenhe doesn't look like your build is actually using the new version.

ragenhe, 2021-01-27:
I ran this from within the container
`npm install zwave-js@6.1.1-0-pr-1505-0102cdc`

is this the correct approach to update?

AlCalzone, 2021-01-27:
I don't think so. ==> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

ragenhe, 2021-01-27:
I'm on it.  Thanks!

ragenhe, 2021-01-28:
@AlCalzone - I have followed those directions, but must be missing a step.

I never see the version # get to 6.1.1-0-pr-1505-0102cdc.

AlCalzone, 2021-01-28:
With the custom build you have to set the branch name to use.
You should be able to find that in the PR header, something like "wants to merge branch xyz into master".

ragenhe, 2021-01-28:
@AlCalzone - Sorry to be a pain.  I attempted `git checkout cc-no-response` and there isn't one in either repo.

Can you see what I should be using instead of `cc-no-responce`?  Please?

AlCalzone, 2021-01-28:
Oh yeah things move fast here. Try the master branch 😅

ragenhe, 2021-01-28:
@AlCalzone - I finally have an update to report.  Using the current 91/28/2021) master branch of both repos, I successfully included the ZCOMBO and with just one additional manual wake up of the device, it was detected 100% and all expected node entries were found on my MQTT broker.

Please let me know if you want to see any of the logs, or need more details.

Thanks for all the help!!
Ragen

AlCalzone, 2021-01-28:
> EDIT: On the HA side I only have a battery_level sensor and a battery_low sensor, but I will continue working on it all until all sensors make it to HA.

Try waking it up a couple of times. Sounds like it is asleep.

closed 2021-01-29 by robertsLando

gregsheremeta, 2021-06-01:
@ragenhe I got mine to include, but like you I only see the battery sensors in HA. Did you get the actual alarm sensors working in HA?

ragenhe, 2021-06-01:
I did.  It all works as it should with the most recent version.