zwave-js/zwave-js #1471: [Bug] Aeotec Single/Dual Nano Switch: originally was becoming "Unknown Node" when added securely, now reverts to insecure inclusion

FreelancerJay, 2021-01-22:
**Describe the bug**

So originally (on Monday, when I had them installed), I was able to add my Aeotec Dual and Single Nano Switches Securely, however they would only display "Unknown Node" in the MQTT interface, and very few values would come up in the Node page.
I reset everything and added them insecurely, which worked fine, since I was paying for the electricians by the hour and we still had a few other bits of work I needed done.
I did notice that if I just excluded the switches and re-included them as new nodes following their directions for insecure inclusion, they would still end up securely included. To get around this I ended up having to factory reset the switches and the Z-Stick. Good thing they're the first things I had included to it!
I was running the "Beta.0" zwavejs2mqtt at that stage.

Today I have tried to duplicate the issue using the dev branch, however the behaviour is different. When I attempt to add the Nano Switch securely, it proceeds as normal for a bit (blue flashing LED), but then flashes a green LED (indicating insecure inclusion) and switches off. It then completes the interview as insecure.

The ZwaveJS log says on line 334:
`03:38:10.475 CNTRLR   [Node 002] Security bootstrapping failed, the node is included insecurely: a secure inclusion timer has elapsed`

**Device information**

Which device(s) is/are affected (make/model)?  
Z-Stick Gen5 Plus (ZW090-B)
- [Tech Spec](https://aeotec.freshdesk.com/support/solutions/articles/6000242203-z-stick-gen5-technical-specifications)
- Has the same model number as the "old" Z-Stick Gen5...
- Firmware 1.10

Nano Switch with Energy Monitoring (ZW116)
- [Tech Spec](https://aeotec.freshdesk.com/support/solutions/articles/6000165731-nano-switch-technical-specifications-)
- Firmware 3.00

What are the node IDs?
Z-Stick: 1
Nano Switch: 2

**Last Known Working Configuration**
- [*] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] Zwavejs2mqtt (latest) docker image
- [*] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
Steps to reproduce the behavior:
1. In ZWaveJS2MQTT, go to 'Control Panel'
2. Click on 'Actions', and then choose 'Start Inclusion'. Click 'Send'
3. In the 'Node Inclusion' popup, choose 'Yes' to 'Start inclusion in secure mode?'
4. Go to the Nano Switch, and click the Action Button twice within 1 second, as per the user guide.
5. Return to ZWaveJS2MQTT, node will appear and then interview as insecure.

**Additional context**
I get the feeling that the "fallback" to insecure inclusion may be a patch/behaviour change introduced recently to work around the secure inclusion issue I was originally having, but I'm not sure. I'm hoping there is still useful info within the logs to help point to the cause of the failure identifying the switch when included securely.

These logs are from immediately after factory reseting both the Nano Switch and the Z-Stick today, so a nice and clean setup with very little else going on. I removed the previous 'store' folder prior to starting up the docker again.

The security key was generated by the ZWaveJS2MQTT interface.

**Logfile:** 

See lines 238 onwards - zwavejs_1.log:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5853482/zwavejs_1.log)

Network Cache Files:
[zwavejs2mqtt-Network Cache Files.zip](https://github.com/zwave-js/node-zwave-js/files/5853593/zwavejs2mqtt-Network.Cache.Files.zip)

Doesn't seem to be anything useful in here but just in case - zwavejs2mqtt.log
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5853486/zwavejs2mqtt.log)

AlCalzone, 2021-01-22:
> I get the feeling that the "fallback" to insecure inclusion may be a patch/behaviour change introduced recently to work around the secure inclusion issue I was originally having, but I'm not sure

No, that is mandated by the specifications and has been this way for a [while](https://github.com/zwave-js/node-zwave-js/pull/1003) (since I added the ability to time out transactions).
[image]
> The Network Key MUST NOT be sent to the new node if a Security Scheme Report Command is received
by the including controller later than 10 seconds after successful inclusion of the node. [...] The S0 bootstrapping
process MUST be terminated if any message times out.

The log message is correct - the `SecurityCCSchemeGet` message does not get the expected response within 10 seconds and thus the secure bootstrapping process gets aborted.
```
03:38:00.470 SERIAL » 0x010a0013020398040025035d                                          (12 bytes)
03:38:00.473 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      3
                      └─[SecurityCCSchemeGet]
03:38:00.474 CNTRLR   the inclusion process was stopped
03:38:00.483 SERIAL « [ACK]                                                                   (0x06)
03:38:00.487 SERIAL « 0x0104011301e8                                                       (6 bytes)
03:38:00.489 SERIAL » [ACK]                                                                   (0x06)
03:38:00.491 DRIVER « [RES] [SendData]
                        was sent: true
03:38:00.507 SERIAL « 0x011800130300000200db7f7f7f7f000004000000000301000028              (26 bytes)
03:38:00.509 SERIAL » [ACK]                                                                   (0x06)
03:38:00.512 DRIVER « [REQ] [SendData]
                        callback id:     3
                        transmit status: OK
03:38:10.475 CNTRLR   [Node 002] Security bootstrapping failed, the node is included insecurely: a s
                      ecure inclusion timer has elapsed.
```

closed 2021-01-22 by AlCalzone

FreelancerJay, 2021-01-22:
Ah ok, that makes sense then. I don’t suppose you saw any sign of why the timeout happened? Like was a invalid responds received and discarded, or did the response turn up later than allowed?

thanks for your time already!

AlCalzone, 2021-01-22:
If there was an invalid response, it would have shown up in the log. There was just no response at all.