zwave-js/zwave-js #1478: [bug] Lock Current Status not updated

snagytx, 2021-01-22:
**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

BE469 - Touchscreen Deadbolt (Node 20)
BE469ZP - Touchscreen Deadbolt Z-Wave Plus (Node 21)

When the door lock for Node 20 is manually locker/unlocked I can see notifications like "Manual lock/unlock operation" but the state of the lock does not change.

When the door lock for Node 21 is manually locker/unlocked I can see notifications like "Manual lock/unlock operation" AND the state of the lock does changes as expected.

**Device information**

Which device(s) is/are affected (make/model)?  
**BE469 - Touchscreen Deadbolt (Node 20)**

What are the node IDs?
**Node 20**

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [x] Previously working device (other platform)
    - Which platform? Vera
    - Have you made any recent configuration changes to the device? Describe.
**Moved to zwave2mqtt & hass**

**Installation information**
How did you install `node-zwave-js`?
- [ ] Zwavejs2mqtt (latest) docker image
- [x] Zwavejs2mqtt (dev) docker image
**zwavejs/zwavejs2mqtt                          dev                 9678962874af        3 hours ago         314MB**
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
Steps to reproduce the behavior:
'The door lock and lock it from the inside. Then check the zwave2mqtt logs and note that the only message is:
`2021-01-22 13:38:16.160 INFO ZWAVE: Node 20: notification: Manual lock operation`

If you do the same for Node 21 you can see:
```
2021-01-22 13:38:01.226 INFO ZWAVE: Node 21: notification: Manual lock operation 
2021-01-22 13:38:01.387 INFO ZWAVE: Node 21: value updated: 98-0-currentMode 0 => 255
2021-01-22 13:38:01.390 INFO ZWAVE: Node 21: value updated: 98-0-outsideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-01-22 13:38:01.393 INFO ZWAVE: Node 21: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-01-22 13:38:01.395 INFO ZWAVE: Node 21: value updated: 98-0-latchStatus open => open
2021-01-22 13:38:01.397 INFO ZWAVE: Node 21: value updated: 98-0-boltStatus unlocked => locked
2021-01-22 13:38:01.399 INFO ZWAVE: Node 21: value updated: 98-0-doorStatus open => open
```

**Additional context**
Add any other context about the problem here.
Relevant part extracted from zwavejs_1.log for the not working node, Node 20:
```
13:38:16.042 SERIAL « 0x010a00040014029840b80087                                          (12 bytes)
13:38:16.044 SERIAL » [ACK]                                                                   (0x06)
13:38:16.045 DRIVER « [Node 020] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
13:38:16.059 SERIAL » 0x01110013140a98803e95e48b36d235e5058886                            (19 bytes)
13:38:16.060 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x05
[... 80 lines cut ...]
13:38:01.380 SERIAL « 0x0123000400151b9881cdb4f6eed2a4a0511cb4bc089814cbfb7d1b2fcc6a0849c (37 bytes)
                      38cc400ac
13:38:01.381 SERIAL » [ACK]                                                                   (0x06)
13:38:01.385 CNTRLR   [Node 021] [~] [Door Lock] currentMode: 0 => 255                  [Endpoint 0]
13:38:01.389 CNTRLR   [Node 021] [~] [Door Lock] outsideHandlesCanOpenDoor: false,false [Endpoint 0]
                      ,false,false => false,false,false,false
13:38:01.392 CNTRLR   [Node 021] [~] [Door Lock] insideHandlesCanOpenDoor: false,false, [Endpoint 0]
                      false,false => false,false,false,false
13:38:01.394 CNTRLR   [Node 021] [~] [Door Lock] latchStatus: "open" => "open"          [Endpoint 0]
13:38:01.396 CNTRLR   [Node 021] [~] [Door Lock] boltStatus: "unlocked" => "locked"     [Endpoint 0]
13:38:01.398 CNTRLR   [Node 021] [~] [Door Lock] doorStatus: "open" => "open"           [Endpoint 0]
13:38:01.400 DRIVER « [Node 021] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[DoorLockCCOperationReport]
                            current mode:           Secured
                            active outside handles: false, false, false, false
                            active inside handles:  false, false, false, false
                            latch status:           open
                            bolt status:            locked
                            door status:            open
```

**Logfile:** 
zwavejs and zwave2mqtt logs

snagytx, 2021-01-22:
I have tried to attached the logs, let me know if they don't show up. trying again in the comment
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5856285/zwavejs_1.log)
[zwave2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5856289/zwave2mqtt.log)

AlCalzone, 2021-01-22:
I can't say for sure, but I guess (based on your logs and because I can't find anything in the manual) that Node 20 (which does not support Z-Wave+) expects us to poll for the current status. 

I'm still in the process of adding an API for that which can be used over MQTT. Once that is in, you'll have to add some kind of automation in HA to poll the lock status when the `Manual lock operation` notification is received.

snagytx, 2021-01-22:
Do you think that it would make sense to monitor the /nodeID_20/nodeID_20/notification/Manual_lock_operation and re-trigger a set lock call? Would this cause an issue if the door is locked and I send a lock request again?

AlCalzone, 2021-01-22:
Yes and no. I would issue a `get` call, but that isn't possible yet.

snagytx, 2021-01-22:
I was thinking to do a set call, since the notification tells me what it needs to be :)

AlCalzone, 2021-01-22:
Thats a dirty workaround, but should do the trick for now.

snagytx, 2021-01-23:
In case somebody cares:

I order to tell Home Assistant to use MQTT Device Trigger I had to post to the following topic:
`homeassistant/device_automation/nodeID_20/Manual_lock_operation/config`
The following payload
```
{
  "automation_type": "trigger",
  "topic": "zwave/nodeID_20/notification/Manual_lock_operation",
  "type": "button_short_press",
  "subtype": "button_1",
  "device": {
    "identifiers": [
      "zwavejs2mqtt_0xf4c9e959_node20"
    ],
    "manufacturer": "Allegion",
    "model": "Touchscreen Deadbolt (BE469)",
    "name": "nodeID_20",
    "sw_version": "88.21"
  }
}
```

The above would automatically create a trigger for "Back Door Lock" called '"First Button" pressed'

Then in an automation I could use the following automation(I used the Automation UI to select for ticker the "Trigger Type" as  "Device" and "Action Type" as "Device") - the following YAML was auto generated:

```
- id: '1611336587134'
  alias: 3 - Back Door Lock
  description: ''
  trigger:
  - platform: device
    domain: mqtt
    device_id: d864eb9d8e0e99f9f9fff9740d232e4d
    type: button_short_press
    subtype: button_1
    discovery_id: nodeID_20 Manual_lock_operation
  condition: []
  action:
  - device_id: d864eb9d8e0e99f9f9fff9740d232e4d
    domain: lock
    entity_id: lock.nodeid_20_lock
    type: lock
  mode: single
```

I'll close the issue considering the above as the solution, even though it's more of a work around.

closed 2021-01-23 by snagytx