zwave-js/zwave-js #1481: [question] SetOutdoorTemp for Stelpro Thermostat?

carldebilly, 2021-01-23:
## Introduction
I have a bunch of Stelpro Ki thermostats that I installed few years ago when I was using SmartThings. Those thermostats were always working great.

## The feature: display outdoor temperature
The have a feature to show outdoor temperature that works well with SmartThings [the code for that is here](https://github.com/stelpro/Ki-Z-Wave-Thermostat/blob/d106ec8c640d3288850e672bf88352beff6f57c7/devicetypes/stelpro/stelpro-ki-thermostat.src/stelpro-ki-thermostat.groovy#L359_L371).

## The failure
I switched to Home Assistant (which is using OpenZWave) and I never found how to use this feature.

## Specs
On SmartThings, it's using the [`sensorMultilevelV3.sensorMultilevelReport`](https://graph.api.smartthings.com/ide/doc/zwave-utils.html#sensorMultilevelV3) to send this information to the device.

[spec sheet on z-wave alliance](https://products.z-wavealliance.org/products/1603):
```
Outdoor Temperature

When connected to a Z-Wave network, the thermostat may display the outdoor temperature provided by an external sensor.
When available, the outdoor temperature is displayed instead of the set point, providing that the outdoor temperature display setting is set to on.
If no temperature is received within a 4 hours timeframe, the outdoor temperature will disappear and the set point will be displayed.
In order to display the outdoor temperature, an external sensor needs to be part of the Z-Wave network. It may be a physical sensor or a weather station app.
Refer to your Z-Wave hub controller instructions manual to find how to associate both devices. On some Z-Wave hub controllers, an app may be used to provide the outdoor temperature.

```

## The hope
Is it possible to code something like this using zwavejs/zwavejs2mqtt ? I'm not afraid to code it (I did the exact same feature for the zigbee version of this thermostat on the Zigbee2Mqtt project).

I'll need some guidance...

robertsLando, 2021-01-23:
@zwave-js-bot import config from ozw with device id 0x0239-0x0001-0x0001

AlCalzone, 2021-01-23:
@robertsLando what needs to happen here is to send a Multilevel Sensor Report to the device like @carldebilly wrote in the first post. It is already possible to do that in zwave-js, but you'll need to expose an API in zwavejs2mqtt for that.

Maybe it could be done similar to what I do in ioBroker.zwave2: https://github.com/AlCalzone/ioBroker.zwave2/blob/master/docs/en/sendCommand.md
Basically, you need to specify the node ID, Command Class Name, API method name and the API method args. Reference implementation (which takes the object from the example) here:
https://github.com/AlCalzone/ioBroker.zwave2/blob/d399cd1be91e8717b0508b0132f5f8dc57184c54/src/main.ts#L1286-L1377