zwave-js/zwave-js #1485: MeterCC.ts creates value labels that cannot be distinguished

ahochsteger, 2021-01-24:
**Describe the bug**

When selecting a meter value in zwavejs2mqtt (e.g. from a Shenzehen Neo NAS-WR01ZE Wall Plug) the selection box shows identical labels for all meter values which cannot be distinguished for which unit it is (kWh, W, V, A).

This is a screenshot from the value selection:
[image]

This is a screenshot that shows that multiple values have identical labels and would not be able to distinguish, if the UI would not output the unit as well:
[image]

Maybe the value labels could be improved a bit as well (shorter with more meaningful names).

**Device information**

Which device(s) is/are affected (make/model)?  Shenzehen Neo NAS-WR01ZE Wall Plug
What are the node IDs? Node 080 (see logs). Device IDs: 600-4135-512 (0x0258-0x1027-0x0200)

The bug affects not only this device but all devices using the meter CC.

**Installation information**
How did you install `node-zwave-js`? Zwavejs2mqtt (dev) docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Go to zwavejs2mqtt settings
2. Click on 'New Value' on the bottom left
3. Select a device supporting the meter CC
4. Open the value select - it shows multiple identically named values, unable to distinguish

**Logfile:**
[zwavejs-log.zip](https://github.com/zwave-js/node-zwave-js/files/5862252/zwavejs-log.zip)

AlCalzone, 2021-01-24:
> would not be able to distinguish, if the UI would not output the unit as well

Adding the unit to the label would be redundant information though.

> Maybe the value labels could be improved a bit as well (shorter with more meaningful names).

What labels would you propose?

ahochsteger, 2021-01-24:
> > Maybe the value labels could be improved a bit as well (shorter with more meaningful names).
> 
> What labels would you propose?

This is for example how OpenHAB names the labels which have always worked well for me:
[image]
See here for more example screenshots: https://community.openhab.org/t/aeotec-power-meter-setup-with-openhab-issues/72932/5

The labels are taken from the device XML files - here for example:
https://github.com/openhab/org.openhab.binding.zwave/blob/9c16de62e85f5c96ba8a287776a4db795d222ad2/src/main/resources/OH-INF/thing/shenzhen/coolcampwrplug_0_0.xml

AlCalzone, 2021-01-24:
They are missing half of what zwave-js supports though.
According to Qubino, we're the first software they tested to display the **rate type** (production, consumption). That is not included in the OpenHAB labels. Also I don't see anything for previous values or the delta time between two measurements in the screenshot.

Meter values are very well standardized, so I see no point in adding the labels in config files. We just need to find a decent format for them. To summarize

1. We have these types and scales (= unit), which includes duplicated scales between the meter types (e.g. kWh)

| Meter type | Meter scales |
| ---------- | ------------ |
| Electric   | kWh          |
|            | kVAh         |
|            | W            |
|            | Pulse count  |
|            | V            |
|            | A            |
|            | Power Factor |
|            | kVar         |
|            | kVarh        |
|            |              |
| Gas        | Cubic meters |
|            | Cubic feet   |
|            | Pulse count  |
|            |              |
| Water      | Cubic meters |
|            | Cubic feet   |
|            | US gallons   |
|            | Pulse count  |
|            |              |
| Heating    | kWh          |
|            |              |
| Cooling    | kWh          |

2. We have two rate types: produced and consumed, which can be **freely** combined with the above
3. And we have three values for each possible combination: `value`, `previousValue` and `deltaTime`.

If you have a good naming scheme that encompasses all of this, I'm all ears.

ahochsteger, 2021-01-24:
I see.

What about the following suggestion for a generic label generation:
```
value: <meterType> <rateType> [<meterScale>]
previousValue: <meterType> <rateType> [<meterScale>] (previous)
deltaTime: <meterType> <rateType> [<meterScale>] (previous time delta)
```
This would generate the following labels:
```
value: Electric consumed [kWh]
previousValue: Electric consumed [kWh] (previous)
deltaTime: Electric consumed [kWh] (previous time delta)
```

AlCalzone, 2021-01-24:
Not exactly much shorter, but unique

PR #1515 (2021-01-29): fix(cc): improve value labels for Meter CC
fixes: #1485
closed 2021-01-29 by AlCalzone