zwave-js/zwave-js #1488: TODO's in log when controller is queried from other controllers

LordMike, 2021-01-24:
I'm not sure if this is (_meant_) to be supported.. But here goes.

**Describe the bug**
If I open up Z-Wave pc controller on another device, and query Node 1, the controller that zwave-js is running, the zwave-js logs the following:

When fetching configuration params:
```
18:47:31.208 SERIAL « 0x010a0004003004700e0001ba                                          (12 bytes)
18:47:31.208 DRIVER   Dropping message because it could not be deserialized: ConfigurationCCProperti
                      esGet: deserialization not implemented
18:47:31.208 SERIAL » [ACK]                                                                   (0x06)
```

When fetching the firmware version:
```
18:38:09.739 SERIAL « 0x0109000400300386137a2e                                            (11 bytes)
18:38:09.741 DRIVER   Dropping message because it could not be deserialized: VersionCCCommandClassGe
                      t: deserialization not implemented
18:38:09.741 SERIAL » [ACK]                                                                   (0x06)
18:38:16.524 SERIAL « 0x0109000400300386137a2e                                            (11 bytes)
18:38:16.525 DRIVER   Dropping message because it could not be deserialized: VersionCCCommandClassGe
                      t: deserialization not implemented
18:38:16.525 SERIAL » [ACK]                                                                   (0x06)
18:38:58.590 SERIAL « 0x010800040030027a01ba                                              (10 bytes)
18:38:58.591 SERIAL » [ACK]                                                                   (0x06)
18:38:58.592 DRIVER « [Node 048] [REQ] [ApplicationCommand]
                      └─[FirmwareUpdateMetaDataCCMetaDataGet]
18:38:58.594 CNTRLR « [Node 048] TODO: no handler for application command
```

**Device information**
Which device(s) is/are affected (make/model)?  
[Aeotec Z-Wave stick, Gen5](https://aeotec.freshdesk.com/support/solutions/articles/6000242203).

What are the node IDs? 
48 for my other controller.

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] Zwavejs2mqtt (latest) docker image
- [x] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
N/A

**Additional context**
N/A

**Logfile:**
See above.

AlCalzone, 2021-01-24:
Yeah that is at the bottom of the list. Almost all commands that are only sent in the default single controller setup can't be decoded yet

LordMike, 2021-01-25:
I found it odd that these messages would make their way to the controller application.. shouldn't they have been handled by the node? :|

If sending a message to node 1, from node 1 itself.. means it comes back to you over the serial interface as you sent it - then I can better understand why interviewing the controller itself has been skipped entirely :O

LordMike, 2021-01-25:
My question is answered, so you can go ahead and close this issue.

AlCalzone, 2021-01-25:
If I understand correctly, while a controller is controlled via Serial API, it is "dumb". It handles the protocol-level stuff, but responding to Command Classes must be handled by the application itself.
Maybe @hanskroner has some more insight here?

LordMike, 2021-01-25:
It's cool and all - makes for more advanced libraries like yours.. but damn.. It poses some new issues ... :)

hanskroner, 2021-01-27:
> If I understand correctly, while a controller is controlled via Serial API, it is "dumb". It handles the protocol-level stuff, but responding to Command Classes must be handled by the application itself.

That's right. The Z-Wave protocol, which runs on the Z-Wave chip, handles the PHY, MAC, and Networking layers - but it leaves the Application layer (i.e., Command Classes) up to the serial host to handle.

The host application, _zwave-js_ in this case, is responsible for setting itself up correctly. It's expected to do things like build a NIF with the list of Command Classes it supports and answer requests from other nodes for those Command Classes, as well as those that are mandated by Z-Wave Certification - just like any other Z-Wave device. Besides being Certification-compliant, it'd be good to get this implemented to ensure devices in the network take advantage of all the capabilities available. For example,I know of devices that ask the Lifeline for its NIF to determine whether it can handle CRC16-encapsulated messages - which come in handy when sending very long frames (like Meter Table data) that are prone to corruption. If the Controller doesn't advertise this Command Class in the NIF, the device won't encapsulate the messages.

Controllers also have the possibility of having Association Groups, Configuration Parameters, and whatever else they want to support. OTA firmware update can also be supported, and can even go as far as being used to update the host itself (not just the Z-Wave firmware) in scenarios where that is practical 😉

AlCalzone, 2021-01-27:
> It's expected to do things like build a NIF with the list of Command Classes it supports and answer requests from other nodes for those Command Classes

Is this different from the `ApplicationNodeInformation` API method mentioned in INS13954?
> Note for Controllers: Because controller libraries store some basic information about themselves from
ApplicationNodeInformation in nonvolatile memory. ApplicationNodeInformation should be set to the
correct values before Application return from ApplicationInitHW(), for applications where this cannot be
done. **The Application must call ZW_SetDefault() after updating ApplicationNodeInformation in order
to force the Z-Wave library to store the correct values.**

I thought about doing that a while back but it seems impractical to hard-reset the controller every time the software is updated.

hanskroner, 2021-01-28:
> Is this different from the `ApplicationNodeInformation` API method mentioned in INS13954?

That's the call that's used for setting up the NIF. There's a few additional calls you need to be looking at for setting the supported Command Classes across non-secure and secure domains, as well as S2 key levels.

> I thought about doing that a while back but it seems impractical to hard-reset the controller every time the software is updated.

The  `ApplicationInitHW()` call is only available within the embedded SDK, so don't worry about that. Remember that the NIF for a node needs to be static for as long as that node is part of a network. In the case of the controller starting a network, if it's NIF changes, the whole network needs to be re-created.

Take for example the current situation _zwave-js_ finds itself in. It supports S0, but not S2. Clients including devices that support S2 are not able to take advantage of it, and devices are being included with S0 only. A few weeks from now, an update to _zwave-js_ adds S2 support. In order to change the controller's NIF, and add S2 as a supported Command Class, you **must** call `ZW_SetDefault()` - effectively orphaning every node in the network. All the devices need to be reset and re-included into the controller. In that process, devices that support S2 will now be included with S2 security. Probably a really important item that you want to highlight in the release notes for the version that add support for this 😉

AlCalzone, 2021-01-28:
> A few weeks from now, an update to zwave-js adds S2 support. In order to change the controller's NIF, and add S2 as a supported Command Class, you must (according to the documentation) call ZW_SetDefault() - effectively orphaning every node in the network. All the devices need to be reset and re-included into the controller.

That doesn't sound like people will like it. Resetting and re-including hundreds of devices that are hidden in a wall 😨

> Not resetting the controller would mean that only new devices joining could use S2, and existing devices will remain at S0.

I don't think this is terrible. We can release a new version, and tell people that they have to hard-reset if they want to upgrade their entire network to S2.

Is it possible to **read** the current NIF of the controller so zwave-js can check if it even needs to be updated?

LordMike, 2021-01-28:
Wouldn't this affect _any_ newly support CC?.. As in, if zwave-js hypothetically suddenly supports multi-channel endpoints, or the CRC encapsulation, or something else - then all devices must be reincluded in order to benefit from it.. 

That could be tricky for a rapidly evolving project like this :P

@hanskroner is there no way to remotely reset a device, to have it "auto include" itself again?

hanskroner, 2021-01-29:
> Wouldn't this affect _any_ newly support CC?.. As in, if zwave-js hypothetically suddenly supports multi-channel endpoints, or the CRC encapsulation, or something else - then all devices must be reincluded in order to benefit from it..
Yes, it's true for any new Command Class.

> That could be tricky for a rapidly evolving project like this :P
There's a reason why Certification requirements for Controllers are so steep. There are no illusions about end users liking to reset their networks every so often, so the "tough" Command Classes were required to be there from the start.

> @hanskroner is there no way to remotely reset a device, to have it "auto include" itself again?
On purpose, there is no way to properly exclude a device from a network without having access to the device. Newer devices that support Smart Start will initiate automatic inclusion through it after resetting to factory defaults, which is the final step a device being excluded takes.

hanskroner, 2021-01-29:
> That doesn't sound like people will like it. Resetting and re-including hundreds of devices that are hidden in a wall 😨
Yeah, it's not fun or desired by anyone. Typically, the devices you'd want to re-included to gain S2 support will be door/window or smoke sensors - which are hard to reach. For these devices, the benefits of S2 likely outnumber the pain of finding a ladder. There's no security benefits between already included S0 devices and S2, but there are performance benefits like reduced traffic (also leading to better battery life on sensors) and the addition of Supervision Command Class support.

> I don't think this is terrible. We can release a new version, and tell people that they have to hard-reset if they want to upgrade their entire network to S2.
>
> Is it possible to **read** the current [...]
I don't think there's an API for reading this from the Z-Wave module, but since it was the host setting it up in the first place, chances are that information is stored somewhere in the host by the driver.

LordMike, 2021-01-29:
In principle, zwave-js could also note which NIF it had, at the time of inclusion of any device (alongside the device), with the goal of recommending end-users to re-include if something important comes by.. Like a device supporting S2, but being included while S2 wasn't supported.

AlCalzone, 2023-07-25:
This should be solved now as part of making the certification tests pass.

closed 2023-07-25 by AlCalzone