zwave-js/zwave-js #1493: Performance bottleneck in getDefinedValueIDs

AlCalzone, 2021-01-25:
[image]

Not sure what that `(bytecode) ~Abort.ExtraWide` is all about but we can try iterating without the `forEach` instead.

AlCalzone, 2021-01-25:
TODO:
- [ ] maintain an index `(nodeId, ccId) => keys[]`
- [ ] look into offering methods to return value and metadata along with the value IDs

robertsLando, 2021-01-26:
@AlCalzone Just curious, what's the problem here?

AlCalzone, 2021-01-26:
HA has a `dump` method which loops through all nodes and reads that node's values using `getDefinedValueIDs`.
That method in turn loops through all CCs it supports and calls that CC's `getDefinedValueIDs`.
That method calls `db.getValues(nodeId, ccId)` then filters out the internal ones.

The problem is that getValues itself loops through **all** values to filter out the matching ones and this work is being repeated for every node and every CC it supports.
So basically, this access pattern has a **terrible** complexity of `O(n⁴)` if we assume that all nodes have approximately the same number of values.

marcelveldt, 2021-01-26:
Just thinking out loud here... Maybe getDefinedValueIDs should return a list of keys (so valueID is in fact an id/key in this context) and you can call getValue or something in those lines to get the complete Value object, including the metadata. 
Value is then what you now call a translatedValueID object. But that's probably more refactoring than needed.

Main issue is that at startup (client connect) we (or any client utilizing the library) have to know the full state of the mesh. What nodes are there, what values do they have etc. So we need to fetch the state once and after that subscribe to all added/changed events to stay updated.

@robertsLando we had reports that getting the initial state took a VERY long time and we traced that down to getting the values (and their metadata).

AlCalzone, 2021-01-26:
I think I found the root issue (see PR), but I have a couple of other things in mind

marcelveldt, 2021-01-26:
Wow, that fix already looks like a huge improvement!

robertsLando, 2021-01-26:
I think that the values db should be sorted or at least the ids grouped by node id

AlCalzone, 2021-01-26:
Since a native map is used under the hood, that's not trivial.

closed 2021-01-26 by AlCalzone