zwave-js/zwave-js #1496: [bug] the label on two configuration values are transposed for Aeotec Siren Gen 5

scyto, 2021-01-26:
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version
1.0.0-rc.1
Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.0.0
zwavejs version: 6.0.0

### Describe the bug

Two parameters (siren volume and siren sound) have  transposed labels for Aeotec Siren Gen5
Both of these parameters work, the labels need swapping.

[image]

I have tested using the values in the table and proved the labels are swapped.  I don't know if this is an zwavejs2mqtt ui bug or a core zwave-js and I am embarrassed to say i couldn't find any definition for this device in github - so i am not sure where to submit the edit?

to be clear siren sound should say siren volume and vice versa.

[image]

### Additional context
nodes json
[Downloads.zip](https://github.com/zwave-js/zwavejs2mqtt/files/5871773/Downloads.zip)

Add any other context about the problem here.

robertsLando, 2021-01-26:
This is the file: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0086/zw080.json

AlCalzone, 2021-01-26:
Yup, the bitmask is switched.
@blhoward2 any chance that is already included in your work?

scyto, 2021-01-26:
i just forked and swapped the bit around already, want me to submit a PR or hold off?
i am not sure how to test in private (as i use the prebuilt docker container)

(i had to think real hard to figure that was the issue and you had mentioned it here all while i was deep in thought :-) funny)

scyto, 2021-01-26:
its midnight here (PST) so i created the PR just incase https://github.com/zwave-js/node-zwave-js/pull/1497

AlCalzone, 2021-01-26:
Pr merged, thanks!

closed 2021-01-26 by AlCalzone