zwave-js/zwave-js #1499: node.getEndpoint call fails sometimes

marcelveldt, 2021-01-26:
This appears to be a race condition as it only manifests itself sometimes.

When you call node.getEndpoint, it will throw an exception at some occasions.
In the WS server we call this right after "value added" to determine the CC version of the CC belonging to the value.
99% of the times this is OK but sometimes this fails, I believe this happens when interview is still busy ?

Code in the WS server where the exception throws:
https://github.com/zwave-js/zwave-js-server/blob/master/src/lib/forward.ts#L142

Capture of the error:

`10:52:59.106 DRIVER   Node 28, Endpoint undefined: Trying to access endpoint instance before Multi Channel interview
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
TypeError: Cannot read property 'getCCVersion' of undefined
    at ZWaveNode.<anonymous> (/usr/lib/node_modules/@zwave-js/server/dist/lib/forward.js:107:56)
    at ZWaveNode.emit (events.js:315:20)
    at ZWaveNode.EventEmitter.emit (domain.js:467:12)
    at ZWaveNode.translateValueEvent (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/node/Node.ts:346:8)
    at ValueDB.emit (events.js:315:20)
    at ValueDB.EventEmitter.emit (domain.js:467:12)
    at ValueDB.setValue (/usr/lib/node_modules/@zwave-js/server/node_modules/@zwave-js/core/src/values/ValueDB.ts:224:10)
    at new ConfigurationCCReport (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:916:22)
    at new CommandClass (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/commandclass/CommandClass.ts:108:13)
    at new ConfigurationCC (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/commandclass/ConfigurationCC.ts:484:3)`

I can simply wrap the call in a try..except or first check if the endpoint is defined but that doesn't really fix the issue.

robertsLando, 2021-01-26:
@marcelveldt In zwavejs2mqtt I ignore the valueAdded event if node isn't ready. I suggest you to add all values when the node is ready

marcelveldt, 2021-01-26:
Hmmm, that is going to be complicated. As you know we run the full state dump at client connect. After that we simply forward the events. So I expect the value added to be signaled only when the value is actually added. 

Will there be another event when the value is actually useable/complete ?

marcelveldt, 2021-01-26:
OK, so this is related to our other issue with the full state dump performance.
@balloob maybe we should split things up a bit and make "dumpNode" a separate call.
Once we receive "node ready" we'll issue a "dumpNode" call to the WS server to retrieve the full state of the node.
At startup/connect we can do the same for all ready nodes.

So proposed workflow:
client connect --> send ws server version
client issues "start_listening" --> send back nodes listing (without values) and start forwarding events
for nodes the client is interested in (e.g. the ready nodes), the client can issue "get_node_state" to get full node dump.

when the client receives a new "node_ready" command it can ask for the dump.
at "node_added" only the (incomplete) node info will be processed

robertsLando, 2021-01-26:
> So I expect the value added to be signaled only when the value is actually added.

That's correct, maybe this could be fixed by moving some CC interviews before others, like the Multi Channel interview. Anyway couldn't you fetch the suppported ccs/metadata only after the node is ready?

marcelveldt, 2021-01-26:
> > So I expect the value added to be signaled only when the value is actually added.
> 
> That's correct, maybe this could be fixed by moving some CC interviews before others, like the Multi Channel interview. Anyway couldn't you fetch the suppported ccs/metadata only after the node is ready?

Yeah, that's where my comment above is all about. Nevertheless it is strange that the value added event is issued when things are not in ready state yet.

robertsLando, 2021-01-26:
> Yeah, that's where my comment above is all about. Nevertheless it is strange that the value added event is issued when things are not in ready state yet.

If value is added it doesn't mean the node is ready

marcelveldt, 2021-01-26:
> > Yeah, that's where my comment above is all about. Nevertheless it is strange that the value added event is issued when things are not in ready state yet.
> 
> If value is added it doesn't mean the node is ready

That is not what I mean. The Value is not yet useable. It throws an exception when you want to grab the endpoint. That should not happen.

robertsLando, 2021-01-26:
>  It throws an exception when you want to grab the endpoint. That should not happen.

I know what you mean but the endpoint informations are fetched from another CC, so you cannot be 100% sure that that CC is already be interviewed at that point.

robertsLando, 2021-01-26:
@marcelveldt I got your same issues when starting with zwave-js but if you get how it works this makes sense

AlCalzone, 2021-01-26:
Yeah, the problem here is that the node is sending a multi channel encapsulated command (from an endpoint), which zwave-js emits as a value event, so it doesn't get missed.

However at that point the `Multi Channel CC` interview was not completed yet, so `zwave-js` does not have any information about endpoints yet. You're trying to read the CC version of that endpoint which is also not known yet at that point, since the endpoint interview happens **after** the `Multi Channel CC` interview of the node.

Long story short: The type declarations reflect the error too.
[image]

Workaround (I assume your TS version is high enough to support optional chaining):
```diff
          args.ccVersion =
-            node.getEndpoint(args.endpoint).getCCVersion(args.commandClass) ||
-            node.getEndpoint(0).getCCVersion(args.commandClass);
+            node.getEndpoint(args.endpoint)?.getCCVersion(args.commandClass) ||
+            node.getCCVersion(args.commandClass);
```
`getEndpoint(0)` is also a no-op, it just returns the node instance itself.

marcelveldt, 2021-01-26:
Thanks for all the help and insight again.
We're going the safer route and only request all these metadata after the node is ready.

closed 2021-01-26 by marcelveldt