zwave-js/zwave-js #1500: [bug] How to create new devices

PoulKlemmensen, 2021-01-26:
Hi.

I have a question on how to create new devices. I have a new device, not yet created and therefore appears as unknown. How and where to start looking?

We are talking about a complete new product from Logic Group. In the tool called: Device ID: 564-293-3 (0x0234-0x0125-0x0003)

robertsLando, 2021-01-26:
@zwave-js-bot import config 0x0234-0x0125-0x0003 from ozw

AlCalzone, 2021-01-26:
See docs: https://zwave-js.github.io/node-zwave-js/#/development/config-files
But I'd check with @blhoward2 who is busy importing hundreds (thousands?) new devices.

robertsLando, 2021-01-26:
Let's see the bot output for now :)

robertsLando, 2021-01-26:
> x Sorry, importing the files yielded no changes.

LOL, I think that device already exists and the Unknown is due to the error whe spoke in the other issue

AlCalzone, 2021-01-26:
No, we don't have it. OZW doesn't seem to have it either. Lets wait for @blhoward2 before we start digging.

PoulKlemmensen, 2021-01-26:
The device is brand new, but I have some prelim. information from Logic Group, that might be usefull. I can email it to you if necessary.

blhoward2, 2021-01-26:
I think we’re down to hundreds left. What is the actual device, do you know?

PoulKlemmensen, 2021-01-26:
It is similar to ZDB5100, but with two building relays for on-off control.  And I have gotten the following description of the configuration parameters : 

/*01*/ {"Relay1 key",          FORMAT_BIT_FIELD, SIZE_8BIT, 0, 0x0F, 0x01},     // bitfield of keys that controls the relay1 output // Btn 1 on/off
/*02*/ {"Relay2 key",          FORMAT_BIT_FIELD, SIZE_8BIT, 0, 0x0F, 0x02},     // bitfield of keys that controls the relay2 output // Btn 2 on/off
/*03*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0}, 
/*04*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},             
/*05*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},  
/*06*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0}, 
 
/*07*/ {"Central Scene",       FORMAT_ENUM, SIZE_8BIT, 0, 1, 1},                // Central Scene notifications
/*08*/ {"Double click option", FORMAT_ENUM, SIZE_8BIT, 0, 1, 1},                // Cleaning Mode
/*09*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},  // NOT USED
/*10*/ {"Enhanced color ctrl", FORMAT_ENUM, SIZE_8BIT, 0, 1, 0},                // Warm white field is used for enhanced color commands
/*11*/ {"Key debounce",        FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 5},
/*12*/ {"Press threshold",     FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 20},  // 20 == 200 milliseconds
/*13*/ {"Hold threshold",      FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 50},  // 50 == 500 milliseconds
/*14*/ {"Global brightness",   FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0xFFFFFF00},
/*15*/ {"AG security mode",    FORMAT_BIT_FIELD, SIZE_16BIT, 0, 0x1FFF, 0x1FFF},
 
/*16*/ {"Key 1 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*17*/ {"Key 1 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*18*/ {"Key 1 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*19*/ {"Key 1 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*20*/ {"LED 1 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*21*/ {"LED 1 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*22*/ {"LED 1 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*23*/ {"LED 1 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*24*/ {"Key 2 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*25*/ {"Key 2 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*26*/ {"Key 2 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*27*/ {"Key 2 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*28*/ {"LED 2 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*29*/ {"LED 2 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*30*/ {"LED 2 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*31*/ {"LED 2 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*32*/ {"Key 3 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*33*/ {"Key 3 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*34*/ {"Key 3 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*35*/ {"Key 3 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*36*/ {"LED 3 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*37*/ {"LED 3 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*38*/ {"LED 3 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*39*/ {"LED 3 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*40*/ {"Key 4 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*41*/ {"Key 4 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*42*/ {"Key 4 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*43*/ {"Key 4 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*44*/ {"LED 4 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*45*/ {"LED 4 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*46*/ {"LED 4 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*47*/ {"LED 4 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00}

blhoward2, 2021-01-26:
That’s a lot and I’m not sure you have enough to do it from that. That said, I compared a few and they seemed to line up. Compare them to the 5100: https://products.z-wavealliance.org/products/3399/configs

If they’re the same I’d see if we or openHab/ozw have that one to start from. If they don’t and you can wait a week or so we’ll likely have one for the 5100. I haven’t turned to the bitmask ones yet.

blhoward2, 2021-01-26:
@AlCalzone can I invoke the bot on here? If not, can you try: Product Type ID: 0x0003
Product ID: 0x0121

robertsLando, 2021-01-26:
@blhoward2 YOu are missing the manufacturer

robertsLando, 2021-01-26:
@zwave-js-bot import config 0x0234-0x0121-0x0003 from ozw

robertsLando, 2021-01-26:
There are a tons of bitmask values there

blhoward2, 2021-01-26:
That’s why I said I wouldn’t want to do it by hand lol. @PoulKlemmensen see how that lines up. If it’s close change the device id info and start from that file.

PoulKlemmensen, 2021-01-27:
I have found the metadata file, where I have copied the most important lines. That will do for now, I guess. I am quite new to the zwavejs2mqtt solution. Need to read the complete guide through first I guess. Been struggling selecting the appropriate solution, HA, Openhab or pure zwavejs2mqtt and Node red.... Think I am going for the last one.

Madj42, 2021-01-28:
@blhoward2 any chance you have inovelli's nzw30, nzw31, and lzw60 on your list of devices to import?  I noticed these were not recognized when I started using the new zwave-js integration on HA.  Pretty sure it's using 6.1.0.   If you want me to post a new issue, I will.

blhoward2, 2021-01-28:
@Madj42 The first two are already supported. The last two are still in my queue.

Madj42, 2021-01-28:
Yeah, mine are the weird ones that are the 30T/31T (Toggle) produts.  Majority of these work but figured I'd ask before submitting a device config for them if it's already being worked on.  Thanks for all your efforts.  If there is something I can help with, let me know.

DanielWinks, 2021-01-28:
To add on to @Madj42 request, @blhoward2:
Seems that the Inovelli Ilumin is missing too, LZW42. 

Device ID: 798-1-5 (0x031e-0x0001-0x0005)

DanielWinks, 2021-01-28:
@zwave-js-bot import config 0x031e-0x0001-0x0005 from ozw

AlCalzone, 2021-01-28:
@zwave-js-bot import config 0x031e-0x0001-0x0005 from ozw

closed 2021-01-29 by AlCalzone

matheus2308, 2021-02-04:
Hi all, quick question, I see that in this thread there was a mention that there were plans to add the sensor LZW60 (from Inovelli), is this still on plans?

The ID for that sensor would be:
0x031e-0x0001-0x000d

blhoward2, 2021-02-04:
Yes. They're intermixed in a much larger import of devices so it's taking some time to get through them. I'm hoping to have them merged in the next week or two.

Madj42, 2021-02-06:
Thanks for your work blhoward.  One thing though, are goy able to confirm it is indeed the nzw30 and 31 devices from  inovelli that are already supported?  Reason I ask is that I have been looking around at the code and found that it was the lzw series that already has configs.  The nzw series from inovelli was their first gen models of switches.  I also checked this against the list that was recently posted and they are not in there.

blhoward2, 2021-02-06:
The Inovelli NZW30/NZW21 were manufactured by Willis Electric for Inovelli. They're under folder 0x015d.

Madj42, 2021-02-06:
Wow, I had no idea.  I feel like an idiot.  Thanks for the quick response.

blhoward2, 2021-02-06:
It's confusing. Inovelli has also forked this repo to presumably submit their own device configs as well so they should be even better supported soon.

PR #1520 (2021-01-29): feat(config): add support for innovelli LZW42
fixes: #1500

**TODO:**
- [x]	Change PR title to be more specific