zwave-js/zwave-js #1504: Aeotec Multisensor 6 incorrectly becomes Asleep when USB powered

maivorbim, 2021-01-26:
Aeotec Multisensor 6 (ZW100) can be either battery or USB powered. When it is USB powered, it should always be awake as per:
[https://aeotec.freshdesk.com/support/solutions/articles/6000115071-difference-between-usb-and-battery-power-and-recommendations-multisensor-6-zw100-](url)

However, the device appears as Asleep in Zwavejs2mqtt control panel. This causes delays when changing zwave configuration parameters, as you have to manually wake up the device via action button to reflect any changes. Expected behaviour when USB powered is for the device to appear as Alive.

**Device information**
Aeotec Multisensor 6 (ZW100) 
Node id 35

Previously working device - Home Assistant 2021.1.5 with default zwave integration allowed to change settings without needing to manually wakeup the device first. The device also worked as a repeater.

**Installation information**
Portainer installation 1.0.0-rc.1 ZwaveJS 6.0.2

**Additional context**
[image]
[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/5876448/zwavejs.log)

Nixon506E, 2021-02-04:
Hoping to reopen this issue except with the Aeotec Water Sensor 6 using the dock and connected to USB power. I am seeing that it is showing as asleep even when connected just like the Multisensors were.

AlCalzone, 2021-02-04:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

While the log is running, re-interview the node (refreshInfo) or however that is called on zwavejs2mqtt.

Nixon506E, 2021-02-04:
Pretty long and I have a couple of reinterviews in this log but node 96 is the water sensor that is showing as asleep.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5927932/zwavejs_1.log)

AlCalzone, 2021-02-04:
@Nixon506E according to the log, Node 96 thinks it can sleep:
```
17:08:46.713 CNTRLR « [Node 096] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Notification Sensor
                      specific device class: Notification Sensor
                      is a listening device: false <== HERE
                      is frequent listening: false <== and HERE
```

Could it be that you included it on battery power and then switched to USB?

Nixon506E, 2021-02-04:
I didnt think I did but I’ll try a repair and see if that solves it

Nixon506E, 2021-02-04:
> I didnt think I did but I’ll try a repair and see if that solves it

I tried multiple times and always got the node to report as sleepy. I am seeing that the configuration value that is read only for powersource is showing 2 which should be bit 0 reading usb power and bit 1 reading as not sleepy so somewhere it is being read incorrectly, aeotec has been known to not follow standards all the time...

AlCalzone, 2021-02-05:
Just to be sure, did you also hard reset the node between exclusion and inclusion?

Nixon506E, 2021-02-05:
I factory reset the node, yes

AlCalzone, 2021-02-05:
can you re-interview it again and make another log?

Nixon506E, 2021-02-09:
I have tried to re-include this node multiple times now with no luck. I am getting a sleepy state even though the config values are telling me the device is USB powered. 

Maybe you can take a look at what was done in this driver for hubitat. 
https://community.hubitat.com/t/release-aeotec-water-sensor-6-w-dock/39292

It looks like that driver uses the config value so maybe they also were having issues with the interview states:
    `if (cmd.parameterNumber==9) {
            Map evt=[name:"powerSource"]
            switch (cmd.configurationValue[0]) {
                case 0:
                    evt.value="mains"
                    evt.descriptionText = "${device.displayName} USB powered"
                    if (txtEnable) log.info evt.descriptionText
                    eventProcess(evt)
                    break
                case 1:
                    evt.value="battery"
                    evt.descriptionText = "${device.displayName} Battery powered"
                    if (txtEnable) log.info evt.descriptionText
                    eventProcess(evt)
                    break
            }
            switch (cmd.configurationValue[1]) {
                case 0:
                    state.sleepy=true
                    break
                case 1:
                    state.sleepy=true
                    break
                case 2:
                    state.sleepy=false
                    break
            }
    }`

AlCalzone, 2021-02-09:
I'm not going to hard-code driver behavior based on a configuration parameter of a single device that is only available later. The protocol info `listening: true/false` is meant exactly for that, because it is available to the controller without having to communicate with the node.

I really need to see that log from your re-interview.

Nixon506E, 2021-02-09:
I definitely can understand that, like I said, aeotec has been known to not adhere to standards sometimes, I am just wondering if maybe that is what is going on and maybe you can confirm. Attached is the latest log with the re-inclusion and reinterview.

[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/5952817/zwavejs_1.2.log)

AlCalzone, 2021-02-09:
Am I correct in assuming the node got ID 101 and 102 in the meantime? If so, both think they are battery-powered and still send wakeup notifications (right until the end of the log):
```
11:34:43.484 CNTRLR « [Node 102] received wakeup notification
```

We have another issue where the opposite was true. The device thought it was USB powered and only became non-listening after a couple of resets: https://github.com/zwave-js/node-zwave-js/issues/1578#issuecomment-774773130

I'm pretty sure that this sensor works, because I have a couple of them myself - all USB powered.

Nixon506E, 2021-02-09:
Yes, it is node 101 then 102. Sounds like I am correct in assuming that aeotec did not properly setup the interview process to change based on if it is in the dock or not, only a config value.

AlCalzone, 2021-02-09:
When changing between battery and USB, you have to exclude **and** hard-reset every device that can do this. Seems like the reset didn't go through in your case 🤷🏻‍♂️

Nixon506E, 2021-02-09:
I did a successful hard reset according to the lights on the sensor every time, not sure what do think as the device knows its usb powered in the configuration params...

AlCalzone, 2021-02-09:
Your firmware is quite old, maybe that changes with an update, which is possible through zwave-js (zwavejs2mqtt) without excluding again:
https://aeotec.freshdesk.com/support/solutions/articles/6000036562-how-to-update-multisensor-6-z-wave-firmware-

AlCalzone, 2021-02-09:
Yup, this could be it:
1.07 2016/3/4 2.Fixed the issue that the **Listen Flag in the NIF will be changed after inclusion**

Nixon506E, 2021-02-09:
It is not a multisensor but rather the water sensor, so no new firmware...
https://aeotec.freshdesk.com/support/solutions/folders/6000186005

AlCalzone, 2021-02-09:
Oh damn I missed that. Might be worth asking the support what the hell is going on with that thing.

Nixon506E, 2021-02-11:
Looks like we can close this one as Aeotec officially said that the water sensor in a dock is still a sleepy device that is always awake and they don't plan on changing it. Thanks for the help, guess I should have reached out to them first.

PR #1544 (2021-01-31): fix(driver): use listening flags to check whether a node can sleep instead of Wake Up CC
fixes: #1538
should help with #1536
fixes: #1504
closed 2021-02-11 by AlCalzone