zwave-js/zwave-js #1511: Better Z-wave network graph

maivorbim, 2021-01-27:
Currently Zwave-js has the option of displaying a SVG graph of all connected nodes. However, the information presented is basic and is lacking additional information such as list of neighbours for each node, latencies and topology/hierarchy.

I find that the following implementation is more clear and helps diagnose network issues faster and could serve as reference: https://github.com/AdamNaj/ZWaveGraphHA

As such, a better network graph would be a nice-to-have feature for zwave-js .

AlCalzone, 2021-01-27:
I think this belongs in https://github.com/zwave-js/zwavejs2mqtt?
Rendering the network graphs is done in the consuming applications not the driver library.

The only thing missing here is the statistics for latency, which is on the TODO list: https://github.com/zwave-js/node-zwave-js/issues/985

robertsLando, 2021-01-27:
@maivorbim that's something not related to zwavejs but zwavejs2mqtt

maivorbim, 2021-01-27:
Right, sorry, i will duplicate it in zwavejs2mqtt then.

closed 2021-01-27 by maivorbim