zwave-js/zwave-js #1517: DoorLock re-interview fails

snagytx, 2021-01-28:
**Describe the bug**

The installation is working fine, I move the R-PI that hosts my Aeoteck Z-Stick and docker zwavejs2mqtt to a different location, which is 1/2 the distance from the BE469 lock and now it has issues getting the node info, after a restart.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?
Node 020

**Last Known Working Configuration**
- [ ] New device
- [x] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version? **same library same configuration**
    - Have you made any recent configuration changes to the device? Describe. **Just moved the R-PI closer to the device - from my office where I have all the computers and monitors, on top of the fridge in the kitchen - which has direct access to the lock and it's maybe 30 feet from the device**

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] Zwavejs2mqtt (latest) docker image
- [x] Zwavejs2mqtt (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual docker build
    - Against which branch of `node-zwave-js`?
    - With which branch of `zwavejs2mqtt`?
- [ ] Other (describe)
- [ ] Manually built (`git clone` - `yarn install` - `yarn run build:full`) 

**To Reproduce**
Steps to reproduce the behavior:
1) Stop docker container
2) Shutdown OS
3) Move R-Pi to a central location
4) Start the OK,
5) Start docker container
6) Interview start stuck at "RestartFromCache", on "Re-Interview" request it changes to "NodeInfo"

**Intersting stuff from the logs**
Dropping message with invalid data (Reason: Nonce 0x73 expired, cannot decode security encapsulated command.) but nothing changed other than the location. Anything that can be done? other than exclude and re-include?

**Additional context**
Add any other context about the problem here.
zwave-js logs while Re-Interviewing only node 020

```
02:28:41.886 CNTRLR   [Node 020] Beginning interview - last completed stage: None
02:28:41.887 CNTRLR   [Node 020] new node, doing a full interview...
02:28:41.887 CNTRLR » [Node 020] querying protocol info...
02:28:41.900 SERIAL » 0x0104004114ae                                                       (6 bytes)
02:28:41.901 DRIVER » [REQ] [GetNodeProtocolInfo]
                        payload: 0x14
02:28:41.907 SERIAL « [ACK]                                                                   (0x06)
02:28:41.909 SERIAL « 0x0109014153dc000440037e                                            (11 bytes)
02:28:41.910 SERIAL » [ACK]                                                                   (0x06)
02:28:41.911 DRIVER « [RES] [GetNodeProtocolInfo]
                        payload: 0x53dc00044003
02:28:41.917 CNTRLR « [Node 020] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Entry Control
                      specific device class: Secure Keypad Door Lock
                      is a listening device: false
                      is frequent listening: true
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
[... 219 lines cut ...]
02:30:00.107 SERIAL » [ACK]                                                                   (0x06)
02:30:00.108 DRIVER « [REQ] [SendData]
                        callback id:     218
                        transmit status: OK
02:30:10.115 CNTRLR   [Node 020] Interview attempt 4/5 failed, retrying in 20000 ms...
02:30:30.131 CNTRLR   [Node 020] Beginning interview - last completed stage: NodeInfo
02:30:30.133 CNTRLR   [Node 020] ManufacturerSpecificCC: doing a complete interview...
02:30:30.133 CNTRLR » [Node 020] querying manufacturer information...
02:30:30.143 SERIAL » 0x010900131402720425db7b                                            (11 bytes)
02:30:30.144 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      219
                      └─[ManufacturerSpecificCCGet]
02:30:30.149 SERIAL « [ACK]                                                                   (0x06)
02:30:30.153 SERIAL « 0x0104011301e8                                                       (6 bytes)
02:30:30.153 SERIAL » [ACK]                                                                   (0x06)
02:30:30.154 DRIVER « [RES] [SendData]
                        was sent: true
02:30:31.628 SERIAL « 0x01180013db00009402be7f7f7f7f010103060400000202041416              (26 bytes)
02:30:31.629 SERIAL » [ACK]                                                                   (0x06)
02:30:31.630 DRIVER « [REQ] [SendData]
                        callback id:     219
                        transmit status: OK
02:30:41.637 CNTRLR   [Node 020] Failed all interview attempts, giving up.
```

snagytx, 2021-01-28:
The whole zwave-js log where multiple startups were captured.

It seems that sometime the Node 020 is detected as "included securely", but it hangs at `ManufacturerSpecificCCGet`

```
02:05:07.545 CNTRLR   [Node 020] [+] [Security] interviewComplete: true      [Endpoint 0] [internal]
02:05:07.559 CNTRLR   [Node 020] The node is included securely.
02:05:07.562 CNTRLR   [Node 020] ManufacturerSpecificCC: doing a complete interview...
02:05:07.563 CNTRLR » [Node 020] querying manufacturer information...
02:05:07.582 SERIAL » 0x010900131402720425f454                                            (11 bytes)
02:05:07.583 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      244
                      └─[ManufacturerSpecificCCGet]
02:05:07.585 SERIAL « [ACK]                                                                   (0x06)
02:05:07.591 SERIAL « 0x0104011301e8                                                       (6 bytes)
02:05:07.591 SERIAL » [ACK]                                                                   (0x06)
02:05:07.592 DRIVER « [RES] [SendData]
                        was sent: true
02:05:07.687 SERIAL « 0x01180013f400000a03bf7f7f7f7f0101030605070002010000b2              (26 bytes)
02:05:07.687 SERIAL » [ACK]                                                                   (0x06)
02:05:07.688 DRIVER « [REQ] [SendData]
                        callback id:     244
                        transmit status: OK
02:05:17.701 CNTRLR   [Node 020] Interview attempt 1/5 failed, retrying in 5000 ms...
```

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5884635/zwavejs_1.log)

snagytx, 2021-01-28:
What is interesting is that the node is still working fine, I unlocked and re-locked the door lock - note that I have the pollValue logic enabled when `Manual lock/unlock operation` notification event is triggered

```
04:05:51.639 SERIAL « 0x010a00040014029840d000ef                                          (12 bytes)
04:05:51.640 SERIAL » [ACK]                                                                   (0x06)
04:05:51.642 DRIVER « [Node 020] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
04:05:51.655 SERIAL » 0x01110013140a9880c1abdeeaab365fa90539f2                            (19 bytes)
04:05:51.656 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      57
                      └─[SecurityCCNonceReport]
                          nonce: 0xc1abdeeaab365fa9
04:05:51.659 SERIAL « [ACK]                                                                   (0x06)
04:05:51.665 SERIAL « 0x0104011301e8                                                       (6 bytes)
04:05:51.666 SERIAL » [ACK]                                                                   (0x06)
04:05:51.667 DRIVER « [RES] [SendData]
                        was sent: true
04:05:51.770 SERIAL « 0x011800133900000b01d07f7f7f7f010103040000000201000013              (26 bytes)
04:05:51.770 SERIAL » [ACK]                                                                   (0x06)
04:05:51.771 DRIVER « [REQ] [SendData]
                        callback id:     57
                        transmit status: OK
04:05:51.829 SERIAL « 0x0125000400141d98817c93e0b674476fef032267ba710d1104b00bc1abf5d0025 (39 bytes)
                      90f73a5d000f7
04:05:51.831 SERIAL » [ACK]                                                                   (0x06)
04:05:51.836 DRIVER « [Node 020] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[NotificationCCReport]
                            notification type:   Access Control
                            notification status: 255
                            notification event:  Manual unlock operation
04:05:51.941 SERIAL » 0x0109001314029840253a34                                            (11 bytes)
04:05:51.942 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      58
                      └─[SecurityCCNonceGet]
04:05:51.945 SERIAL « [ACK]                                                                   (0x06)
04:05:51.951 SERIAL « 0x0104011301e8                                                       (6 bytes)
04:05:51.952 SERIAL » [ACK]                                                                   (0x06)
04:05:51.952 DRIVER « [RES] [SendData]
                        was sent: true
04:05:53.319 SERIAL « 0x011800133a00008901d07f7f7f7f010103040000000202041481              (26 bytes)
04:05:53.320 SERIAL » [ACK]                                                                   (0x06)
04:05:53.321 DRIVER « [REQ] [SendData]
                        callback id:     58
                        transmit status: OK
04:05:53.368 SERIAL « 0x0112000400140a988016cbe3389cbda1f5d0004c                          (20 bytes)
04:05:53.369 SERIAL » [ACK]                                                                   (0x06)
04:05:53.370 DRIVER « [Node 020] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceReport]
                          nonce: 0x16cbe3389cbda1f5
04:05:53.380 SERIAL » 0x011d001314169881a80a44ea20a4abf33c8967167887d4a94b2faeb7253b1f    (31 bytes)
04:05:53.381 DRIVER » [Node 020] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      59
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 22
                        └─[DoorLockCCOperationGet]
04:05:53.384 SERIAL « [ACK]                                                                   (0x06)
04:05:53.391 SERIAL « 0x0104011301e8                                                       (6 bytes)
04:05:53.392 SERIAL » [ACK]                                                                   (0x06)
04:05:53.393 DRIVER « [RES] [SendData]
                        was sent: true
04:05:53.505 SERIAL « 0x011800133b00000b01d07f7f7f7f010103040000000201000011              (26 bytes)
04:05:53.505 SERIAL » [ACK]                                                                   (0x06)
04:05:53.506 DRIVER « [REQ] [SendData]
                        callback id:     59
                        transmit status: OK
[... 124 lines cut ...]
04:06:00.480 SERIAL » [ACK]                                                                   (0x06)
04:06:00.481 DRIVER « [REQ] [SendData]
                        callback id:     64
                        transmit status: OK
04:06:00.538 SERIAL « 0x0123000400141b98816734cf31ae7010b17bf52e3242aad134301bd996a6d2933 (37 bytes)
                      272d00090
04:06:00.539 SERIAL » [ACK]                                                                   (0x06)
04:06:00.541 CNTRLR   [Node 020] [~] [Door Lock] currentMode: 0 => 255                  [Endpoint 0]
04:06:00.544 CNTRLR   [Node 020] [~] [Door Lock] outsideHandlesCanOpenDoor: false,false [Endpoint 0]
                      ,false,false => false,false,false,false
[... cut ...]
04:06:00.552 DRIVER « [Node 020] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[DoorLockCCOperationReport]
                            current mode:           Secured
                            active outside handles: false, false, false, false
                            active inside handles:  false, false, false, false
                            latch status:           open
                            bolt status:            locked
```

AlCalzone, 2021-01-28:
> 02:28:44.825 DRIVER   Dropping message with invalid data (Reason: Nonce 0x73 expired, cannot decode
>                       security encapsulated command.): [...]

That's the problem. I'm not sure why the message is discarded, because nonce 0x73 was just created. The driver now thinks the node is not included securely (that heuristic is not optimal) and won't communicate securely.

AlCalzone, 2021-01-28:
Does that happen too when you re-interview?

Re-including is **almost never** useful if the driver does stupid stuff - so don't.

AlCalzone, 2021-01-28:
Oh wait - that was a re-interview

> Interview start stuck at "RestartFromCache", on "Re-Interview" request it changes to "NodeInfo"

You played yourself btw. Re-Interviewing deletes all previously known info, so the driver now knows **less** about the device and has the same problem as before.

snagytx, 2021-01-28:
The node is not always considered as not secured, I think I caught the only time when it drops that one important message. 

But I do see the logs saying that it’s included securely but then it doesn’t respond to manufacturer or association. What is your recommendation? Just restart docker? Is there another option to restart the startup  process for a single node without using re-interview? Maybe leave the lock without power for 5 minutes to do a reset?

AlCalzone, 2021-01-28:
Normally, waking up the device should be enough. zwave-js is not perfect yet in that regard - I have a coupld of things I need to change.

However, the problem above means you're stuck with re-interviewing for now.

snagytx, 2021-01-28:
I got it. I'll see how I'll handle it. Maybe remove it, do a hard reset and re include it.

AlCalzone, 2021-01-28:
> Maybe remove it, do a hard reset and re include it

No point in doing that. Try to re-interview and see if the problem goes away. If not, wait for me to fix it. If it does, don't push the re-interview button again ;)

snagytx, 2021-01-29:
I played a bit more and noticed that even in ZWavePCController once securely included, using ZWavePCController, I can see all the info but then if I refresh info about the lock the secure part is not pulled. It might be something specific to the lock that it get's into a state or something when it doesn't share the secure info when interviewed a 2nd time - after inclusion.

I had no issue to pull the info of BE469ZP even though it was included using zwavejs2mqtt, note that I added the same network key in both.

I can't really explain why BE469 worked fine for a while, I even went back to old docker images that worked, but then decided not to work anymore.

Because I can't spend more time on this and I need something reliable I decided to upgrade BE469 to BE469ZP - BE469 was maybe 6-8 years old anyway.

AlCalzone, 2021-01-29:
It is possible that this is also going to be fixed by https://github.com/zwave-js/node-zwave-js/pull/1516, but that PR isn't entirely ready yet.
If you do find some spare time you could give that one a test run.

snagytx, 2021-01-29:
That pull seems to be more for ZWave+, this device id not ZWave+ compatible, since I used ZWavePcController v5.39.472.0 from Silicon Labs and zwave2mqtt, the openzwave version, with the same outcome I really suspect this is a door lock issue - it got stuck in a bad state - even the "[How to Reset a Schlage Connect BE468 or BE469 Smart Lock](https://www.acmelocksmith.com/blog/how-to-factory-reset-schlage-smart-locks/)" steps didn't help - same steps are described in the lock manual.

AlCalzone, 2021-01-30:
I've made some inquiries and it does look like the device is behaving really badly when it comes to the nonce (encryption) handling. I don't think I can support that behavior without introducing non-spec-compliant behavior into zwave-js.

closed 2021-01-30 by AlCalzone

labaland, 2021-02-07:
Same here. firs time it did work. now i cant get it back!! factory reset lock and new zwavejs setup. it can be added but i dont get ANY values

AlCalzone, 2021-02-08:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.