zwave-js/zwave-js #1518: Measure command time, time out responses accordingly

AlCalzone, 2021-01-28:
During a node capability discovery or interview, as well as traffic generated due to user activation, a
controlling nodes MUST timeout waiting for responses (reports) as part of the capability discovery or
controlling scenarios.
Two timers named CommandTime and ReportTime are used for timing out during a node discovery
interview. Illustrations are given for secure and non-secure cases in Figure 1 and Figure 2
* CommandTime is measured by the application
* ReportTime timeout SHOULD be set to CommandTime + 1 second.

[image]

LordMike, 2021-01-28:
#985  ? :)

AlCalzone, 2021-01-28:
Yes but this issue also affects dynaically changing timeouts in the state machine.

LordMike, 2021-01-28:
I could sense that, but here's to hoping :)

PR #3756 (2021-11-21): feat: compute message RTT, time out node updates faster
With this PR, we measure the round-trip-time for Serial API commands, which is especially important when sending commands to a node (time until ACK from the node). Adhering to the specs, we now time out waiting for a node response after *Command time* + 1 second, where *Command time* is:
a) the RTT for simple commands
b) the sum of the RTTs of the individual commands (get nonce + original command) for S0/S2 encapsulated commands

We previously waited 10 seconds (configurable through the `timeouts.report` driver option), which is too long. [... cut]
PR #3773 (2022-03-02): chore: prepare release v9
This PR is meant to collect the changes that will go into the upcoming v9 release.

fixes: #951
fixes: #1518
closes: #3651
closes:  #3120
fixes: #4272
fixes: #3091
closed 2022-03-02 by AlCalzone