zwave-js/zwave-js #1536: Aeon Labs Minimote appears as dead after network restart, despite being a battery device

kpine, 2021-01-30:
**Describe the bug**

After I include the Minimote, it is found and the status is Asleep. When I restart zwavejs2mqtt, the node is detected as Dead, and all the device information is blank (not reloaded from cache). Once I click on a button on a remote, it completes the interview process and keeps it in the Alive status, never going Asleep, as if it were a mains-powered device. The device information is filled back in.

The device was included while on battery power.

**Device information**

Which device(s) is/are affected (make/model)?  Aeon Labs Minimote
What are the node IDs? Node 008

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X] Previously working device (other platform)
    - Which platform? OZW
    - Have you made any recent configuration changes to the device? Describe. No.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [X] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Start inclusion process
2. Include Minimote
3. Notice everything looks ok
4. Restart zwavejs2mqtt
5. Notice the device details (mfg, product) are blank, and the device status is Dead
6. Press a button, device details appear and the device is marked incorrectly as Alive (should be asleep eventually).

**Additional context**

As discussed in the HA discord, this problem occurs because the device does not advertise the Wake Up CC during the interview, despite supporting it. I noticed the OZW device config file has several workarounds, including manually advertising the Wake Up CC, which is probably why I never noticed a problem there. Those are missing from the zwave-js config file.

After restarting the network, I noticed that if the very first thing I do is manually wake up the device (hold down the learn button for a few seconds), then zwave-js will decide to add the Wake Up CC to the node's list of CCs. Because of this, the attached cache file contains it. From this point on, restarting zwavejs2mqtt works fine and the device functions as expected (RestartFromCache, Asleep). If I send the wake up notification after pressing a button (thus finished the interview), the Wake Up CC will never be added.

Perhaps overriding the command classes in the config like OZW would be a good idea, to force zwave-js to think it supports the wake up CC during the inclusion. This would make a better experience. See [config file](https://github.com/OpenZWave/open-zwave/blob/master/config/aeotec/dsa03202.xml).

**Logfile:** [zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5897185/zwavejs_1.log)
**Cache files:** [logs.zip](https://github.com/zwave-js/node-zwave-js/files/5897186/logs.zip)

AlCalzone, 2021-01-30:
First breadcrumb: zwave-js assumes this device supports WakeUp, but querying the version tells us otherwise:
```
17:44:13.341 DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]
                      └─[VersionCCCommandClassReport]
                          CC:      Wake Up
                          version: 0
17:44:13.346 CNTRLR   [Node 008]   does NOT support CC Wake Up (0x84)
```
That doesn't have an effect though until the driver restarts - then the WakeUp CC is really missing from the list.

AlCalzone, 2021-01-30:
Okay, it seems that this behavior is not spec-compliant. To get around it, we should force-add the WakeUp CC as supported with a compat flag.

PR #1540 (2021-01-31): fix(config): force Wake Up as supported for Aeon Labs Minimote
fixes: #1536
closed 2021-01-31 by AlCalzone