zwave-js/zwave-js #1543: Why is 'lock jammed' listed as a variable rather then an event?

itewk, 2021-01-31:
I am trying to get my August Lock Pro (https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x033f/asl-03.json) to properly report if the lock is jammed or not.  When the lock jams I get this event:

```
{
    "type": "event",
    "event": {
        "source": "node",
        "event": "value updated",
        "nodeId": 8,
        "args": {
            "commandClassName": "Notification",
            "commandClass": 113,
            "endpoint": 0,
            "property": "Access Control",
            "propertyKey": "Lock state",
            "newValue": 11,
            "prevValue": 11,
            "propertyName": "Access Control",
            "propertyKeyName": "Lock state",
            "metadata": {
                "type": "number",
                "readable": true,
                "writeable": false,
                "min": 0,
                "max": 255,
                "label": "Lock state",
                "states": {
                    "0": "idle",
                    "11": "Lock jammed"
                },
                "ccSpecific": {
                    "notificationType": 6
                }
            },
            "value": 11,
            "ccVersion": 8
        }
    }
}
```

When I then un-jam the lock whether manually or automatically then I get:

```
{
    "type": "event",
    "event": {
        "source": "node",
        "event": "notification",
        "nodeId": 8,
        "notificationLabel": "Manual lock operation"
    }
}
```
or
```
{
    "type": "event",
    "event": {
        "source": "node",
        "event": "notification",
        "nodeId": 8,
        "notificationLabel": "RF lock operation"
    }
}
```

The problem is there is never another `"event": "value updated",` to clear the state of the lock jammed back to `0` / `idle`.  What I can't tell is if this is some particularity of the August lock that it doesnt send a new notification that the lock is unjammed, or if this has to do with how this library is differentiating between which events are value update events and which are asynchronous notifications

Digging around the code if found https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/notifications.json#L386-L393 defining the lock jammed as a variable and then down in the events seciton https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/notifications.json#L571-L576 there is a jumnp of 0x0A to 0xC skipping the 0x0B for lock jammed.

I am curious if/why not lock jammed isn't listed in that events section and is instead listed in variables section? Should the `0x0b` state be an event instead? Or is this something just weird with how August Lock is woring in which case need some specific override for the august device (if that is even possible in the config).

Before I did any PRs was hoping for some insight on what you all think the "correct" behavior here is for z-wave vs what I am seeing with my august lock.

AlCalzone, 2021-01-31:
The notifications in zwave-js (you found the correct .json) merely mirror the specifications in a format that is easier to parse than an Excel document. According to the specs, `Lock jammed` is a variable that can be reset to idle.
I'm not sure which version of the CC your lock supports. If it is 8 or higher, the device should send an idle notification by itself. If it is lower, zwave-js has a fallback timer that should set the variable to idle after 5 minutes.

itewk, 2021-01-31:
@AlCalzone how do i check which verson of CC lock supports?

itewk, 2021-01-31:
@AlCalzone also, where do you find the specs? i was trying to find something like that before I asked question, but couldn't find.

AlCalzone, 2021-01-31:
I have the relevant parts in the `specs` folder in this repo. Most of them are listed on Silabs website.

> how do i check which verson of CC lock supports?

Some device manuals mention it. If not, looking the cache file `<homeid>.json` contains the info.

itewk, 2021-01-31:
running in home-assistant via the in progress addon so not sure how to find the cache files but looking at the container logs I see the following:
```
09:14:26.872 CNTRLR   [Node 008] VersionCC: doing a complete interview...,
09:14:26.873 CNTRLR » [Node 008] querying node versions...,
09:14:28.334 CNTRLR « [Node 006] received response for node versions:,
                        library type:      Enhanced Slave (0x03),
                        protocol version:  4.61,
                        firmware versions: 1.59,
09:14:28.334 CNTRLR » [Node 006] querying CC versions...,
09:14:28.336 CNTRLR » [Node 006]   querying the CC version for Door Lock...,
09:14:29.788 CNTRLR « [Node 007] received response for node versions:,
                        library type:      Enhanced Slave (0x03),
                        protocol version:  4.61,
                        firmware versions: 1.59,
09:14:29.789 CNTRLR » [Node 007] querying CC versions...,
09:14:29.789 CNTRLR » [Node 007]   querying the CC version for Door Lock...,
09:14:31.212 CNTRLR « [Node 009] received response for node versions:,
                        library type:      Enhanced Slave (0x03),
                        protocol version:  4.61,
                        firmware versions: 1.59,
09:14:31.213 CNTRLR » [Node 009] querying CC versions...,
09:14:31.214 CNTRLR » [Node 009]   querying the CC version for Door Lock...,
09:14:32.659 CNTRLR « [Node 008] received response for node versions:,
                        library type:      Enhanced Slave (0x03),
                        protocol version:  4.61,
                        firmware versions: 1.59,
09:14:32.660 CNTRLR » [Node 008] querying CC versions...,
09:14:32.660 CNTRLR » [Node 008]   querying the CC version for Door Lock...,
09:14:34.069 CNTRLR   [Node 006]   supports CC Door Lock (0x62) in version 2,
09:14:34.070 CNTRLR » [Node 006]   querying the CC version for User Code...,
09:14:35.576 CNTRLR   [Node 007]   supports CC Door Lock (0x62) in version 2,
09:14:35.578 CNTRLR » [Node 007]   querying the CC version for User Code...,
09:14:37.041 CNTRLR   [Node 009]   supports CC Door Lock (0x62) in version 2,
09:14:37.041 CNTRLR » [Node 009]   querying the CC version for User Code...,
09:14:38.506 CNTRLR   [Node 008]   supports CC Door Lock (0x62) in version 2,
09:14:38.506 CNTRLR » [Node 008]   querying the CC version for User Code...,
09:14:39.935 CNTRLR   [Node 006]   does NOT support CC User Code (0x63),
09:14:39.935 CNTRLR » [Node 006]   querying the CC version for Manufacturer Specific...,
09:14:41.372 CNTRLR   [Node 007]   does NOT support CC User Code (0x63),
09:14:41.373 CNTRLR » [Node 007]   querying the CC version for Manufacturer Specific...,
09:14:42.839 CNTRLR   [Node 009]   does NOT support CC User Code (0x63),
09:14:42.840 CNTRLR » [Node 009]   querying the CC version for Manufacturer Specific...,
09:14:44.256 CNTRLR   [Node 008]   does NOT support CC User Code (0x63),
09:14:44.257 CNTRLR » [Node 008]   querying the CC version for Manufacturer Specific...,
09:14:45.712 CNTRLR   [Node 006]   supports CC Manufacturer Specific (0x72) in version 2,
09:14:45.716 CNTRLR   [Node 006]   skipping query for Security (0x98) because max implemented versio,
                      n is 1,
09:14:45.717 CNTRLR » [Node 006]   querying the CC version for Version...,
09:14:47.170 CNTRLR   [Node 007]   supports CC Manufacturer Specific (0x72) in version 2,
09:14:47.171 CNTRLR   [Node 007]   skipping query for Security (0x98) because max implemented versio,
                      n is 1,
09:14:47.172 CNTRLR » [Node 007]   querying the CC version for Version...,
09:14:48.588 CNTRLR   [Node 009]   supports CC Manufacturer Specific (0x72) in version 2,
09:14:48.589 CNTRLR   [Node 009]   skipping query for Security (0x98) because max implemented versio,
                      n is 1,
09:14:48.589 CNTRLR » [Node 009]   querying the CC version for Version...,
09:14:50.019 CNTRLR   [Node 008]   supports CC Manufacturer Specific (0x72) in version 2,
09:14:50.020 CNTRLR   [Node 008]   skipping query for Security (0x98) because max implemented versio,
                      n is 1,
09:14:50.021 CNTRLR » [Node 008]   querying the CC version for Version...,
09:14:51.406 CNTRLR   [Node 006]   supports CC Version (0x86) in version 2,
09:14:51.407 CNTRLR » [Node 006]   querying the CC version for Z-Wave Plus Info...,
09:14:52.882 CNTRLR   [Node 007]   supports CC Version (0x86) in version 2,
09:14:52.883 CNTRLR » [Node 007]   querying the CC version for Z-Wave Plus Info...,
09:14:54.329 CNTRLR   [Node 009]   supports CC Version (0x86) in version 2,
09:14:54.330 CNTRLR » [Node 009]   querying the CC version for Z-Wave Plus Info...,
09:14:55.777 CNTRLR   [Node 008]   supports CC Version (0x86) in version 2,
09:14:55.779 CNTRLR » [Node 008]   querying the CC version for Z-Wave Plus Info...,
09:14:57.194 CNTRLR   [Node 006]   supports CC Z-Wave Plus Info (0x5e) in version 2,
09:14:57.194 CNTRLR   [Node 006]   skipping query for Transport Service (0x55) because max implement,
[... 17 lines cut ...]
09:14:58.630 CNTRLR » [Node 007]   querying the CC version for Battery...,
09:15:00.050 CNTRLR   [Node 009]   supports CC Z-Wave Plus Info (0x5e) in version 2,
09:15:00.051 CNTRLR   [Node 009]   skipping query for Transport Service (0x55) because max implement,
                      ed version is 0,
09:15:00.053 CNTRLR   [Node 009]   skipping query for Security 2 (0x9f) because max implemented vers,
[... cut ...] as jammed? though i guess if the device doesn't specifically give an "unjammed" notification, what is there to do? I see a bunch of code in the old home assistant zwave intigration working around various locks notifications to accuratly determine the jammed status, so I guess the answer to my issues will have to be something similar up at the home assistant level listenint for other events such as the rf or manual lock operation and clearning the jammed status then. *sigh* nothing is ever simple

AlCalzone, 2021-01-31:
To answer your question - the Notification CC version should show up during that phase of the interview. It is just not in that log section.

itewk, 2021-01-31:
@AlCalzone 
```
09:14:04.713 CNTRLR « [Node 008] received response for protocol info:,
                      basic device class:    Routing Slave,
                      generic device class:  Entry Control,
                      specific device class: Secure Keypad Door Lock,
                      is a listening device: false,
                      is frequent listening: true,
                      is a routing device:   true,
                      is a secure device:    unknown,
                      is a beaming device:   true,
                      maximum baud rate:     40000 kbps,
                      version:               4,
```
Would that mean CC version 4?

I have left it sitting for well over an hour now and the value does not seem to have gone back to 0.

AlCalzone, 2021-01-31:
No it should look like

    [Node 008]   supports CC Notification in version X

itewk, 2021-01-31:
> supports CC Notification in version

```
09:15:15.633 CNTRLR   [Node 006]   supports CC Notification (0x71) in version 8,
09:15:15.633 CNTRLR » [Node 006]   querying the CC version for Multi Channel Association...,
09:15:17.082 CNTRLR   [Node 007]   supports CC Notification (0x71) in version 8,
09:15:17.083 CNTRLR » [Node 007]   querying the CC version for Multi Channel Association...,
09:15:18.556 CNTRLR   [Node 009]   supports CC Notification (0x71) in version 8,
09:15:18.556 CNTRLR » [Node 009]   querying the CC version for Multi Channel Association...,
09:15:19.978 CNTRLR   [Node 008]   supports CC Notification (0x71) in version 8,
```

so that reads as vesion 8, which in theory means the lock should be clearing the `Lock State` propety back to idle, but its not....so, any hooks to work around when device isn't behaving as expected?

itewk, 2021-01-31:
would https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/notifications.json#L386-L393 need to be updated to be like:

```
			{
				"name": "Lock state",
				"states": {
				        "0x00": {
						"label": "Idle"
					}
					"0x0b": {
						"label": "Lock jammed"
					}
				}
			},
			```
to pick up any sent idle event?

I was try9ijng to find some way to log out all zwave traffic to see if the lock is sending out some message when it becomes unjammed that isn't getting picked up or something, but couldn't figure out a way to see the raw zwave traffic

AlCalzone, 2021-01-31:
No, the idle state is not listed explicitly. Instead, the variables that cannot be idled are marked with `"idle": false` or something like that.

itewk, 2021-01-31:
@AlCalzone i am using it within the zwave-js-server right within the home assistant addon, any suggestion for enabling the log in that context?

AlCalzone, 2021-01-31:
Guess you need to ask the HASS devs

shbatm, 2021-03-18:
@itewk Did you ever find a resolution for this? I'm finding that my Schlage BE469P locks are not clearing the Jammed notification (does not reset to idle) and was wondering if I'm missing something obvious?

Using the ZwaveJS2MQTT Add-on for Home Assistant. "Stuck" status shows in Dashboard, Home Assistant, and on MQTT broker.

AlCalzone, 2021-03-18:
Version 6.6.3 (and 7) contains a bugfix which might explain the old behavior:
> Notification variables are now auto-idled after 5 minutes as it was intended, not after 5 hours.

I guess this was the root cause here too.

itewk, 2021-03-18:
@shbatm in the case of the August lock I watched the traffic for a while and August sends an event when the lock gets jamed but does not send an Event when it gets unjamed (putt it back to idle), you get all the Notifications that the lock has been operated. Therefor logic would have to be put in speicfic to the lock to infer that if the lock was in jammed state, but then a different lock operation occured, it must not be jammed any more. all this sort of "work around the weirndess of each lock" code did/does exist in the old zwave project, i started playign with how to port it over, but then lost steam.

This is all the special, literally labeled work around, logic that was dealing with this in the old zwave project, something similar would need to be ported over to deal with these sort of infernce states: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zwave/lock.py#L261-L341

AlCalzone, 2021-03-18:
> Therefor logic would have to be put in speicfic to the lock to infer that if the lock was in jammed state, but then a different lock operation occured, it must not be jammed any more.

Lock jammed is a notification variable, which gets auto-idled by zwave-js, unless the device supports Notification CC V8 or newer. In previous versions, the timeout for this was mistakenly set to 5 hours, not 5 minutes which is why you wouldn't see any change back.

If that is not enough, a heuristic for events that can clear the "jammed" status could be added here:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/node/Node.ts#L2479

itewk, 2021-03-18:
@AlCalzone i dont remember seeing https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/node/Node.ts#L2479 when i alst looked at the code base. looks like a promising place to deal with the issue. I would say the 5 minute auto idle isn't a good solution because you end up in a false reporting state in 2 casses

a) door gets jamed and then imediatlly unjammed, status would report jammed for another 5 minutes while it isn't jammed and is secured
b) door is still jammed after 5 minutes, but the flag gets cleared, but actually still jammed

neither of those false reporting states is good, especially B because that could report a secure state when not in one.

shbatm, 2021-03-18:
Thanks for the feedback and looking back into this. I agree with @itewk that there are some gotchas with a 5-minute reset of the jammed status. I would actually be OK if it just sent the notification event like for other operations (e.g. Manual unlock operation), then the downstream software can determine if it's still jammed or not (e.g. if receiving a Manual lock operation after a Jammed event).

@AlCalzone - For my Schlage BE469ZPs, it's not clearing the event at all (5 min or 5 hours). I jammed the back door lock yesterday afternoon during testing and it's still showing Jammed in my Z-Wave JS Dashboard, the front door got jammed several days ago, and it's still showing jammed as well. I will update to >=6.6.3 and see if it's still having issues.

Zwavejs Version - 6.6.2
Zwavejs-server Version - 1.1.1
App Version - 2.3.0

Thanks all, awesome work on this project!

AlCalzone, 2021-03-18:
@shbatm If you could make a zwave-js log for good measure, you should see if it changes the value. That's how we discovered that the timeout was 5 hours.

shbatm, 2021-03-18:
Attached are zwavejs and zwavejs2mqtt logs from while I was messing with the lock. I updated to 6.6.3 (using ZWaveJS2MQTT addon for Home Assistant) and am still seeing the same problem, the lock jammed status is not being reset by the lock itself or by the zwavejs after 5 minutes.

The zwavejs2mqtt log has some commentary on the actions I was performing. I did want to note that prior to re-interviewing the node after the upgrade, it was changing the bolt status to LOCKED whenever I jammed the lock. This could definitely be a problem if it was more than just a hiccup.

Both log levels are on INFO and only my lock nodes are being logged for zwavejs, please advise if different verbosity or more complete logs are required.

[zwavejs_355.log](https://github.com/zwave-js/node-zwave-js/files/6166656/zwavejs_355.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6166657/zwavejs2mqtt.log)

AlCalzone, 2021-03-18:
The commentary helps, but I need the zwave-js log with a loglevel of `debug`.
Just the part where you jam and then wait 5 minutes should be enough.

shbatm, 2021-03-18:
Here you go.

1) Manual Unlock
2) Jammed
3) Wait 2 minutes, manual lock
4) Wait 7 minutes, RF unlock

No reset of jammed status. Jamming the lock still sets the bolt to "LOCKED" despite it being open.

[zwavejs_355.log](https://github.com/zwave-js/node-zwave-js/files/6166891/zwavejs_355.log)

AlCalzone, 2021-03-19:
Did you remove log lines between the blocks you commented? If so, please don't - I want to avoid that the log is missing info you might have overlooked.

>  Jamming the lock still sets the bolt to "LOCKED" despite it being open.

But the device reports that:
```
14:44:23.249 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[DoorLockCCOperationReport]
                            current mode:           Unknown
                            active outside handles: false, false, false, false
                            active inside handles:  false, false, false, false
                            latch status:           open
                            bolt status:            locked
                            door status:            open
```
IMO it shouldn't, since the bolt status isn't actually locked. You should report that to the manufacturer IMO.
At least it reports the current lock mode as `Unknown`, so you can use that info to check if the lock is locked. 

> EDIT 2: Still showing Lock Jammed after 6 hours.

Which device is that? This one? https://products.z-wavealliance.org/products/3706?selectedFrequencyId=2
If so, it supports Notification CC V8, for which the specifications mandate that the device itself sends the idle notification:
[image]

The rest of the log looks fine to me.

shbatm, 2021-03-19:
> Did you remove log lines between the blocks you commented? If so, please don't - I want to avoid that the log is missing info you might have overlooked.

Nope. Just added line break and comments.

> You should report that to the manufacturer IMO.
At least it reports the current lock mode as Unknown, so you can use that info to check if the lock is locked.

I'll see if I can find the right way to notify them. They're pretty good about support, but usually just involves them sending me a new lock...

> Which device is that? This one? 

Yes, that should be it. I can confirm the exact hardware version later. I also have an older non-Z-wave+ version I haven't checked yet. 

So is it worth creating a work around for devices that don't comply with V8? OZW would report all Access Control notifications under the same topic, instead of splitting them into events and a few separate parameters. Would it be possible to also fire off a Lock Jammed event/notification the same way it does for a normal operation (e.g. 'Manual locked operation'), in addition to storing the value as it is now? That would at least allow for tracking the lock jammed status in separate software (e.g. In Home Assistant). I can pick up the jammed event and if another operation follows I know it was cleared. Right now there's no way for me to see if the lock gets jammed a 2nd time because no event is triggered and the value isn't updated (since it never gets reset).

AlCalzone, 2021-03-19:
You have to distinguish between notification variables and notification events. zwave-js mirrors the Z-Wave specs here (and I think they actually make sense in this regard). Variables represent state, events just have a meaning the instant they are received.

A jammed lock is a state, you locking the device is a one-time event (which affects the locked state as a result).

> OZW would report all Access Control notifications under the same topic, instead of splitting them into events and a few separate parameters.

`zwave-js` reports all notification **events** the same way. How these get transformed into topics is handled in `zwavejs2mqtt`. IMO it makes sense to have separate topics for notification **variables**, because these represent states that are independent of each other.

> Right now there's no way for me to see if the lock gets jammed a 2nd time because no event is triggered and the value isn't updated (since it never gets reset).

That's a device bug though. I'm open to adding a workaround to enable auto-idling on non-compliant V8 devices, which would restore the behavior you're expecting.

However, I really want users who are affected by broken devices (like you) to put some pressure on manufacturers, even if it is just telling them: _"Your stuff doesn't work correctly, fix it!"_
Otherwise, they'll keep their attitude _"🤷🏻‍♂️ but it works in SmartThings/OZW/..."_ when in reality those systems added device-specific workarounds to handle the broken stuff many manufacturers just threw on the market and then forgot.

shbatm, 2021-03-19:
> You have to distinguish between notification variables and notification events. zwave-js mirrors the Z-Wave specs here (and I think they actually make sense in this regard). Variables represent [...]

Thanks for the clarification. I looked at the specs in more detail and I understand now why you're differentiating.

> That's a device bug though. I'm open to adding a workaround to enable auto-idling on non-compliant V8 devices, which would restore the behavior you're expecting.

Agree it is a device issue. I will try and figure out the correct avenue to report to Schlage; unfortunately, while I agree pressure on the OEMs is the right way to get them to conform to the standards they say they are conformant with, I also know that the workarounds are going to be unavoidable -- I've haven't seen any published firmware updates for Schlage Z-Wave locks in the several years I've had them, so even if they fix future versions, currently installed locks probably wouldn't see the change.

Let me know if there's anything I can do to help with testing a work around; don't think I have the time at the moment to try and code it myself. 

I do agree with @itewk's comments above: any workaround needs to account for false-positives if possible, and watching for a successful follow-up operation and/or the failed electronic transition counts would be more accurate than a blind auto-idle timer.

> a) door gets jammed and then immediately unjammed, status would report jammed for another 5 minutes while it isn't jammed and is secured
> b) door is still jammed after 5 minutes, but the flag gets cleared, but actually still jammed

AlCalzone, 2021-03-19:
So to summarize, we actually need three things here:

1. compat flag to enable auto-idling of values even if the CC is V8+
2. definitions in `notifications.json` which notification variables must not be auto-idled (jammed should be one of them)
3. more heuristics to handle the implicit dependencies between certain notification variables and events

AlCalzone, 2021-04-05:
Just a heads up:
1. is done and opt-in for all devices. We no longer auto-idle notification values by default.

itewk, 2021-04-05:
@AlCalzone coolio, thanks

matteskonto, 2021-05-01:
I have a similar issue with ID-150 from ID-lock where the Lock jammed value do not reset and I guess it was autoreset before because it was working some month ago.

I am hesitant to re-interview the node to reset the value for testing because I have had issue with battery drain and now it is working great so I don't want to disturb it unnecessarily

Apple2T4ch, 2021-07-29:
I have this issue with my Kwikset 914C lock. Tried reinterviewing and excluding/including again. My notification CC is 8.

jeeftor, 2021-09-08:
I think I just saw this issue on my august pro. I’ll dig into the logs. And try to figure out how to harass august for an update

jeeftor, 2021-09-09:
Looks like latest release of ha

https://www.home-assistant.io/blog/2021/09/01/release-20219/￼

Z-Wave JS Updates

If you are a Z-Wave siren/doorbell owner, you probably know how frustrating it was to wait for Home Assistant to add entity support for the Sound Switch CC (Command Class). This is caused by the blazing fast speed the upstream Z-Wave JS project is moving forward at; Often much faster than Home Assistant is able to keep up with.

It also might be that Home Assistant cannot (or is likely not to) support the specific command class you care about; Nevertheless, that shouldn’t stop you from being able to create automations for those devices!

In this release, we have introduced the new zwave_js.value_updated automation trigger type which will allow you to trigger an automation off of any Command Class value that Z-Wave JS supports

Might be useful?

itewk, 2021-09-09:
@jeeftor its been a while since i touched this and will need to upgrade my system and play around with it again. this might make making some automation easier. the fundemental problem really is with how august doesn't send a specific "unjammed" event. you have to infer it from some other event like the door opening, closing, locking, unlocking after a jammed event. but even that i remember was difficult to deal with previously. this new feature sounds like it may do the trick. i can't make any promises when i will wake up on a saturday morning and want to muck with my homeassistant again, but im sure the fancy will strick me.

jeeftor, 2021-09-09:
It's something I'd also like to look into from an automation side. I switched to the native august from the zwave recently and realized I couldn't disable auto lock programmatically so I thought I'd look at zwave again. 

My thought was If I can get a jammed event and then any other event we can probably mimic the ozw functionality in a blueprint or something and/or roll it into ha somehow. 

But I have 3 small kids and almost never have free time :(

mmiller7, 2022-03-27:
Ran into this with a Yale 256 deadbolt trying to move to new zwave and its a real pain because a VERY common usecase any time we have visitors is someone unlocks the door, exits, and then they don't fully close the door latched before tapping lock so it jams in an unlocked state.

I've also noticed this same issues appears to be affecting the keypad-lockout (too many tries) sensor that is stuck "true" ever since I tested that earlier today...so now I can't get either of those updates.

I know the lock is sending events because if I turn on MQTT I can see timestamps sent over...but there seems to be no other way to get this data because none of the events I've found show an event fired and the booleans are stuck "on".

I'm not sure why this is so difficult to fix that its been going for over a year -- either the lock is working (locking and unlocking) or its jammed.  If its jammed, it won't be locking or unlocking.  If its locking and unlocking, its not jammed.  Any successful lock or unlock clears the jam in real life and I would expect the software objects to match that.

jeeftor, 2022-03-27:
I think because we would have to write software to induce a behavior that’s not coming directly from the lock. Aka a software reset of the status. I think somebody mentioned open Zwave used to do this maybe.

kpine, 2022-03-27:
> I think because we would have to write software to induce a behavior that’s not coming directly from the lock. Aka a software reset of the status. I think somebody mentioned open Zwave used to do this maybe.

OZW doesn't model notification as events or states, and there is no idle reset. It's just a single state variable for the notification type. When one event happens you lose the previously set value (this is for 1.4, 1.6 does idle in some cases, and models the notifications slightly different, but it's still a single variable). This is why many users think this worked in OZW, when it really didn't. OZW just replaces the values for every event.

For example
1. Lock sends Jammed notification. OZW sets its "access control" variable is set to 0xb.
2. User unlocks the lock successfully, lock sends Manual unlock notification, OZW sets its "access control" variable to 0x1.

Well, no surprise there, user doesn't worry about the lock jam state because the value in the sensor was replaced with the unlock notification. Since Z-Wave JS treats the unlock as an event, the jam state is not implicitly cleared.

The same is true for the separate alarm_type and alarm_value. The previous values are just lost when new ones come in.

If you want to replicate the behavior of OZW in HA, you can create a template sensor that is triggered from the `zwave_js_notification` event, and the Value Updated Device Trigger. Set the last numerical state of the sensor based on either the last value triggered. Only caveat with trigger-based templates are that their state will be "Unknown" on startup, until one of the events occurs.

mmiller7, 2022-03-27:
> > I think because we would have to write software to induce a behavior that’s not coming directly from the lock. Aka a software reset of the status. I think somebody mentioned open Zwave used to do this maybe.
> 
> OZW doesn't model notification as events or states, and there is no idle reset. It's just a single state variable for the notification type. When one event happens you lose the previously set value (this is for 1.4, 1.6 does idle in some cases, and models the notifications slightly different, but it's still a single variable). This is why many users think this worked in OZW, when it really didn't. OZW just replaces the values for every event.
> 
> For example
> 
>     1. Lock sends Jammed notification. OZW sets its "access control" variable is set to 0xb.
> 
>     2. User unlocks the lock successfully, lock sends Manual unlock notification, OZW sets its "access control" variable to 0x1.
> 
> 
> Well, no surprise there, user doesn't worry about the lock jam state because the value in the sensor was replaced with the unlock notification. Since Z-Wave JS treats the unlock as an event, the jam state is not implicitly cleared.
> 
> The same is true for the separate alarm_type and alarm_value. The previous values are just lost when new ones come in.
> 
> If you want to replicate the behavior of OZW in HA, you can create a template sensor that is triggered from the `zwave_js_notification` event, and the Value Updated Device Trigger. Set the last numerical state of the sensor based on the last value triggered. Only caveat with trigger-based templates are that their state will be "Unknown" on startup, until one of the events occurs.

This seems (in my mind) like the preferred thing to have happen because generally that is a more accurate reflection of reality (it stays jammed until I walk over and cycle the bolt, then it stays locked or unlocked until something else happens).

I'm about 12 hours into my adventure of figuring out how to get the same functionality as before, and I only just figured out that I can (and MAYBE how to) see what data is sent when the "Value Updated" happens to go along with the lock/unlock `zwave_js_notification`.  I'm now at the point I need to try and figure out how I can use this to make......"something".......that gives me a status field for the various types of alarms.

jeeftor, 2022-03-28:
[image]

I have zero faith this will matter at all but I filed a support ticket

jeeftor, 2022-03-28:
I haven't looked into this project much but it sounds like adding custom code for august and other locks is probably a theoretical "breaking" of how ZwaveJS is supposed to work? If we were to add a patch would that be done in this lib or in Home Assistant somehow?

mmiller7, 2022-03-28:
> [image]
> 
> I have zero faith this will matter at all but I filed a support ticket

For what its worth, I also wrote Yale support and I fully expect their reply to be "your zwave hub should handle it better".

I know Yale claims they have a new Z-Wave module (the RF modules are separate boards so you can buy a lock and then buy Bluetooth, WiFi, Z-Wave, Zigbee, etc) and allegedly the new one will be S2-security enabled...but it sounds like its not past the press-release stages and at the moment looks like everything Z-Wave is sold out across nearly all brands/models/devices so I expect even if a new module with new standards is a "fix" it will be a long time coming.  Something something "chip shortage" as everyone is pointing at these days.

jeeftor, 2022-03-28:
So i sent an email to CertAdmin@Z-WaveAlliance.org - no idea if this will help - but if August is incorrectly implementing the v8 command "thing" and ARE certified that's maybe an issue?

Z-Wave Certification Support

> Z-Wave Certification Team for questions regarding the overall certification process or questions regarding certification interoperability testing done by the independent test partners: CertAdmin@Z-WaveAlliance.org

mmiller7, 2022-03-29:
For what its worth, I got a reply from Yale basically saying try remove/re-include and pointing the finger back at the ZwaveJS software needing to "support it".

`Answering your questions, there is no newer version for the Z-wave module. We suggest contacting the Z-wave provider to confirm if the new Z-wave controller is fully compatible with the Yale Smart Lock.`

itewk, 2022-03-29:
insert spiderman meme here.

mmiller7, 2022-03-29:
Only thing I can think of would be cite a specific make/model of pieces involved and then write up a specific test-case with every step, the current status, action to take, expected status, and actual result.

So that would probably be a lot of lines
Observe X
Take action Y
Observe <specific data field>
Verify <data field value>
Expected: (what you think it should be)
Actual: (what is missing, or the wrong value)

That's how we write up test cases for Q&A at work to send back to engineers when something fails, the full environment, steps to reproduce the problem, and observations at every step compared to expected results.

Unfortunately though, its so easy to then also turn around and get "works for me, closed" or "it works the way we think it should".

jeeftor, 2022-03-30:
Has anybody reached out to august about this?

I got another reply from Z-Wave

> I can confirm that this product was certified as compliant with Notification CC v8 requirements
> Did you address the issue with manufacturer? If yes, what did they reply?

I currently have a support ticket open with August but I don't think they understand what's going on. They want me to send a video of the app 🤦🏼 

If anybody has already gone this route please let me know and I can forward that info to Z-Wave....

jeeftor, 2022-03-30:
[image]

So... thanks august... :) 

I think my next step will be to run a network traffic capture on the z-Wave data stream - showing the jammed state being set and never unset... except I'm not 100% sure how to do this - but I assume I can find it in zwavejs2mqtt somewhere

itewk, 2022-03-30:
@jeeftor im pretty sure i already captured that in https://github.com/zwave-js/node-zwave-js/issues/1543#issuecomment-770409993. i went through all those experiments back when i started this. but probably worth capturing a new set of data.

jeeftor, 2022-03-30:
I just sent some logs to Betina at the zwave consortium - additionally I used my work address which _maybe_ looks more official :)

jeeftor, 2022-04-06:
I just reached back out to zwave to see if they can give me any info on "what happens next" ... which I suspect is  🦗🦗🦗🦗

jeeftor, 2022-04-09:
> Sorry for not getting back to you earlier.
>  
> What we have concluded is that the lock should provide a state idle notification.
> I have therefore requested a test house to verify, and I have notified Yale/August that we are looking into this.
> If there was a slip in verification, the manufacturer will be advised to correct the issue and re-certify.
> If the product is correctly verified, then you must reach out to the manufacturer or the supplier
>  
> It will take some time before I get a result, but when I know more, I will let you know.
>  
> Kind Regards,

jfroy, 2022-04-21:
I'm in the same situation with a Yale YRD256 lock. But that's a V4 notification product/module, so if I understand this issue correctly, there's no requirement for the lock to idle the state/variable.

It seems to me, taking a step back, that zwave-js should have infrastructure for product "quirks", much like the Linux kernel. It's fine to have a "we follow the spec to the T" approach, but the real world is messy and products have defects. I'm new to this project and HA in general, so I don't know if this has been discussed before.

jeeftor, 2022-04-24:
@AlCalzone - Any way we can get some sort of patch/fix for our "situation" onto the roadmap?

jeeftor, 2022-04-24:
@jfroy - I believe the old openzwave implementation for HA actually did handle this situation -> whereas zwavejs doesn't currently - however - since home assistant hired @AlCalzone - maybe there is hope for the future!!!

AlCalzone, 2022-04-26:
This issue is still open, I'll get to it at some point.

evanhfox, 2022-06-12:
> This issue is still open, I'll get to it at some point.

Indeed this is still an issue. Some excellent troubleshooting documented in this issue thus far. Have this issue with 2 x of my Yale YRD256 locks.

jeeftor, 2022-08-17:
Still an issue! :)

mbaran5, 2022-11-12:
Finding this here coming up on two years after it was initially reported. I have a new Yale Assure Lock 2 with the YRMZW2-US module and am facing this issue. Did anyone come up with a programatic way to work around it? I know I can clear the status from the HA side with a simple automation but it'll still always reflect incorrectly on the Z-Wave JS side.

[20 more replies]

closed 2023-04-04 by zwave-js-bot