zwave-js/zwave-js #1551: [feat] Add links to z-wave manual or zwave-js configs file

LordMike, 2021-02-01:
**Is your feature request related to a problem? Please describe.**
Sometimes when working on devices, it's useful to read more about configs or behaviour in the manuals available. In z-wave, this is doubly easy, as the z-wave standardization seems to mean that manuals are available at a central location.

**Describe the solution you'd like**
Alongside the export button/link for the node json, you could add a link to the relevant zwave-js config file, and the manual on z-wave.info (or other place .. I'm not entirely sure how to find them).

### Config file:

For the config file, you can link to something like this. In some cases, the config file has a version range on it (some of the files in this directory have that). Maybe zwave-js can tell you which config file is being used? - alternatively just linking to the model number for that manufacturer is a good start (remember to not link when manufacturer is 0x0000 :)).

> https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x019a/11-01-011.json

### Manuals
Sometimes when searching for model numbers on google, I find these links:

> https://manuals.zwave.eu/backend/make.php?lang=EN&sku=ZWA012-C&cert=ZC12-20060075&type=mini
> http://manuals-backend.z-wave.info/make.php?lang=en&sku=ZW112-B&cert=ZC10-16010007
> http://manuals-backend.z-wave.info/make.php?lang=en&sku=ZWA008-C&cert=ZC10-19066568

If possible, it'd be cool to link to these, or better manuals if available.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

AlCalzone, 2021-02-01:
@blhoward2 should be easy to add, now that we know the Z-Wave alliance IDs for most devices, right?

blhoward2, 2021-02-01:
Yes, the links are in the file for most. I'll take a look.

LordMike, 2021-02-01:
> 
> 
> @blhoward2 should be easy to add, now that we know the Z-Wave alliance IDs for most devices, right?

As I wrote, I think it could also be cool to link to zwave-js's config file for the device.. Can zwavejs2mqtt see _which_ config file is in use?

blhoward2, 2021-02-01:
I'm not sure about that aspect. They're meant to be invisible to the user and UI space is at a premium. We're not linking to the file from our website, which is under development, either. @AlCalzone @robertsLando Thoughts?

LordMike, 2021-02-01:
Well, to be fair, zwave-js' API has nothing to do with a UI - making the information available is a purely programmatic thing.. :)

blhoward2, 2021-02-01:
The zwave-js server can chose to make the config file available if they like in the integration. They have the ability to tell which it is.

LordMike, 2021-02-01:
> We're not going to add a field in the file with the file's name so that part of your suggestion is merely a UI change.

I would never suggest that.. :D But it's cool that it's available.. Means I made my issue in the right repository :)

Do you know how those other manual links are made (http://manuals-backend.z-wave.info/make.php?lang=en&sku=ZW112-B&cert=ZC10-16010007)?.. I've found that removing the `cert` parameter yields a template page, so it's necessary, but I have no idea how to arrive at this value.. I suspect it's a verification something certificate perhaps.

blhoward2, 2021-02-01:
The links to the manufacturer manual in the page? We have those available to us from the zwave alliance database, from which that page is partially built.

robertsLando, 2021-02-01:
About the link to the config file that could be provided by zwavejs or calculated on runtime

robertsLando, 2021-02-01:
Or, even better, we could build an heklper function on zwavejs side to get the config file link from a node. Like `node.getConfigLink()`

LordMike, 2021-02-01:
> 
> 
> About the link to the config file that could be provided by zwavejs or calculated on runtime

blhoward2, 2021-02-01:
> Or, even better, we could build an heklper function on zwavejs side to get the config file link from a node. Like `node.getConfigLink()`

Isn't there already a function? We look up the file by device type/id in importConfig.ts.

AlCalzone, 2021-02-01:
ConfigManager does not parse the metadata yet

blhoward2, 2021-02-01:
Different question. If a front end wants to include a link to the config file, do they have a way to identify the appropriate file through to link to? We do that in importConfig now using configManager.

PR #1584 (2021-02-09): feat(config): add links to device manuals
Provides the info to integrate a link to the manual, which will close #1551. @marcus-j-davies @robertsLando
closed 2021-02-09 by AlCalzone