zwave-js/zwave-js #1552: [bug] Honeywell (Jasco) switches displaying Unknown manufacturer 99 / 18770 / Unknown product 12856

ragenhe, 2021-02-01:
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.0.1
zwavejs version: 6.1.0

### Describe the bug

Honeywell (Jasco) light switch(es) are showing Unknown manufacturer 99 / 18770 / Unknown product 12856.
I have deployed three of them, all behave / display the same way.

### To Reproduce

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

I didn't expect this switch to have an issue detecting, as GE/Jascos are detected without issue.

### Additional context

The switches are fully functional.
[image]
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/5896948/zwavejs2mqtt-store.zip)

AlCalzone, 2021-02-01:
@blhoward2 did you get to that device yet?

ragenhe, 2021-02-01:
@AlCalzone @blhoward2 

Looks like the device _was included this morning on the updated dev docker.

Thanks guys!!

closed 2021-02-01 by ragenhe