zwave-js/zwave-js #1553: [bug] SWITCH_MULTILEVEL SET missing on Philio PSE02 Siren

Gis70, 2021-02-01:
### Version

Build/Run method

- [X ] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.0.1
zwavejs version: 6.1.0

### Describe the bug

Hi, i'm new on Zwave-js2MQTT, i'm come from Jeedom where they have their proper zwave adaptation based on OZW 1.4.

My issue is that's i can't find SWITCH_MULTILEVEL SET for my Philio PSE02 Siren.

So i'm not able to change type of sound as in the manual :

[image]

### To Reproduce

Steps to reproduce the behavior:

[image]

### Expected behavior

Have a menu where I can send the sound ID

I can try to modify directly a config file with your help. I opened the json file of this device and compared it with a device which have SWITCH_MULTILEVEL but it seems it's not in this file.

Thanks a lot for your help.

Gis70, 2021-02-01:
@robertsLando  As i said previously in my edited message, logs are in silly mode and it's the Node 2, the node 13 is shutter, and at the moment it's OK. (begin my merge of all zwave devices)

Do i launch a re interview ?

AlCalzone, 2021-02-01:
Yes, please re-interview that device and post the logfile and the `<homeid>.json` cache file here.

Gis70, 2021-02-01:
@AlCalzone @robertsLando this is my lasts logs just after a interview :   
`14:43:35.520 CNTRLR   [Node 002] Beginning interview - last completed stage: None`

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5904601/zwavejs_1.log)

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5904603/zwavejs2mqtt.log)
[e5248c75.json.log](https://github.com/zwave-js/node-zwave-js/files/5904605/e5248c75.json.log)

Is it the right json file ?

AlCalzone, 2021-02-01:
```
14:43:36.207 CNTRLR « [Node 002] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Manufacturer Specific
                      · Version
                      · Association Group Information
                      · Powerlevel
                      · Device Reset Locally
                      · Security
                      · Firmware Update Meta Data
```
Doesn't look like it supports Multilevel Switch CC. Which manual do you have?
In [this one](https://www.homecontrols.com/homecontrols/products/pdfs/EZ-Ezlo/EZPHLPSE02US-User-Manual.pdf) there is no mention of that command.

AlCalzone, 2021-02-01:
zwave-js has an api to do that, I think @robertsLando recently exposed it.

Gis70, 2021-02-01:
Ok, if @robertsLando can explain to me how to send such commands. At the end it will be sent from MQTT to zwave-js2mqtt

Thx for your help.

AlCalzone, 2021-02-01:
So, basically you need to call the `sendReport` method for the `Notification` CC with this argument
```jsonc
{
  notificationType: 0x01, // or 0x06 or 0x07 or 0x0A
  notificationEvent: /* see table */,
}
```

robertsLando, 2021-02-01:
@Gis70 https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command

closed 2021-02-01 by robertsLando

Gis70, 2021-02-01:
Thx guys, you should told me, Read The Fucking Manual ;-)

AlCalzone, 2021-02-01:
I can't since my part is not documented yet 😅

Gis70, 2021-02-04:
Hello @robertsLando and @AlCalzone i'm trying to do what it's explained there https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command but i'm not doing well.

My topic Name is zwave
What is _client

Does it look like to that :    (i'm trying simple example to getNodes)

`zwave/_CLIENTS/ZWAVE_GATEWAY-Controleur/api/getNodes()`

robertsLando, 2021-02-04:
`zwave/_CLIENTS/ZWAVE_GATEWAY-Controleur/api/getNodes/set`

Gis70, 2021-02-04:
Ok it's working for getNodes but it's this name :
[image]

But i'm not able to send my command : 

I would send on the nodeId 2 :  
[image]

[image]

For example I want the door chime :  can you help be to send the right syntax

`zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/api/sendCommand/set`

Value :  `{"args":[{"nodeId":2,"commandClass":113,"endpoint": 3,},"startLevelChange",[{6}]]}`

Debug :   

[image]

One thing for the non coder like me should be to make a special page where we can add device compatible and what command to make, a little like for zigbee2mqtt :  [](https://www.zigbee2mqtt.io/information/supported_devices.html)  what do you think ?

robertsLando, 2021-02-05:
What is `[{6}]`?? That's not a valid json

> One thing for the non coder like me should be to make a special page where we can add device compatible and what command to make, a little like for zigbee2mqtt : https://www.zigbee2mqtt.io/information/supported_devices.html what do you think ?

It's on todos

Gis70, 2021-02-18:
Hi @robertsLando @AlCalzone , I need you again, i found on google that we have to modify 1st a parameter here : put in to 128
[image]

So OK with GUI for this.

Then i found we must using BASIC_SET to select sound :  

[image]

How to send BASIC_SET via API or GUI in feature ?

robertsLando, 2021-02-18:
You can using mqtt apis and the `sendCommand` api: https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command

Gis70, 2021-02-18:
{ "args": [ { "nodeId": 2, "commandClass": 6,"endpoint": 0, }, "BASIC_SET", ]}

Publish a JSON like above where commandClass is the number to select sound (here 6 for beep beep )?