zwave-js/zwave-js #1555: GE 14291 switch failing to discover double tap error in logs

crzykidd, 2021-02-01:
**Describe the bug**

When using zwavejs2mqtt 1.0.2 and zwavjs 6.1.1 the GE 14291 switch gives an error on discovery.  This is due to firmware version 5.20 not support double tap, so there is a failure getting the discovery_payload.   
This impacts firmware 5.20 on the GE/Jasco 14291/14292 

**Device information**

Which device(s) is/are affected (make/model)?  GE/Jasco 14291/14292 
What are the node IDs? 13

**Last Known Working Configuration**
- [ ] New device
- [X] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version? 6.0
    - Have you made any recent configuration changes to the device? Describe.
   just upgraded to 6.1.1

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [X ] `zwavejs2mqtt` (latest) docker image
- [X ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Discover switch
2. watch logs

**Additional context**
Discussion in Slack:  
https://zwave2mqtt.slack.com/archives/C01DQ24SDKL/p1612220970387700

**Logfile:** 
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5907524/zwavejs_1.log)

jarrettv, 2021-02-02:
I have a **GE/Jasco 14291 / ZW4005** that seems fine. However I am now seeing same issue with **GE/Jasco 14294 / ZW3005**

blhoward2, 2021-02-02:
@AlCalzone For the 14294, @jarrettv saw the error on his with version 5.26 but then it did discover the command class 32 value on a re-interview. I have a bunch of these also with fw 5.26 that are working fine. Is that an interview error? I did confirm that an earlier version of the fw <5.25 doesn't describe having double-tap in the zwa file so I went ahead and split those, but that wouldn't affect @jarrettv.

jarrettv, 2021-02-02:
`021-02-01 17:55:43.950 INFO ZWAVE: Node 9: interview completed, all values are updated
2021-02-01 17:55:43.952 ERROR GATEWAY: Error while discovering value 9-32-0-event of node 9: Cannot read property 'discovery_payload' of undefined Cannot read property 'discovery_payload' of undefined
TypeError: Cannot read property 'discovery_payload' of undefined
    at Gateway.discoverValue (/home/jv/Dev/zwavejs2mqtt-jv/lib/Gateway.js:1828:25)
    at Gateway.onNodeStatus (/home/jv/Dev/zwavejs2mqtt-jv/lib/Gateway.js:358:12)
    at ZwaveClient.emit (node:events:379:20)
    at ZwaveClient.EventEmitter.emit (node:domain:470:12)
    at ZwaveClient.onNodeStatus (/home/jv/Dev/zwavejs2mqtt-jv/lib/ZwaveClient.js:317:10)
    at ZwaveClient.onNodeInterviewCompleted (/home/jv/Dev/zwavejs2mqtt-jv/lib/ZwaveClient.js:408:16)
    at ZWaveNode.emit (node:events:391:22)
    at ZWaveNode.EventEmitter.emit (node:domain:470:12)
    at ZWaveNode.interview (/home/jv/Dev/zwavejs2mqtt-jv/node_modules/zwave-js/src/lib/node/Node.ts:1021:8)
    at Driver.interviewNode (/home/jv/Dev/zwavejs2mqtt-jv/node_modules/zwave-js/src/lib/driver/Driver.ts:869:10)
2021-02-01 17:55:44.075 INFO ZWAVE: Node 19: value updated: 112-0-3 101 => 101
2021-02-01 17:55:44.138 INFO ZWAVE: Node 20: value updated: 112-0-3 101 => 101
2021-02-01 17:55:44.313 INFO ZWAVE: Node 21: value updated: 112-0-3 101 => 101
2021-02-01 17:55:44.439 INFO ZWAVE: Node 22: value updated: 112-0-3 101 => 101
2021-02-01 17:55:44.591 INFO ZWAVE: Node 24: value updated: 112-0-3 101 => 101
2021-02-01 17:55:44.709 INFO ZWAVE: Node 25: value updated: 112-0-9 1 => 1
2021-02-01 17:55:44.872 INFO ZWAVE: Node 26: value updated: 112-0-5 170 => 170
2021-02-01 17:55:45.074 INFO ZWAVE: Node 27: value updated: 112-0-5 170 => 170
2021-02-01 17:55:45.285 INFO ZWAVE: Node 28: value updated: 112-0-5 170 => 170
2021-02-01 17:55:45.410 INFO ZWAVE: Node 29: value updated: 112-0-3 101 => 101
2021-02-01 17:55:45.459 INFO ZWAVE: Node 30: value updated: 112-0-5 170 => 170
2021-02-01 17:55:45.512 INFO ZWAVE: Node 31: value updated: 112-0-5 170 => 170
2021-02-01 17:55:46.169 INFO ZWAVE: Node 32: value updated: 112-0-6 5 => 5
2021-02-01 17:55:46.301 INFO ZWAVE: Node 33: value updated: 112-0-3 101 => 101
2021-02-01 17:55:46.365 INFO ZWAVE: Node 37: value updated: 112-0-4 101 => 101
2021-02-01 17:55:46.541 INFO ZWAVE: Node 17: value updated: 112-0-4 101 => 101
2021-02-01 17:55:46.671 INFO ZWAVE: Node 19: value updated: 112-0-4 101 => 101
2021-02-01 17:55:46.736 INFO ZWAVE: Node 20: value updated: 112-0-4 101 => 101
2021-02-01 17:55:46.902 INFO ZWAVE: Node 21: value updated: 112-0-4 101 => 101
2021-02-01 17:55:47.031 INFO ZWAVE: Node 22: value updated: 112-0-4 101 => 101
2021-02-01 17:55:47.193 INFO ZWAVE: Node 24: value updated: 112-0-4 101 => 101
2021-02-01 17:55:47.310 INFO ZWAVE: Node 25: value updated: 112-0-10 3 => 3
2021-02-01 17:55:47.472 INFO ZWAVE: Node 26: value updated: 112-0-6 5 => 5
2021-02-01 17:55:47.675 INFO ZWAVE: Node 27: value updated: 112-0-6 5 => 5
2021-02-01 17:55:47.885 INFO ZWAVE: Node 28: value updated: 112-0-6 5 => 5
2021-02-01 17:55:48.001 INFO ZWAVE: Node 29: value updated: 112-0-4 101 => 101
2021-02-01 17:55:48.050 INFO ZWAVE: Node 30: value updated: 112-0-6 5 => 5
2021-02-01 17:55:48.099 INFO ZWAVE: Node 31: value updated: 112-0-6 5 => 5
2021-02-01 17:55:48.770 INFO ZWAVE: Node 32: value updated: 112-0-7 1 => 1
2021-02-01 17:55:48.911 INFO ZWAVE: Node 33: value updated: 112-0-4 101 => 101
2021-02-01 17:55:48.974 INFO ZWAVE: Node 37: value updated: 112-0-5 10 => 10
2021-02-01 17:55:49.146 INFO ZWAVE: Node 17: value updated: 112-0-5 10 => 10
2021-02-01 17:55:49.280 INFO ZWAVE: Node 19: value updated: 112-0-5 33 => 33
2021-02-01 17:55:49.344 INFO ZWAVE: Node 20: value updated: 112-0-5 33 => 33
2021-02-01 17:55:49.514 INFO ZWAVE: Node 21: value updated: 112-0-5 22 => 22
2021-02-01 17:55:49.641 INFO ZWAVE: Node 22: value updated: 112-0-5 24 => 24
2021-02-01 17:55:49.792 INFO ZWAVE: Node 24: value updated: 112-0-5 10 => 10
2021-02-01 17:55:49.913 INFO ZWAVE: Node 25: value updated: 112-0-11 1 => 1
2021-02-01 17:55:50.082 INFO ZWAVE: Node 26: value updated: 112-0-7 1 => 1
2021-02-01 17:55:50.288 INFO ZWAVE: Node 27: value updated: 112-0-7 1 => 1
2021-02-01 17:55:50.558 INFO ZWAVE: Node 28: value updated: 112-0-7 1 => 1
2021-02-01 17:55:50.681 INFO ZWAVE: Node 29: value updated: 112-0-5 10 => 10
2021-02-01 17:55:50.728 INFO ZWAVE: Node 30: value updated: 112-0-7 1 => 1
2021-02-01 17:55:50.776 INFO ZWAVE: Node 31: value updated: 112-0-7 1 => 1
2021-02-01 17:56:01.170 INFO ZWAVE: Node 33: value updated: 112-0-5 10 => 10
2021-02-01 17:56:01.234 INFO ZWAVE: Node 37: value updated: 112-0-6 99 => 99
2021-02-01 17:56:01.405 INFO ZWAVE: Node 17: value updated: 112-0-6 99 => 99
2021-02-01 17:56:01.539 INFO ZWAVE: Node 19: value updated: 112-0-6 99 => 99
2021-02-01 17:56:01.602 INFO ZWAVE: Node 20: value updated: 112-0-6 99 => 99
2021-02-01 17:56:01.774 INFO ZWAVE: Node 21: value updated: 112-0-6 99 => 99
2021-02-01 17:56:01.899 INFO ZWAVE: Node 22: value updated: 112-0-6 99 => 99
2021-02-01 17:56:02.053 INFO ZWAVE: Node 24: value updated: 112-0-6 99 => 99
2021-02-01 17:56:02.169 INFO ZWAVE: Node 25: value updated: 112-0-12 3 => 3
2021-02-01 17:56:32.442 INFO ZWAVE: Node 29: value updated: 112-0-6 99 => 99
2021-02-01 17:56:53.180 INFO ZWAVE: Node 32: value updated: 117-0-local 0 => 0
2021-02-01 17:56:53.181 INFO ZWAVE: Node 32: value updated: 117-0-rf 0 => 0
[... cut ...]
    at ZwaveClient.onNodeInterviewCompleted (/home/jv/Dev/zwavejs2mqtt-jv/lib/ZwaveClient.js:408:16)
    at ZWaveNode.emit (node:events:391:22)
    at ZWaveNode.EventEmitter.emit (node:domain:470:12)
    at ZWaveNode.interview (/home/jv/Dev/zwavejs2mqtt-jv/node_modules/zwave-js/src/lib/node/Node.ts:1021:8)
    at Driver.interviewNode (/home/jv/Dev/zwavejs2mqtt-jv/node_modules/zwave-js/src/lib/driver/Driver.ts:869:10)
2021-02-01 17:57:09.357 INFO ZWAVE: Node 26: interview completed, all values are updated`

jarrettv, 2021-02-02:
```2021-02-01 18:06:10.568 INFO ZWAVE: Calling api refreshInfo with args: [ 25, [length]: 1 ]
2021-02-01 18:06:10.584 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
2021-02-01 18:06:10.749 INFO ZWAVE: Node 25: value added: 114-0-manufacturerId => 99
2021-02-01 18:06:10.750 INFO ZWAVE: Node 25: value added: 114-0-productType => 18756
2021-02-01 18:06:10.750 INFO ZWAVE: Node 25: value added: 114-0-productId => 12344
2021-02-01 18:06:10.828 INFO ZWAVE: Node 25: value added: 134-0-libraryType => 3
2021-02-01 18:06:10.829 INFO ZWAVE: Node 25: value added: 134-0-protocolVersion => 4.34
2021-02-01 18:06:10.829 INFO ZWAVE: Node 25: value added: 134-0-firmwareVersions => 5.26
2021-02-01 18:06:11.339 INFO ZWAVE: Node 25: metadata updated: 38-0-Up
2021-02-01 18:06:11.340 INFO ZWAVE: Node 25: metadata updated: 38-0-Down
2021-02-01 18:06:11.391 INFO ZWAVE: Node 25: value added: 38-0-targetValue => undefined
2021-02-01 18:06:11.392 INFO ZWAVE: Node 25: value added: 38-0-currentValue => 12
2021-02-01 18:06:11.396 INFO ZWAVE: Node 25: metadata updated: 112-0-3
2021-02-01 18:06:11.397 INFO ZWAVE: Node 25: metadata updated: 112-0-4
2021-02-01 18:06:11.397 INFO ZWAVE: Node 25: metadata updated: 112-0-7
2021-02-01 18:06:11.398 INFO ZWAVE: Node 25: metadata updated: 112-0-8
2021-02-01 18:06:11.398 INFO ZWAVE: Node 25: metadata updated: 112-0-9
2021-02-01 18:06:11.399 INFO ZWAVE: Node 25: metadata updated: 112-0-10
2021-02-01 18:06:11.399 INFO ZWAVE: Node 25: metadata updated: 112-0-11
2021-02-01 18:06:11.400 INFO ZWAVE: Node 25: metadata updated: 112-0-12
2021-02-01 18:06:11.454 INFO ZWAVE: Node 25: value added: 112-0-3 => 0
2021-02-01 18:06:11.506 INFO ZWAVE: Node 25: value added: 112-0-4 => 0
2021-02-01 18:06:11.558 INFO ZWAVE: Node 25: value added: 112-0-7 => 1
2021-02-01 18:06:11.610 INFO ZWAVE: Node 25: value added: 112-0-8 => 3
2021-02-01 18:06:11.663 INFO ZWAVE: Node 25: value added: 112-0-9 => 1
2021-02-01 18:06:11.719 INFO ZWAVE: Node 25: value added: 112-0-10 => 3
2021-02-01 18:06:11.776 INFO ZWAVE: Node 25: value added: 112-0-11 => 1
2021-02-01 18:06:11.830 INFO ZWAVE: Node 25: value added: 112-0-12 => 3
2021-02-01 18:06:12.599 INFO ZWAVE: Node 25: value added 25-38-0-targetValue => undefined
2021-02-01 18:06:12.600 INFO ZWAVE: Node 25: value added 25-38-0-duration => [Duration: undefinedseconds]
2021-02-01 18:06:12.602 INFO ZWAVE: Node 25: value added 25-38-0-currentValue => 12
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-38-0-Up => undefined
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-38-0-Down => undefined
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-134-0-libraryType => 3
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-134-0-protocolVersion => 4.34
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-134-0-firmwareVersions => 5.26
2021-02-01 18:06:12.603 INFO ZWAVE: Node 25: value added 25-134-0-hardwareVersion => undefined
2021-02-01 18:06:12.604 INFO ZWAVE: Node 25: value added 25-114-0-manufacturerId => 99
2021-02-01 18:06:12.604 INFO ZWAVE: Node 25: value added 25-114-0-productType => 18756
2021-02-01 18:06:12.604 INFO ZWAVE: Node 25: value added 25-114-0-productId => 12344
2021-02-01 18:06:12.604 INFO ZWAVE: Node 25: value added 25-112-0-3 => 0
2021-02-01 18:06:12.604 INFO ZWAVE: Node 25: value added 25-112-0-4 => 0
2021-02-01 18:06:12.605 INFO ZWAVE: Node 25: value added 25-112-0-7 => 1
2021-02-01 18:06:12.605 INFO ZWAVE: Node 25: value added 25-112-0-8 => 3
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-112-0-9 => 1
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-112-0-10 => 3
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-112-0-11 => 1
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-112-0-12 => 3
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-43-0-sceneId => undefined
2021-02-01 18:06:12.606 INFO ZWAVE: Node 25: value added 25-43-0-dimmingDuration => undefined
2021-02-01 18:06:12.609 INFO ZWAVE: Node 25: value added 25-32-0-currentValue => undefined
2021-02-01 18:06:12.609 INFO ZWAVE: Node 25: value added 25-32-0-targetValue => undefined
2021-02-01 18:06:12.610 INFO ZWAVE: Node 25 ready: GE/Jasco - 14294 / ZW3005 (In-Wall Dimmer Switch)
2021-02-01 18:06:12.611 INFO ZWAVE: Node 25: interview completed, all values are updated```

and 

```2021-02-01 18:57:54.093 INFO ZWAVE: Calling api refreshInfo with args: [ 9, [length]: 1 ]
2021-02-01 18:57:54.105 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
2021-02-01 18:57:54.510 INFO ZWAVE: Node 9: value added: 114-0-manufacturerId => 99
2021-02-01 18:57:54.511 INFO ZWAVE: Node 9: value added: 114-0-productType => 18770
2021-02-01 18:57:54.511 INFO ZWAVE: Node 9: value added: 114-0-productId => 12342
2021-02-01 18:57:54.710 INFO ZWAVE: Node 9: value added: 134-0-libraryType => 3
2021-02-01 18:57:54.711 INFO ZWAVE: Node 9: value added: 134-0-protocolVersion => 4.5
2021-02-01 18:57:54.711 INFO ZWAVE: Node 9: value added: 134-0-firmwareVersions => 5.20
2021-02-01 18:57:56.927 INFO ZWAVE: Node 9: value added: 37-0-currentValue => false
2021-02-01 18:57:56.928 INFO ZWAVE: Node 9: value added: 37-0-targetValue => undefined
2021-02-01 18:57:56.929 INFO ZWAVE: Node 9: metadata updated: 112-0-3
2021-02-01 18:57:56.930 INFO ZWAVE: Node 9: metadata updated: 112-0-4
[... cut ...]
2021-02-01 18:58:00.226 INFO ZWAVE: Node 9: value added 9-43-0-sceneId => undefined
2021-02-01 18:58:00.226 INFO ZWAVE: Node 9: value added 9-43-0-dimmingDuration => undefined
2021-02-01 18:58:00.228 INFO ZWAVE: Node 9: value added 9-32-0-currentValue => undefined
2021-02-01 18:58:00.228 INFO ZWAVE: Node 9: value added 9-32-0-targetValue => undefined
2021-02-01 18:58:00.228 INFO ZWAVE: Node 9 ready: GE/Jasco - 14291 / ZW4005 (In-Wall Paddle Switch)
2021-02-01 18:58:00.229 INFO ZWAVE: Node 9: interview completed, all values are updated```

blhoward2, 2021-02-02:
This is caused by a different bug, at the zwavejs2mqtt level. https://github.com/zwave-js/zwavejs2mqtt/issues/421

PR #1556 (2021-02-03): fix(config): remove double tap support from small number of early ge devices
Closes #1555. A small number of early run 14291/14292 do not support double tap. They run firmware version 5.20. I'm also splitting 14294 as a precaution.
closed 2021-02-02 by blhoward2