zwave-js/zwave-js #1558: Multilevel switch currentValue reports level 255 when setting targetLevel to last (255)

kpine, 2021-02-02:
**Describe the bug**

If you turn on a light using the last level (targetValue 255) zwave-js immediately returns the currentValue as 255, which is a "deprecated" value. I assume this is some new behavior from #1522.

Should the application handle the value 255, or should zwave-js not send it? Unfortunately, if nothing is sent, the problem of flickering UIs returns.

As of now, HA only expects values from 1-99 so the 255 results in HA thinking the light is as bright as the sun (or close to it!). 

**Device information**

Which device(s) is/are affected (make/model)? All dimmers (state exists until a new value is reported, that time various by device). If an application prefers to report targetValue for V4+ switches, it will not be observable.
What are the node IDs? 2

**Last Known Working Configuration**
- [ ] New device
- [X] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version? v6.1.0
    - Have you made any recent configuration changes to the device? Describe. No

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch:  b1b4f74ef95da72f87e68b474e513867d7e33c68
    - `zwavejs2mqtt` branch:  v1.0.1
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Set the multilevel switch targetValue to 255 (turn on to last level)
2. Watch the currentValue go to 255
3. Wait a few seconds for a refresh
4. Watch the currentValue show the actual value

**Additional context**

**Logfile:** [zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5908984/zwavejs_1.log)

blhoward2, 2021-02-02:
I think this is just an issue of the 99 or 255 scale, no? Are you on the most recent? If so, I think @AlCalzone  and @robertsLando need to coordinate which scale they're using for devices as zwavejs2mqtt is defining the 0-99 scale and node-zwave-js is returning 255. https://github.com/zwave-js/zwavejs2mqtt/pull/295

AlCalzone, 2021-02-02:
@kpine's analysis is correct. This is due to the immediate response for the `currentValue` properties. 255 is an edge case I forgot to handle.

Unfortunately, this means that you'll have to handle the missing (or delayed) update somehow in the UI. There's no sound fix for this at the zwave-js side that also avoids flickering.
If we remember the last value somehow, that adds a lot of complexity and there's no guarantee that it is the correct one.
If we don't update `currentValue` (my preferred solution) you will have flickering for devices that don't send updates on their own.

I believe that in the end, there's some change warranted on the HA side. If the zwave-js API call succeeds, it is at least guaranteed that the command was acknowledged by the device. This doesn't mean it will execute it, but you could use that information to update the UI (or keep it from switching back).

kpine, 2021-02-03:
I agree with all that you said. HA should ultimately handle this in some way, after all there must be some other kinds of devices out there (not Z-Wave) that don't report instant states. Assuming an optimistic value as you are doing would work the same way.

Given #1561, it looks like HA can assume it will always receive values of 0-99 then, at least for currentValue? Should it guard against other values?

I also wonder if some HA users will see this for the multilevel switch V4 devices. HA checks the CC version, if it's 4+ it uses targetValue to report the brightness value instead. If HA turns on a V4 switch in this same way, it seems the targetValue will be 255 until the switch reports something, causing the same problem? In that cause, it should guard against 255 anyways (I don't have one of these switches to test).

marcelveldt, 2021-02-03:
@kpine we reverted the targetValue check in HA...
https://github.com/home-assistant/core/pull/45884

PR #1561 (2021-02-03): fix(cc): verify that targetValue is a valid currentValue
fixes: #1558
closed 2021-02-03 by AlCalzone