zwave-js/zwave-js #1573: Home Energy Meter G2 - inaccurate values

snagytx, 2021-02-03:
**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.
When the device sends the periodic reports, 7-112-0-101 set to 6927 and 7-118-0-111 set to 240, the reported values get intermingled. Looking only for (W) I can see:

```
16:45:55.073 CNTRLR   [Node 007] [~] [Meter] value[66049]: 1404.107 => 1379.211         [Endpoint 0]
16:45:55.079 CNTRLR   [Node 007] [~] [Meter] value[66049]: 981.774 => 1379.211          [Endpoint 1]
16:45:55.275 CNTRLR   [Node 007] [~] [Meter] value[66049]: 1379.211 => 960.679          [Endpoint 1]
16:45:55.375 CNTRLR   [Node 007] [~] [Meter] value[66049]: 422.333 => 418.532           [Endpoint 2]
```

Note that Endpoint1 changes to Endpoint0 value first, then to the correct value.

**Device information**

Which device(s) is/are affected (make/model)?  **Device ID: 134-28-2 (0x0086-0x001c-0x0002)**
What are the node IDs? **Node007**

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [x] `zwavejs2mqtt` (1.0.3) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
Set:

7-112-0-101 to 6927
7-118-0-111 to 240

wait for the report.

**Additional context**
Add any other context about the problem here.

**Logfile:** 
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5920160/zwavejs_1.log)

AlCalzone, 2021-02-03:
That looks like this device does not treat endpoint 0 as a mirror of endpoint 1. I can provide a fix later.

AlCalzone, 2021-02-03:
@snagytx can you test against the `fix-root-to-endpoint-mapping` branch?

snagytx, 2021-02-03:
@AlCalzone your change seems to have fixed the issue, I did noticed a behavior with Device ID: 134-100-258 (0x0086-0x0064-0x0102) (AEON Labs  - Multisensor 6), which  I don't think it's related to this change - but I'll open a new issue for it.

PR #1574 (2021-02-03): fix: skip root-to-endpoint mapping if preserveRootApplicationCCValueIDs is set
and enable the flag for Aeon Labs Home Energy Meter G2

fixes: #1573
closed 2021-02-03 by AlCalzone