zwave-js/zwave-js #1582: ZTS-110 Z Wave Thermostat cant change temp heat or cool

typhoonss92, 2021-02-04:
**Describe the bug**

Unable to change the temp for Heat or Cool on the thermostat. I have tried from the integration, and cards in HA. 
ZTS-110 Z Wave Thermostat
I unplugged from the wall and removed batterys to see if that would help. Also removed and readded the node.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?
Node Id 4

**Last Known Working Configuration**
This was working about a week ago on core-2021.1.5 I believe I am on core-2021.20 now
Supervisor-2021.01.7 and Home Assistant OS 5.10
I had Zwave installed for years and started having issues, so I migrated to Zwave_js. I noticed that there are not as many sensors for the device and it seems there is not a fan entity for this thermostat in zwave_js
    
**Installation information**
I installed with integration and add-on Z-wave JS

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**
When I try to change the temp on the thermostat it just changed back to what it was set at after a few seconds. I looked in the Log from the add-on section and I see this over and over

Error in driver ZWaveError: Thermostat Setpoint: "setpoint" must be further specified by a numeric property key
    at Proxy.ThermostatSetpointCCAPI.<computed> (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/commandclass/ThermostatSetpointCC.ts:162:10)
    at ZWaveNode.setValue (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/node/Node.ts:667:14)
    at Function.handle (/usr/lib/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:15:44)
    at Object.node (/usr/lib/node_modules/@zwave-js/server/dist/lib/server.js:30:91)
    at Client.receiveMessage (/usr/lib/node_modules/@zwave-js/server/dist/lib/server.js:61:99)
    at WebSocket.<anonymous> (/usr/lib/node_modules/@zwave-js/server/dist/lib/server.js:35:45)
    at WebSocket.emit (events.js:315:20)
    at WebSocket.EventEmitter.emit (domain.js:467:12)
    at Receiver.receiverOnMessage (/usr/lib/node_modules/@zwave-js/server/node_modules/ws/lib/websocket.js:825:20)
    at Receiver.emit (events.js:315:20) {
  code: 35,
  context: undefined,
  transactionSource: undefined
}

Additionally here is a dump of from the zwave_js integrations page
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/5922785/zwave_js_dump.zip)

**Logfile:**

AlCalzone, 2021-02-04:
Looks like HA is trying to use an incomplete value ID (i.e. setpoint for an unspecified thermostat mode)
@marcelveldt

typhoonss92, 2021-02-04:
I don't know if someone fixed this but there was an update to zwave_js this morning, ran this and rebooted the server. Now it changed the temp and stays changed. Not sure if there is still any backend issues but it seems to be resolved on my end.

AlCalzone, 2021-02-04:
Thanks, closing this - feel free to reopen if the issue persists.

closed 2021-02-04 by AlCalzone