zwave-js/zwave-js #1587: Configurable startup order / value refresh delay

jcam, 2021-02-04:
**Is your feature request related to a problem? Please describe.**
I have some devices with a million entities (rgbw dimmers with multiple inputs, scene control, etc) that take a very long time to get to the "complete" state. 
Devices with higher node ids don't seem to want to get to the RestartFromCache state until the lower-numbered rgbw dimmers get to Complete or at least well along to it. 
Is there a way to force it to run through all nodes to RestartFromCache before it begins the value refresh stage?

And, once it begins that value refresh stage, is there a way to essentially tell it 'slow your roll!' so it doesn't spam the network quite so much? I can control the lowered numbered nodes while it is still running the value refreshes, but it will take 10-20 seconds before a light turn on request will go through, and i usually have to send it 2 or 3 times... I would rather have it take a good bit longer to get to the Complete stage and have usable/functional control while waiting for all valueids to update than have multiple minutes of unresponsiveness.

I have seen a few others report similar things, someone in the HA discord #zwave this morning was asking if there is a "network_complete" state/event because their automations don't necessarily work right if it isn't "done"

**Describe the solution you'd like**
1. delay beginning the valueid updates until all nodes reach RestartFromCache
2. controllable delay between each valueid fetch, rather than just as-fast-as-possible

AlCalzone, 2021-02-04:
I think this goes into the direction of: do we need to refresh all values at all? ==> https://github.com/zwave-js/node-zwave-js/discussions/1432

> asking if there is a "network_complete" state/event because their automations don't necessarily work right if it isn't "done"

Once a node is ready, it should be workable. Obviously I cannot give a blanket statement for all automations out there - we can think about that on a case by case basis.

jcam, 2021-02-04:
I knew there was a discussion about this started already but didn't find it, thanks.

Now that you mention it, I think I agree with LordMike... if the controller missed some updates while it was offline, so be it. It wouldn't be all that different from if there was a bunch of congestion on the mesh and some updates got lost due to collisions. Need to level set that the stored state is best effort, rather than guaranteed.

In a way, this sort of falls in with the polling discussion as well.
I would want my garage door controllers and door locks to be 100% accurate with the stored state, so I would probably set up once every 10 minutes polling on those, just to be sure they're right. Whether a missed push update came about because of a power outage, network congestion, too-much-rain, someone deciding to turn on the tesla coil, or whatever else, isn't all that important, but periodically checking that it is right, is. (i.e. i'm on vacation, the house sitter leaves the door unlocked or the garage open, I'd like to be able to look at a glance and know that at least as of 10 minutes ago, house is secure)

AlCalzone, 2021-02-11:
Please check https://github.com/zwave-js/node-zwave-js/pull/1714 if that improves the situation for you.

jcam, 2021-02-11:
I'm not sure this will have any effect on my issue, as the nodes in question are not sleepy/battery nodes.

AlCalzone, 2021-02-11:
Before this fix, sleeping devices would start with status unknown instead of asleep after restart from cache.
This would cause the first message for them to be transmitted and time out, effectively blocking the send queue for 10s per sleeping device.

AlCalzone, 2021-03-14:
obsolete with https://github.com/zwave-js/node-zwave-js/pull/2030

PR #1787 (2021-03-08): refactor: do not query all values after cache restart
It is time to part with old habits. With this PR (when done) we no longer query all node values on restart and instead leave it up to the user to query them on demand. Because this might require changes in applications, I'm gonna mark this as breaking (although it is a good break).

As a result, traffic and startup times should be reduced dramatically once all nodes have been interviewed. 

fixes: #1587
fixes: #1884
fixes: #1909
resolves the discussion in https://github.com/zwave-js/node-zwave-js/discussions/1432
closed 2021-03-14 by AlCalzone