zwave-js/zwave-js #1588: Add First Alert as a manufacturer and  ZSMOKE Smoke Alarm and ZCOMBO Smoke and CO Alarm

tonygauderman, 2021-02-04:
Would like First Alert as a manufacturer, and their ZSMOKE and ZCOMBO Smoke and Smoke and CO Alarms added.

Below is the manufacturers page:
https://support.firstalert.com/s/article/Z-Wave-alarms

jcam, 2021-02-04:
Are these definitely not supported yet?
First Alert is a BRK brand, and I see ZSMOKE and ZCOMBO under the BRK manufacturer:
https://devices.zwave-js.io/?selectedPath=0x0138:0x0001:0x0001:0.0

stboch, 2021-02-04:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0138/zcombo.json 

but my 3 zcombo's are all Manufacture ID 312 and not 138 as listed above... not sure my units are circa 2014...

jcam, 2021-02-04:
0x138 is decimal 312

stboch, 2021-02-04:
Ok this might be an issue with the devices that are on battery? Not sure why even thought it is beginning the interview process but doesn't seem to move past "ProtocolInfo"

`16:32:19.160 CNTRLR   [Node 035] Beginning interview - last completed stage: None
16:32:19.160 CNTRLR   [Node 035] new node, doing a full interview...
16:32:19.160 CNTRLR » [Node 035] querying protocol info...
16:32:19.170 CNTRLR « [Node 035] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Alarm Sensor
                      specific device class: Unused
                      is a listening device: false
                      is frequent listening: false
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
                      version:               4
16:32:19.171 CNTRLR   [Node 035] The node is now awake.
16:32:19.173 CNTRLR   [Node 035] Interview stage completed: ProtocolInfo
16:32:19.173 CNTRLR » [Node 035] pinging the node...
16:32:25.721 CNTRLR   [Node 035] Querying the node info failed
16:32:25.721 CNTRLR   [Node 035] Interview attempt 1/5 failed, retrying in 5000 ms...
16:32:30.723 CNTRLR   [Node 035] Beginning interview - last completed stage: ProtocolInfo
16:32:30.723 CNTRLR » [Node 035] pinging the node...
16:32:31.939 CNTRLR   [Node 035] The node did not respond after 1 attempts.
                      It is probably asleep, moving its messages to the wakeup queue.
16:32:31.940 CNTRLR   [Node 035] The node is now asleep.
16:32:31.944 CNTRLR   [Node 035] ping failed: The node is asleep
16:32:31.944 CNTRLR   [Node 035] ping failed: The node is asleep
16:32:31.944 CNTRLR » [Node 035] querying node info...
16:32:31.946 CNTRLR » [Node 035] querying node info...
16:32:31.955 DRIVER   unexpected response, discarding...
16:43:07.754 CNTRLR « [Node 035] received unsupported notification {
                          "nodeId": 35,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 13,
                          "alarmLevel": 255
                      }
16:43:48.097 CNTRLR   [Node 035] Beginning interview - last completed stage: None
16:43:48.097 CNTRLR   [Node 035] new node, doing a full interview...
16:43:48.097 CNTRLR » [Node 035] querying protocol info...
16:43:48.104 CNTRLR « [Node 035] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Alarm Sensor
                      specific device class: Unused
                      is a listening device: false
                      is frequent listening: false
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
                      version:               4
16:43:48.105 CNTRLR   [Node 035] The node is now awake.
16:43:48.106 CNTRLR   [Node 035] Interview stage completed: ProtocolInfo
16:43:48.106 CNTRLR » [Node 035] pinging the node...
16:43:54.632 CNTRLR   [Node 035] Querying the node info failed
16:43:54.633 CNTRLR   [Node 035] Interview attempt 1/5 failed, retrying in 5000 ms...
16:43:56.187 CNTRLR « [Node 035] ping successful
16:43:56.187 CNTRLR » [Node 035] querying node info...
16:43:57.118 CNTRLR « [Node 035] node info received
                      supported CCs:
                      · Basic
                      · Battery
                      · Configuration
                      · Association
                      · Notification
                      · Manufacturer Specific
                      · Version
                      controlled CCs:
16:43:57.123 CNTRLR   [Node 035] Interview stage completed: NodeInfo
[... 37 lines cut ...]
16:43:58.105 CNTRLR » [Node 035] querying CC versions...
16:43:58.105 CNTRLR » [Node 035]   querying the CC version for Basic...
16:43:58.215 CNTRLR   [Node 005] Beginning interview - last completed stage: None
16:43:58.215 CNTRLR   [Node 005] new node, doing a full interview...
16:43:58.215 CNTRLR » [Node 005] querying protocol info...
16:43:58.412 CNTRLR   [Node 035]   supports CC Basic (0x20) in version 1
16:43:58.412 CNTRLR » [Node 035]   querying the CC version for Wake Up...
16:43:58.574 CNTRLR   [Node 035]   supports CC Basic (0x20) in version 1
16:43:58.574 CNTRLR » [Node 035]   querying the CC version for Wake Up...
16:43:59.635 CNTRLR   [Node 035] Beginning interview - last completed stage: NodeInfo
16:43:59.635 CNTRLR   [Node 035] ManufacturerSpecificCC: doing a partial interview...
16:43:59.638 CNTRLR   [Node 035] VersionCC: doing a complete interview...
16:43:59.638 CNTRLR » [Node 035] querying node versions...
16:44:12.092 CNTRLR   [Node 035] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
16:44:18.903 CNTRLR   [Node 035] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
16:44:25.853 CNTRLR   [Node 035] The node did not respond after 3 attempts.
                      It is probably asleep, moving its messages to the wakeup queue.
16:44:25.854 CNTRLR   [Node 035] The node is now asleep.
16:44:42.873 CNTRLR « [Node 035] received unsupported notification {
                          "nodeId": 35,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 12,
                          "alarmLevel": 255
                      }`

AlCalzone, 2021-02-04:
Something weird is happening there (double interview processes). Did you `refreshInfo`? If so, restart zwave-js, don't `refreshInfo` and try to wake up the device manually - which seemed to work but was garbled by the duplicated interview.

stboch, 2021-02-04:
When restarting the wake time for the device is fairly short lived... I was a few different options to try and get it to be reinitialized, sorry new to zwave-js and still learning. 

Now its in the NodeInfo state and doesn't seem to be moving along...

tonygauderman, 2021-02-05:
These devices are notorious for sleeping quickly before the interview process is done.  Somehow the device add process works with SmartThings reliably, but no idea how.

dshokouhi, 2021-02-05:
Try to go through the pairing process twice to complete the interview for the ZCOMBO  perform the first one and watch the logs and once it stops do the inclusion steps on the device again.

tonygauderman, 2021-02-05:
Yea, been doing that, and it adds, but it's a generic Node.  I do get battery level and battery level alarm, but that's it.  Each time my logs like very similar to the ones attached above.

blhoward2, 2021-02-05:
These devices have always required multiple wakeups to fully interview. You need to do the whole battery door open, closed while holding button thing, until you see Interview Complete in the logs.

kpine, 2021-02-05:
> Yea, been doing that, and it adds, but it's a generic Node. I do get battery level and battery level alarm, but that's it. Each time my logs like very similar to the ones attached above.

Make sure you are running v6.1.2. The older ZCOMBO uses Alarm V1 reports, and support was just added for that.

closed 2021-02-08 by zwave-js-assistant[bot]