zwave-js/zwave-js #1592: Fibaro Door Sensor 2, Door open/closed unavailable.

valuta1995, 2021-02-04:
**Describe the bug**
Added a new device to a completely clean network. Fibaro Door/window sensor 2. Most entities associated with it are visible, however the open/close binary sensor reports as unavailable. The ready message has come in.

**Device information**
Which device(s) is/are affected (make/model)?  https://manuals.fibaro.com/door-window-sensor-2/
What are the node IDs? nodeId: 9 (in attached dump)

**Last Known Working Configuration**
- [x] Previously working device (other platform)
    - Which platform? OpenOZW through (installed through home-assistant OS on raspi4)
    - Have you made any recent configuration changes to the device? None
    - Moved from OpenOZW to node-zwave-js (installed through home-assistant OS on raspi4)
    
**Installation information**
How did you install `node-zwave-js`?
- [X] Other: Through home-assistant os: Add-on version 1.4, and integration

**To Reproduce**
Steps to reproduce the behavior:
Start with an empty network, include the fibaro sensor in the network and wait for it to set up (ready). Observe outputs from the sensor. Temperature and tamper switches behave as expected. Door open/close does not (unavailable).

**Additional context**
Add any other context about the problem here.

**Logfile:** While I understand the need for a log file and understand how to upload one, I have no idea how to pass the required ENV variable to the home-assistant add-on.
[zwave_js_dump.jsonl](https://github.com/zwave-js/node-zwave-js/files/5927444/zwave_js_dump.jsonl.txt)

AlCalzone, 2021-02-04:
Can you open the zwavejs2mqtt UI and create a logfile there? As far as I know that should be possible.
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report

valuta1995, 2021-02-04:
Thanks for the response. I do not have zwavejs2mqtt, but if this is the accepted way of getting these logs I can install it first thing tomorrow morning. It is a testing system at the moment, so if you need me to do something to reset it completely to help figure this out, that is fine too.

AlCalzone, 2021-02-04:
I've talked to the HA devs. They'll add a way to extract logs from HA, so if you can wait a bit that is also okay.

valuta1995, 2021-02-04:
I have no rush, so I'd you're busy at the moment I'll wait it out.

Petro31, 2021-02-11:
@valuta1995 are you running a supervised installation or HassOS?

valuta1995, 2021-02-11:
@Petro31 I am indeed.

Petro31, 2021-02-11:
@valuta1995 Supervised is not the same as HassOS, so the answer should be 'HassOS' or 'Supervised'.  Based on that, I would assume you're running HassOS.

To get the logs, you need to SSH into the system and get the logs via CLI commands.

Otherwise, if you go to the addon, you can copy/paste the last 100 log lines from the logs tab.

Petro31, 2021-02-11:
Also, I have this device.  If need be and can exclude / include it.  The device works for me, however it was not found as a 'known device'.

valuta1995, 2021-02-11:
Petro, I completely misread the question as "are you running your installation on HassOS". To which the answer is yes. Your reply has prompted me to take a look at my logs (and history). As it turns out, an upade, two days ago seems to have fixed the issue.

[image]

closed 2021-02-11 by valuta1995