zwave-js/zwave-js #1596: Add support for Fantem Door/Window Sensor and MultiSensor

akoolenbourke, 2021-02-04:
Fantem OEM Aeotec devices so should have the same capabilities. These are the values for the sensor that ZWave JS shows in the Manufacturer Specific section (Using the ZWaveJS2MQTT addon)

Door/Window Sensor (Fantem)
Man: 362 == 0x16a 
Prod Type: 514 == 0x202
Prod ID: 112 == 0x4a

Motion Sensor
Man: 362 == 0x16a
Prod Type: 514 == 0x202
prod Id: 100 == 0x64

AlCalzone, 2021-02-04:
I've added a new issue template to get a hold of these kinds of requests to make it easier for us. Could you please add the missing information to your request:

---

(first two points are done)

**Did you check if we really don't have it?**

- [ ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

If so, it helps in creating the config file.

akoolenbourke, 2021-02-04:
Thanks.

Not in DB
Not merged or pending PR
It's not in the TODO List. The zwave alliance site lists it for China and US/Canada/Mexic but not for AUS/NZ which is my territory.

The door/window sensor and the multisensor are there for the Aeotec (OEM) items for my territory.

Door/Window
https://products.z-wavealliance.org/products/1616?selectedFrequencyId=4

MultiSensor
https://products.z-wavealliance.org/products/1404?selectedFrequencyId=4

[OomiMultiSensor.pdf](https://github.com/zwave-js/node-zwave-js/files/5928998/OomiMultiSensor.pdf)
[Oomi Door Window Sensor manual.pdf](https://github.com/zwave-js/node-zwave-js/files/5928999/Oomi.Door.Window.Sensor.manual.pdf)

cdnninja, 2021-04-25:
I have added the door window sensor as a pull request.  I will work on the Mutli sensor another day.

cdnninja, 2021-04-28:
From what I found I think these devices already exist with different product IDs.  This guy: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x016a/ft112-k.json and https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x016a/ft100.json. I will keep reading to see if I can confirm.  If they are the same I will update these with additional product IDs.

PR #2597 (2021-05-12): fix(config): align Fantem Door Window Sensor to Aeotec files
Resolves #1596. Aligns to AEON files and merging with FT112-K.  Looks like the letters are just revisions of the device for markets, much like AEON host numerous productIDs.  The product IDs match AEON so could re-use them but at this point thinking we just add as we find official mention.
closed 2021-05-12 by AlCalzone