zwave-js/zwave-js #1602: [bug] manual updates on schlage BE468 zwave lock aren't sent to MQTT

radiocolin, 2021-02-04:
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version

Build/Run method

- [ ] Docker
- [ ] PKG
- [X] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.0.3.ed6fbe3
zwavejs version: 6.1.1

### Describe the bug

Schlage BE468 is not updating properly after receiving manual lock events.

### To Reproduce

Manually lock or unlock the lock.

### Expected behavior

The lock will update its status.

### Additional context

The gateway is receiving notifications of manual lock/unlock events, but these events aren't consistently producing an MQTT event. The MQTT event sometimes fires if a manual lock is triggered immediately after a software unlock. The majority of the time, the manual operation is noted but a new MQTT event is not triggered.

In the logs below, I toggle the lock manually four times, then unlock and lock it via MQTT, then toggle it manually again. My actions are noted in italics.

Logs:

*manual operation*
2021-02-03 22:19:23.365 INFO ZWAVE: Node 5: notification: Manual unlock operation 
2021-02-03 22:19:27.161 INFO ZWAVE: Node 5: notification: Manual lock operation 
2021-02-03 22:19:31.750 INFO ZWAVE: Node 5: notification: Manual unlock operation 
2021-02-03 22:19:36.758 INFO ZWAVE: Node 5: notification: Manual lock operation 
*operate via MQTT*
2021-02-03 22:19:39.849 INFO MQTT: Message received on zwave/frontstoop/frontdoorlock/98/0/targetMode/set, '255'
2021-02-03 22:19:39.851 INFO ZWAVE: Writing 255 to 5-98-0-targetMode
2021-02-03 22:19:47.778 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 255
2021-02-03 22:19:47.779 INFO ZWAVE: Node 5: value updated: 98-0-outsideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:19:47.781 INFO ZWAVE: Node 5: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:19:47.782 INFO ZWAVE: Node 5: value updated: 98-0-latchStatus open => open
2021-02-03 22:19:47.783 INFO ZWAVE: Node 5: value updated: 98-0-boltStatus unlocked => locked
2021-02-03 22:19:47.786 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus open => open
*operate via MQTT*
2021-02-03 22:19:52.322 INFO MQTT: Message received on zwave/frontstoop/frontdoorlock/98/0/targetMode/set, '0'
2021-02-03 22:19:52.323 INFO ZWAVE: Writing 0 to 5-98-0-targetMode
2021-02-03 22:19:54.875 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 255 => 0
2021-02-03 22:19:54.877 INFO ZWAVE: Node 5: value updated: 98-0-outsideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:19:54.879 INFO ZWAVE: Node 5: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:19:54.881 INFO ZWAVE: Node 5: value updated: 98-0-latchStatus open => open
2021-02-03 22:19:54.882 INFO ZWAVE: Node 5: value updated: 98-0-boltStatus locked => unlocked
2021-02-03 22:19:54.884 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus open => open
2021-02-03 22:20:00.306 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 0
2021-02-03 22:20:00.309 INFO ZWAVE: Node 5: value updated: 98-0-outsideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:20:00.311 INFO ZWAVE: Node 5: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-02-03 22:20:00.314 INFO ZWAVE: Node 5: value updated: 98-0-latchStatus open => open
2021-02-03 22:20:00.317 INFO ZWAVE: Node 5: value updated: 98-0-boltStatus unlocked => unlocked
2021-02-03 22:20:00.319 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus open => open
*manual operation*
2021-02-03 22:20:05.341 INFO ZWAVE: Node 5: notification: Manual lock operation

towerhand, 2021-02-04:
Would you like the logs for the RF (un)lock which is reporting properly to compare or is it totally unrelated?

AlCalzone, 2021-02-04:
Won't hurt I guess :)

towerhand, 2021-02-05:
2021-02-04 17:49:31.098 INFO ZWAVE: Node 22: notification: RF lock operation 
2021-02-04 17:49:31.102 DEBUG MQTT: Publishing to zwave2jsmqtt/yale_door_lock/notification/RF_lock_operation: { time: 1612482571100, value: null } with options { qos: 0, retain: true }

I'm adding another log just incase you would like more info, I waited for the interview process to finish then (un)lock manually and (un)lock with remotely (RF). Removed everything from the log prior the first manual unlock so its "only" 11 MBs instead of +200 MBs.
[zwavejs2mqtt.log.zip](https://github.com/zwave-js/node-zwave-js/files/5929202/zwavejs2mqtt.log.zip)

Ikcelaks, 2021-02-05:
The original HA zwave integration had functional work-arounds for the BE469 and other door locks: https://github.com/home-assistant/core/commits/dev/homeassistant/components/zwave/lock.py

They applied two workarounds for BE469.

1. When receiving a Notification message, they updated the lock state according to the notification received.
2. When receiving Door Lock CC messages, they generated a corresponding notification event.

I really liked this setup, because it meant that both the lock state and the last received notification event give accurate representations of the current lock status.

DaveWBernard, 2021-02-05:
I have the same issue with a Keywe lock. This was solved in open-zwave: [https://github.com/OpenZWave/open-zwave/pull/1955](https://github.com/OpenZWave/open-zwave/pull/1955)

Comments from the commit:

AlCalzone, 2021-02-05:
@towerhand I know it is a bit confusing but I cannot use the zwavejs2mqtt logs - I need the one node-zwave-js creates.

@DaveWBernard could you also make a log of the device interaction?

towerhand, 2021-02-05:
Sorry @AlCalzone, didn't even know there are different logs. Any pointers on how to download them when using zjs2m with docker compose?

I found this: https://zwave-js.github.io/node-zwave-js/#/development/troubleshooting?id=providing-the-necessary-information-for-an-issue but I'm not sure if it works with docker compose and if it saves it on the same folder as the zjs2m logs.

AlCalzone, 2021-02-05:
@towerhand check the zjs2mqtt docs, there is a section on that.

towerhand, 2021-02-05:
Got it, I was confused because the toggle for the zwave-js logs is next to the toggle for enabling the zwave-js websocket server so I thought it was for logging that instead.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/5935673/zwavejs2mqtt-store.zip)

dnikles, 2021-02-06:
When this does get sorted out, it would also be helpful to have the unlock method reported as well. I'm currently getting this from sensor.schlage_allegion_be469zp_connect_smart_deadbolt_alarm_type in the deprecated zwave integration. It returns 19 for keypad unlock and 22 for manual unlock.

AlCalzone, 2021-02-06:
Any chance you got that as a zwave-js log?

thegame3202, 2021-02-06:
> Any chance you got that as a zwave-js log?

Did you see mine above? Do you need another or more detail? Happy to get more!

AlCalzone, 2021-02-06:
My comment was meant for @dnikles - thought he was talking about something else. Didn't have a chance to look at the logs yet.

Ikcelaks, 2021-02-07:
> When this does get sorted out, it would also be helpful to have the unlock method reported as well. I'm currently getting this from sensor.schlage_allegion_be469zp_connect_smart_deadbolt_alarm_type in the deprecated zwave integration. It returns 19 for keypad unlock and 22 for manual unlock.

I have the older BE469 lock, and the alarm type information is coming through in HA as zwave-js-event notification events. The zwave-js integration isn't creating working senors from these notifications, but zwave-js is sending them, and they are creating events in HA that you can listen for. So, I'm pretty sure this library's driver is handling them fine for this purpose.

dnikles, 2021-02-07:
> I have the older BE469 lock, and the alarm type information is coming through in HA as zwave-js-event notification events. The zwave-js integration isn't creating working senors from these [...]

You are correct, thank you. I am able to listen to event zwave-js-event for label "Keypad unlock operation" or "Manual unlock operation". This addresses my previous concern.

AlCalzone, 2021-02-07:
Please test https://github.com/zwave-js/node-zwave-js/pull/1641
If anyone needs an npm package to test, I can release one.

towerhand, 2021-02-07:
I’ll test in a few minutes, gotta build it.

Ikcelaks, 2021-02-07:
Could the PR be expanded to also cover notification Event 0x05 (keypad lock) and 0x06 (keypad unlock)? Otherwise this only half fixes the issue for the BE469.

AlCalzone, 2021-02-07:
PR updated, should be supported now

closed 2021-02-09 by AlCalzone

drobtravels, 2021-02-21:
With my Yale YRD210, I'm still not having the lock status is not updating.  This occurs whether using keypad or manually.  Im on Zwavejs 6.4.0.  Let me know if I should open a separate issue.

I believe the relevant log lines are below.  I'll try to do some further digging / troubleshooting tomorrow.

Log when I unlocked the door manually:
```
20:53:24.191 CNTRLR   [Node 086] [~] [Notification] notificationMode: "unkno [Endpoint 0] [internal]
20:53:24.192 CNTRLR   [Node 086] [~] [Notification] alarmType: 22 => 22                 [Endpoint 0]
20:53:24.192 CNTRLR   [Node 086] [~] [Notification] alarmLevel: 1 => 1                  [Endpoint 0]
```

Locking the door manually
```
20:53:24.565 CNTRLR   [Node 086] [~] [Notification] notificationMode: "unkno [Endpoint 0] [internal]
20:53:24.565 CNTRLR   [Node 086] [~] [Notification] alarmType: 22 => 21                 [Endpoint 0]
20:53:24.566 CNTRLR   [Node 086] [~] [Notification] alarmLevel: 1 => 1                  [Endpoint 0]
```

When I use the keyboard, alarmType is 24 for unlock

AlCalzone, 2021-02-21:
Your device uses the old (unstandardized) version of the CC so you'll have to wait for #1713 or poll the lock status with an automation.

ragaimeena, 2021-09-29:
Sorry to ask this question here, please direct me to the right thread if this is inappropriate to ask here
I am about to migrate my Schlage lock from smartthings to HA. Is this integration and PR fully functional now through the regular integration process or do I have to change any json files manually?
Thank you

PR #1641 (2021-02-09): feat(cc): map Notification CC (un)lock operations to (Door) Lock CC
With this PR, we handle manual (un)lock operations reported through Notification CC and set the current lock status for (Door) Lock CC accordingly.

fixes: #1602