zwave-js/zwave-js #1608: Store timestamp along with the value

AlCalzone, 2021-02-05:
as requested in https://github.com/zwave-js/zwavejs2mqtt/issues/391

chrish987, 2021-02-19:
any potential timeline on this feature?

AlCalzone, 2021-02-19:
Tbh, there are more pressing issues on my end and the changes will be pretty invasive since the value storage is a core part of the library and used **everywhere**.

chrish987, 2021-02-20:
does that mean it is unlikely to be ever implemented? or weeks or months away?

AlCalzone, 2021-02-21:
I do see some value in this (e.g. auto-idling notifications across a restart could benefit). But looking at my issue pile, it is rather months away.

AlCalzone, 2022-09-28:
@raintonr I'd probably do it in the ValueDB, but I haven't thought it through completely. This will also need to consider edge cases like updates that don't actually come from the device.

In any case, a PR for this likely won't land before 2023 -> https://github.com/zwave-js/node-zwave-js/issues/5129

closed 2023-03-10 by AlCalzone