zwave-js/zwave-js #1609: Qubino ZMNHJD1 not working

efesser, 2021-02-05:
**Describe the bug**

Two of my 3 Qubino ZMNHJD1 are not working. They are visible on the network but the entities (lights and switches) are reported "Unavailable" in my Home Assistant installation.

**Device information**

Device working as expected: Node 10 - Qubino ZMNHJD1, **firmware version reported 1.1**

Devices **not** working: Node 12 and 13 - Qubino ZMNHJD1, **firmware version reported 2.0**

**Last Known Working Configuration**
- [x] Previously working device (other platform)
    - OpenZWave 1.4 (deprecated Home Assistant integration)
    
**Installation information**
- `Z-Wave JS` Home Assistant add-on
- `Z-Wave JS to MQTT` Home Assistant add-on (for getting the log files)

**To Reproduce**
Start network, include devices, after everything is ready, the entities are unavailable in Home Assistant for 2 of the 3 nodes, even after restarting everything and healing network.

**Additional context**
Could it be related to the differences between firmware 1.1 and 2.0?
The working node is reporting a temperature sensor (value is -999.9 because there is no sensor connected), whereas the two others are not reporting it.

**Logfile:** 
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5931669/zwavejs2mqtt.log)
[zwavejs_377.log](https://github.com/zwave-js/node-zwave-js/files/5931670/zwavejs_377.log)

AlCalzone, 2021-02-05:
I can't speak for the "unavailable", that is something specific to HASS. /cc @marcelveldt

What I can say however is that the nodes 12 and 13 don't respond to most queries, so many of the application values are simply empty (✔= response, ❌= no response):

| CC                        |    Node 10    | Node 12 | Node 13 |
| ------------------------- | :-----------: | :-----: | :-----: |
| Multi Channel             |       ✔       |    ✔    |    ✔    |
| Association               | not supported | invalid | invalid |
| Multi Channel Association |       ✔       |    ✔    |    ✔    |
| Basic                     |       ✔       |   ❌    |   ❌    |
| Binary Switch             |       ✔       |   ❌    |   ❌    |
| Binary Sensor             |       ✔       |   ❌    |   ❌    |
| Configuration             |       ✔       |   ❌    |   ❌    |
| Multilevel Switch         |       ✔       |   ❌    |   ❌    |
| Multilevel Sensor         |       ✔       |   ❌    |   ❌    |

I can't tell why that is, because the commands are the same for all three nodes.

marcelveldt, 2021-02-05:
An entity will render as unavailable if Home Assistant if something is wrong:
- the node does not have the "ready" state yet
- the value is missing 

So exactly what @AlCalzone is describing, the values are not there

AlCalzone, 2021-02-05:
Hm, IMO you should at least be able to control the thing, even if it has no values.

etienn01, 2021-02-05:
@AlCalzone I just tried to set `[12-38-0-targetValue] Target value` from the Control Panel and it works.
The entity became "available" in Home Assistant after I called "Refresh Values" on that node and reloaded the integration. 

However, when I switched back from the `Z-Wave JS to MQTT` add-on to `Z-Wave JS`, they're again showing as unavailable.

AlCalzone, 2021-02-05:
That's a HASS issue, nothing I can do about it.

etienn01, 2021-02-05:
Thanks for your help @AlCalzone, I'll report on HASS side 👍

closed 2021-02-05 by etienn01

marcelveldt, 2021-02-05:
> Thanks for your help @AlCalzone, I'll report on HASS side 👍

No need, we're already on it ;-)

etienn01, 2021-02-05:
Nice @marcelveldt! Can you link me to an issue I can keep an eye on?