zwave-js/zwave-js #1612: FGR-222 support missing

lorenzstorm1, 2021-02-05:
**Which device is missing?**
Fibaro FGR-222

Enter Manufacturer and Device name here
Fibaro FGR-222

**What are the IDs?**

Manufacturer ID: 0x010F
Product type: ?
Product ID: ?  

**Did you check if we really don't have it?**
you just got FGR**M**-222

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://manuals.fibaro.com/roller-shutter-2/

If so, it helps in creating the config file.

AlCalzone, 2021-02-05:
Unless we know the product type and ID, we cannot act on this. It might actually be that the label is just wrong. I have several "Roller Shutter 2".

lorenzstorm1, 2021-02-05:
I found out the Product ID is: 0x010F.0x0300.0x0102

Source:
http://manuals-backend.z-wave.info/make.php?lang=en&sku=FIBEFGR-222&cert=ZC08-11080006

AlCalzone, 2021-02-05:
Interesting, we have that as a device ID for the FGR-221:
https://devices.zwave-js.io/?selectedPath=0x010f:0x0102:0x0300:0.0

I guess some of the other IDs there might be wrong too.

PR #1651 (2021-02-08): fix: correct label for FGR222
Closes #1612. @AlCalzone ready for review
closed 2021-02-08 by AlCalzone