zwave-js/zwave-js #1614: Bug with Qubino ZMNHDD1

MacKaarstad, 2021-02-05:
I noticed some strange behaviour with Qubino Flush Dimmer (ZMNHDD1) when dimming.
It seems that the dim/transtition time when controlled from HASS is set by the parameter setting "Dimming time when key pressed"(parameter 66). 
When I change parameter 66 to 0(s) I can see instant response, but this makes dimming with the switch impossible. Default value is 3(s).

AlCalzone, 2021-02-05:
Sorry, what is the question / issue?

geirra, 2021-02-05:
Param 66 on dimmer controlls physical dimming time 
`Choose the time during which the Dimmer will move between the min. and max. dimming
values during a continuous press of the push-button I1 or by an associated device. `

Did you test tweaking param 68 instead?

MacKaarstad, 2021-02-06:
Param 66 shuld only control physical dimming time. But now it also controlls dimming time when controlled from HASS.

Param 68 does not affect this.

AlCalzone, 2021-02-06:
What a device does when it's configuration is changed is entirely up to the device. We don't actively use configuration parameters to influence the dimming behavior - that is up to the user.

It might be that the meaning of your config parameters has changed with a firmware update and our config files don't reflect that. If that is the case, we'd need to know which firmware changed the parameters how.

MacKaarstad, 2021-02-06:
Qubino does not distribute their firmware files, and I have not updated it. This exact unit does not behave this way on the OZW 1.6 beta or in HomeSeer. 

What can I do to assist in solving this issue?

AlCalzone, 2021-02-06:
To be honest I still don't understand what exactly the issue is. Maybe you could demonstrate with a log and explanations what you (don't) expect to happen.

MacKaarstad, 2021-02-06:
Not able to post logs at the moment.

I don't expect the dimmer to fade slowly from actual level to new level when it is controlled from HASS. 

The dimmer used to jump to new level instantaneously when controlled from software. But now it seem to pick up the time set in param 66 and use that for transtition time between the actual and new level.

Example with param 66 set to 3:
light is off
*switch on to full brightness* 
light turns on, but is taking 3s to reach full brightness
(The same goes for turning off the light)

Param 66 is set to 0:
light is off
*switch on to full brightness* 
light turns on and is instantly at full brightness
(The same goes for turning off the light)

AlCalzone, 2021-02-06:
Oh so this is a duplicate of #1321 then. When not specified, devices use the default transition duration (which is working as intended). And in your case the default transition duration can be configured with the mentioned parameter.

AlCalzone, 2021-02-07:
Duplicate of #1321

closed 2021-02-07 by AlCalzone