zwave-js/zwave-js #1615: No on/off switch configuration detected for TKB Home - TZ69G - Smart energy plug in switch

robinsmidsrod, 2021-02-05:
**Describe the bug**

The on/off switch configuration is not detected, so only meter values are reported to Home Assistant.

**Device information**

Which device(s) is/are affected (make/model)?

TKB Home - TZ69G `(0x0118-0x3119-0x0116)`
https://products.z-wavealliance.org/products/1640?selectedFrequencyId=1

What are the node IDs? `15` 

**Last Known Working Configuration**
- [X] Previously working device (other platform)
    - Which platform? OpenZWave 1.6
    - Have you made any recent configuration changes to the device? Not really. I tried excluding it and including it again, but it didn't really change anything. I've added a log of the inclusion process.
    
**Installation information**
How did you install `node-zwave-js`?
- [X] `zwavejs2mqtt` (latest) docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configuration Panel' in Zwavejs2Mqtt
2. Click on 'Node 15'
3. Scroll down to 'Values'
4. There is no entry for 'Binary Switch', only 'Meter', 'Version' and 'Manufacturer Specific'.

**Additional context**
In OpenZWave 1.6 I had the same issue as well, but after twiddling a lot with the XML configuration I was able to get something working. I've attached the log from the inclusion process in zwavejs, the JSON generated in zwavejs and the custom XML definition from OpenZWave 1.6 that I tweaked myself. Hopefully this should allow you to add the required command classes to make it work properly.

[tkbhome_tz69g_node_15.log](https://github.com/zwave-js/node-zwave-js/files/5935820/tkbhome_tz69g_node_15.log)
[tkbhome_tz69g_node_15.json.txt](https://github.com/zwave-js/node-zwave-js/files/5935823/tkbhome_tz69g_node_15.json.txt)
[wenzhou_tz69.xml.txt](https://github.com/zwave-js/node-zwave-js/files/5935822/wenzhou_tz69.xml.txt)

AlCalzone, 2021-02-07:
Unfortunately that is not the correct log. I need the zwave-js log (the one that begins with)
```
███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
  ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
 ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
```

robinsmidsrod, 2021-02-08:
@AlCalzone Here is the log you (hopefully) need. The node ID in question is **15**. I do see some indications in the log that it claims to not support the BinarySwitch CC. I guess my file from OpenZWave should at least point you in the right direction of what needs to be done to make it work.

[zwavejs_6.log.txt](https://github.com/zwave-js/node-zwave-js/files/5942615/zwavejs_6.log.txt)

AlCalzone, 2021-02-08:
Yeah, there is something weird. It identifies as a Binary Switch type, but then claims NOT to support the Binary Switch CC.

AlCalzone, 2021-02-08:
The device is not behaving correctly. The Binary Switch CC needs to be marked as supported through the config file.

robinsmidsrod, 2021-02-08:
@AlCalzone Is there something you need me to do? Maybe test out a config file or something like that? I'm running the zwavejs2mqtt 1.0.5 docker container from Docker Hub. I can switch to a dev release on Docker Hub if that is the simplest way to test things out.

I'm also not surprised this device behaves weird. While I was using it with OpenZWave 1.6 I think it responded to CC 37 for the switch-related behavior and CC 133 for association/lifeline behavior (see the attached ozw XML file). Also, it would not store configuration at all, so if I tried to switch the behavior of the power on/off LED, it would revert back to default settings if I power-cycled it. It would also spam a lot of metering messages, even though I set a slower reporting interval. So all in all, not a very well made device.

AlCalzone, 2021-02-08:
You can try to edit the config file yourself and add the CC `0x25` using a compat flag:
https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=commandclassesadd

Otherwise I should find some time later tonight.

robinsmidsrod, 2021-02-08:
@AlCalzone Can this modification be done by just adding a modified file in the store/ directory, or do I have to rebuild the container myself? I'm not familiar with node-js development (yarn etc.). Looking at the container I notice that the JSON file is present twice in the generated container:

```
/usr/src/app # find . | grep tz69
./node_modules/@zwave-js/server/node_modules/@zwave-js/config/config/devices/0x0118/tz69.json
./node_modules/@zwave-js/config/config/devices/0x0118/tz69.json
```

Which one of these should I overwrite?

It would be so much easier to test compatibility improvements like this one if it was possible to read a device config file from the store/ directory (which is typically bind-mounted via zwavejs2mqtt) before falling back to the bundled JSON files.

AlCalzone, 2021-02-08:
I'm not sure why the server has a different dependency here. @robertsLando something to do with linking dependencies incorrectly maybe?

Actually you should be able to mount the file itself as a volume, overwriting whats in the container. I'm not sure which one though - that needs to be sorted out.

robinsmidsrod, 2021-02-08:
@AlCalzone I've bind-mounted both the files, as you weren't sure which one to modify, and added this at the end of my tz69.json:

```
    "compat": {
        "commandClasses": {
            "add": {
                // Adds the CC BinarySwitch
                "0x25": {
                    /** Whether the endpoint or node can react to this CC */
                    "isSupported": true,
                    /** Whether the endpoint or node can control other nodes with this CC */
                    "isControlled": false,
                }
            }
        }
    }
```

This was the minimum of information I could add to make it work. This device doesn't seem to support security at all, and I'm unsure which version of the BinarySwitch it should support, if it is version 1, 2 or something else. I'm not sure if `isControlled` has the right value, but from the documentation of this device it seems to only support Lifeline association to a single node. Have I understood it correct that this means isControlled=false? If so, that means this can be added to tz69.json and I think I should be happy. My lava lamp can now be controlled again via Home Assistant (using websocket API).

AlCalzone, 2021-02-08:
isControlled means that the device controls the CC in other devices, e.g. via an association. For our purposes this is mostly informational, so it can be left out.

AlCalzone, 2021-02-08:
@robinsmidsrod could you give the linked PR a quick glance whether that was the correct file?

robinsmidsrod, 2021-02-09:
@AlCalzone The PR looks good. It is the correct file. But as I mentioned, if I only had the `isSupported: true` (and no `isControlled: false`) on the root endpoint, it didn't generate a BinarySwitch section for me. It might've been just my understanding of how to get the node to update its configuration (I used both re-interview and refresh values commands). But if you claim that is enough I'll trust you if you say so. :)

What is the correct approach to force usage of the new file on a node? I updated the file, restarted the zwavejs2mqtt container, waited for the mesh to load, then I sent a re-interview node command on this particular problematic node and waited for it to complete.

AlCalzone, 2021-02-09:
`isControlled` is not evaluated by zwave-js

> What is the correct approach to force usage of the new file on a node? I updated the file, restarted the zwavejs2mqtt container, waited for the mesh to load, then I sent a re-interview node command on this particular problematic node and waited for it to complete.

Sounds correct

PR #1666 (2021-02-09): fix(config): force Binary Switch support for TKB Home TZ69
fixes: #1615
closed 2021-02-09 by AlCalzone